dhrone/pydPiper #1: 1st column is always on on a WEH LCD

JedS, 2017-03-13:
Hi
First I would like to congrats you for your job... Very useful.
I try to used it on a Audiophonics RasDac V3 (RPi3+ Audiophonics DAC I-Sabre V3 ES9023 TCXO).
On this system, the LCD is a Winstar WEH001602A model.
Everything works fine except the 1st column on the screen is always on.
[image]
I launch pipPyper with the following command line :
`/usr/bin/pyhton2 pydPiper.py --volumio --pages pages_fixed.py`
I tried every pages you provide, but the problem remains the same.
If I kill the pydPiper process, the column remains on.
If I don't use pydPiper, my screen is totally black, so I guess it's not a hardware nor connection problem.

dhrone, 2017-03-14:
That's really odd.  I've got a couple of the WEH displays and have not had any issues.  I am driving the WEH displays as if they are the WEG style which is how I get all of the flexibility with fonts, scroll bars, etc.  Try running the test script that is part of the winstar_weg.py display driver.
`/usr/bin/python2 winstar_weg.py`
You do need to be in the pydPiper/displays directory when you run this.  It should provide a simulated music display.  Let me know if the pixels in column 0 all light up during this test.

JedS, 2017-04-28:
It was an hardware problem (LCD was defective).

closed 2017-04-28 by JedS