dhrone/pydPiper #2: no active widget on MoodeAudio

cortegedusage, 2017-03-13:
Hi,
Thank you for your work, It is very easy to setup.
But on MoodeAudio I get "No active Widget" when
starting it with sudo python pydPiper.py --mpd

am I missing something?

Many thanks

dhrone, 2017-03-14:
That message appears when there is no widget that is currently enabled.  This can sometimes happen during start-up when the song data has not been received yet though normally it clears after a few seconds once the data gets updated.  If the "no active widget" message does not clear it is likely either a problem with the data collection system or the pages file has an error.  Can you share your pages file?

cortegedusage, 2017-03-14:
i'm running it with the pages_fixed.py and without any prefix.
both without result

dhrone, 2017-03-19:
Ok.  I'll try a fresh install of Moode tomorrow with the pages_fixed.py to see what is going wrong.

dhrone, 2017-03-26:
Ok.  Finally got around to looking at this.  There is currently an inconsistency between the different music sources in regards to the two variables 'encoding' and 'stream'.  I need to go through each of the services and standardize what each produces.  In the mean time a quick fix for Moode is to change the conditional values in the seqPlay sequence from 'stream' to 'encoding'.    I've added a pages_fixed_moode.py file to the repo that shows the changes.

cortegedusage, 2017-03-28:
Thank you fort tour effort, it works now!

closed 2017-03-28 by cortegedusage