dhrone/pydPiper #3: Timezone not working

ostenax, 2017-07-01:
Hi, very nice project.
I have the problem using timezone. It seems that the problem ist the pages_fixed.py file. There you have this lines of code with a reference to Us timezone:

'time': { 'type':'text', 'format':'{0}', 'variables':['utc|timezone+**_US/Eastern_**|strftime+%-I:%M'], 'font':'large', 'just':'right', 'varwidth':T$

Replaced with this:
'time': { 'type':'text', 'format':'{0}', 'variables':['utc|timezone+Europe/Berlin|strftime+%H:%M'], 'font':'large', 'just':'right', 'varwidth'$

The timezone Germany and 24Hour is used. But this should set over the configfile what doesnt work. My Python knowhow is not good enough to solve the problem.

I couldn't find the reason. Could you verify?

Regards Axel

cortegedusage, 2017-10-09:
It looks like this is hardcodes into pages.py.
@dhrone is this correct?

dhrone, 2017-10-09:
Yes this is correct.  Keep in mind that the dockerized version contains a copy of the entire pydPiper distribution so any changes you make to a local copy will not be referenced.  It is possible to mount the local disk's copy of pydPiper into the docker container.  To do this change docker run command in the init script to the following...

/usr/bin/docker run --network=host --privileged -v /home/volumio/pydPiper:/app/pydPiper:ro dhrone/pydpiper:raspdacv3 python /app/pydPiper.py --volumio

This assumes that you cloned the current version of pydPiper into the /home/volumio directory e.g.
cd /home/volumio
git clone https://github.com/dhrone/pydPiper

This will allow cause the docker container to use the local copy of the files, not the copy stored within the container itself.

Also, one other warning.  I made a change to the attributes of the scroll function.  The only pages file that I modified to use this new format is pages.py.  If you use any of the other pages examples they will likely fail until I get a chance to update them.

cortegedusage, 2017-10-09:
Wow, thanx for the extensive explanation. Will try this t when I find the
time. But it makes sence now why any changes to the docker image didnt
last...

Thank you!

cortegedusage, 2017-10-14:
sorry to say but when I change pydpipervolumio.service in the systemd folder with the above change
nothing happens, pydpiper still starts with the "old" version...
I tried some things, but I'm not familiar with docker at all so I'm kind of stuck...

dhrone, 2017-10-16:
I'll have to test it to see if there is a problem with the command.  I can also walk you through how to make the edits yourself by modifying the docker container.

cortegedusage, 2017-10-16:
I don't know what would be the easiest way for you.
maybe the easiest way would be change the pages.py in github
and change the two lines mentioning the time zone and let the script pick
the local time from the computer it's running.
(might be slightly more difficult than I think it is, that's up to you)
This way anyone could install it and it would just work out of the box. We
would loose the wheather option, that's true (but I'm not using that)

either way, it's a splendid peace of software I must say, and it seems
pretty stable, more stable than other software for displays for volumio.
Thanks!

Dennis

dhrone, 2017-10-21:
I screwed up the command I gave you.  It should have been...
/usr/bin/docker run --network=host --privileged -v /home/volumio/pydPiper:/app:ro dhrone/pydpiper:raspdacv3 python /app/pydPiper.py --volumio

Anyway, I made some significant changes to the argument processing.  You can now specify almost everything that was included in the pydPiper_config.py file through the start-up command.  This makes it a lot easier to localize to your specific configuration.  I've released a new version (v0.3-alpha) and have modified the instructions on the front page to explain how to install it.  

I've just run through a from scratch installation using Volumio 2.296 and it seems to work perfectly (at least for the hardware I have in front of me which is a Raspberry Pi 3 using a HD44780 LCD over an I2C interface).  

Give it a shot and let me know what you think.

dhrone, 2017-10-28:
This appears resolved.  Note that I have now added a new variable to the system called localtime.  This is a <moment> type variable which leverages the TIMEZONE setting.  TIMEZONE can either be set within pydPiper_config.py or through a new command line argument --timezone < tz >.  A list of valid tz values can be found at https://en.wikipedia.org/wiki/List_of_tz_database_time_zones.     Also keep in mind that if you do edit pydPiper_config.py you will also need to include a mount in the docker run command to give the container access to the file.

closed 2017-10-28 by dhrone