dhrone/pydPiper #4: Doesn't work with new volumio2

cortegedusage, 2017-08-03:
I'm trying to get the script to work with volumio2.
but after installing every dependency needed, it keeps crashing on start.
it gives:
volumio@Woonkamer:/usr/local/bin/pydPiper-0.251-alpha$ sudo python2 pydPiper.py --volumio --pages pages_fixed.py
pydPiper starting
Loading font from /usr/local/bin/pydPiper-0.251-alpha/displays/fonts/latin1_5x8_fixed.fnt
Uncaught exception
Traceback (most recent call last):
  File "pydPiper.py", line 534, in <module>
    mc = music_controller(services_list, dc, showupdates)
  File "pydPiper.py", line 93, in __init__
    self.initservices()
  File "pydPiper.py", line 122, in initservices
    musicservice = sources.musicdata_volumio2.musicdata_volumio2(self.musicqueue, pydPiper_config.VOLUMIO_SERVER, pydPiper_config.VOLUMIO_PORT, exitapp )
AttributeError: 'module' object has no attribute 'musicdata_volumio2'
Traceback (most recent call last):
  File "pydPiper.py", line 534, in <module>
    mc = music_controller(services_list, dc, showupdates)
  File "pydPiper.py", line 93, in __init__
    self.initservices()
  File "pydPiper.py", line 122, in initservices
    musicservice = sources.musicdata_volumio2.musicdata_volumio2(self.musicqueue, pydPiper_config.VOLUMIO_SERVER, pydPiper_config.VOLUMIO_PORT, exitapp )
AttributeError: 'module' object has no attribute 'musicdata_volumio2'

when started with mpd it works.

the only thing google gives is that it could be a circular dependency, but I cannot seem to find it.

many thanks
Dennis

dhrone, 2017-08-03:
It looks like when the system tried to instantiate the musicdata_volumio2 it failed.  Can you provide the specific volumio2 image that you are using?

cortegedusage, 2017-08-03:
I was using the latest. 2.246 I believe.

<https://github.com/dhrone/pydPiper/issues/4#issuecomment-320012037>, or mute
the thread

dhrone, 2017-08-03:
Ok. I'll pull that down and see if I run into the same issue.

cortegedusage, 2017-08-03:
That would be great, but there is no rush in it. I Will be leaving Tomorrow
on hollidays and the only pi' s I'l be seeing will be Apple pies...
It would be great though if you could fix it.

I like the Idea of the script pulling info From volumio API instead of mpd
or spop.
Unfortunately Volumio is still heavilly under construction, which brakes
things From time to time.

Thanks for your effort.

Dennis.

dhrone, 2017-08-06:
Just installed 2.2.46 and it has (from a python perspective) a very different configuration from the previous version I tested.  I've been meaning to play with docker in order to pull out of this figure out the dependencies for every distro problem.  Let me see if I can get pydPiper running from within a docker image and then distribute that with docker installation instructions.  Should cut down on some of these installation problems.   I've already confirmed that I can get docker running without complication on Volumio2 v2.246.

cortegedusage, 2017-08-10:
I did notice I had to install an awfull extra packages to get to the point
where it stopped complaining :)

Thanks for the effort.

Dennis

cortegedusage, 2017-08-23:
Just out of curiousity, did you get any further in installing docker/pydpiper?

Dennis

dhrone, 2017-08-23:
I did but still work to be done.  I've gotten docker installed, and am able to configure a docker image with python and the dependent python modules (e.g. Moment, pyLMS, redis).  I still need to get docker configured to allow access to the GPIO pins and the network ports that allow communications with Volumio.

dhrone, 2017-10-02:
Good news.  I'm almost done getting pydPiper Dockerized and have it working on the most recent Volumio 2 version.  Will likely get a release out by the weekend.

cortegedusage, 2017-10-03:
wow great work!

dhrone, 2017-10-07:
Got it working but haven't tested extensively.  I'm assuming that you're using a Raspdac V3.  If not, let me know what you are running and I can modify the instructions.  Also, I should mention that all of the localization settings are only accessible by modifying the docker image.  I'll provide instructions for that later.  For now, the system will run with default settings such as eastern timezone, no weather access, etc.

Quick installation instructions

WARNING:  These instructions are only for the Raspdac V3 running Volumio 2.  If you are using a Raspdac V2 or any other combination of RPi, OLED, LCD, or distribution it will not work.  Also, it is largely untested.  YMMV.

Install the latest version of Volumio 2 (currently 2.282) onto your SD card

Before removing SD card from your computer, place file named 'ssh' in /boot.  This file does not need to contain any content.  It just needs to exist in order to enable the SSH daemon when Volumio gets booted.

Place the SD card into your RPi and boot

Log in to the RPi using SSH
ssh volumio@volumio.local (the default password is volumio)

Issue the following commands to install pydPiper
git clone https://github.com/dhrone/pydPiper
pydPiper/volumio_install.sh
(enter the root password)

A lengthy install process will now run...

When the install script stops executing in about 20 seconds pydPiper should start working.

cvdbout, 2017-10-07:
Is this only working for RaspDac for now?  I've just installed it for using with my LCD (HD44780) and most recent version of Volumio and I'm getting the message shown in the first post in this thread.

dhrone, 2017-10-07:
Let me know your specific config and I'll see what I can do. I need the display type (hd44780, ssd1306) interface style (parallel, i2c), and display dimensions (lines, columns). I also need the address (i2c) or pin connection info (parallel). FYI, I've got i2c support for ssd style displays but not for hd44780 displays. It shouldn't be a big deal to add hd44780 support but I would not be able to test it as I do not have one.

cvdbout, 2017-10-07:
Thank you for the reply!

Ideally, my set-up would be as follows:

HD44780 i2c, 20x4.
i2c address 0x27.

If i2c can't be added let me know and I can let you know the pin configuration (I've two different set-ups, the one with i2c is preferred)

dhrone, 2017-10-07:
Can you also post the error message you got when you ran the ne version?

cvdbout, 2017-10-07:
As follows, word for word as the original post:

pydPiper starting
Loading font from /home/volumio/pydPiper-0.251-alpha/displays/fonts/latin1_5x8_fixed.fnt
Uncaught exception
Traceback (most recent call last):
  File "pydPiper.py", line 534, in <module>
    mc = music_controller(services_list, dc, showupdates)
  File "pydPiper.py", line 93, in __init__
    self.initservices()
  File "pydPiper.py", line 122, in initservices
    musicservice = sources.musicdata_volumio2.musicdata_volumio2(self.musicqueue, pydPiper_config.VOLUMIO_SERVER, pydPiper_config.VOLUMIO_PORT, exitapp )
AttributeError: 'module' object has no attribute 'musicdata_volumio2'
Traceback (most recent call last):
  File "pydPiper.py", line 534, in <module>
    mc = music_controller(services_list, dc, showupdates)
  File "pydPiper.py", line 93, in __init__
    self.initservices()
  File "pydPiper.py", line 122, in initservices
    musicservice = sources.musicdata_volumio2.musicdata_volumio2(self.musicqueue, pydPiper_config.VOLUMIO_SERVER, pydPiper_config.VOLUMIO_PORT, exitapp )
AttributeError: 'module' object has no attribute 'musicdata_volumio2'

cortegedusage, 2017-10-07:
I got it working, with the install script.
Need to change the Timezone and i'm office thanks

dhrone, 2017-10-08:
Could you confirm that you got it to work and if yes, what you did to get it working?  The version installed from the volumio_install.sh script should not work with an HD44780.  FYI, from the output you posted, it appears you are running the last pydPiper release version (0.251-alpha), not the development version I am actively working on.

cvdbout, 2017-10-08:
dhrone, different person, different setup - original poster got it working, so should I start a new post/move over to the other question I posted to allow them to close this?

cortegedusage, 2017-10-08:
works like a charm!

p1234321, 2017-10-11:
Hello,

I also tried with the instructions from above. but when i want to start it with "sudo python pydPiper.py --volumio --pages pages_test_ssd1306.py" i get:

Traceback (most recent call last):
  File "pydPiper.py", line 8, in <module>
    import json, threading, logging, Queue, time, sys, getopt, moment, signal, commands, os, copy, imp, urllib2
ImportError: No module named moment

please help...

best regards

paul

dhrone, 2017-10-11:
It's not clear to me which directions that you followed.  The command that you've issued (sudo python pydPiper.py --volumio --pages pages_test_ssd1306.py) does not run pydPiper within the docker container that I've set up.  It runs it on the base volumio distribution which does not contain the required python dependencies.  That's the reason that the exception was thrown.  It also looks like you are trying to get an SSD style display working.  That is not yet supported with the dockerized version of pydPython but I may get it working this weekend.  Please let me know what your specific situation is (version of Volumio, type of display, dimensions of display, I2C address, etc)  I may be able to test your specific configuration.

p1234321, 2017-10-11:
Thanks a lot for the fast response! Yes it is true that I want to use it with an ssd1306 display. 
Looking forward to your release! Amazing project!!!

thanks

Paul
_______
paul pritz
hugo-wolf-gasse 10/81
8010 Graz
tel: 06503332443

dhrone, 2017-10-11:
Would still be great to get confirmation on which volumio distribution you are using, and the dimensions of your display in pixels.

p1234321, 2017-10-11:
I’m using volume 2.285 and an sad 1306 with 128x64 pixels.

Best regards

paul
_______
paul pritz
hugo-wolf-gasse 10/81
8010 Graz
tel: 06503332443

dhrone, 2017-10-21:
I just released a new version of the software that should work with Volumio and the SSD1306(128x64).  Unfortunately, I do not have my SSD1306 display here so will have to test next week.  Feel free to give it a try yourself though.  Instructions on the home page.  The ExecStart command for your system should be something similar to...

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver ssd1306_i2c  --width 128 --height 64 --i2caddress 0x3D --timezone 'US/Eastern' --temperature fahrenheit  --pages pages_ssd1306.py

p1234321, 2017-10-24:
Thanks you so much!!!
I’m going to try it this weekend!

Best regards

paul

p1234321, 2017-10-26:
Thank you so much!
Works like a charm!
One question left:
Is it possible to change the settings after installation. I would like to take an other display with different i2c dress….

Best regards
Paul

wimpie007, 2017-10-27:
@dhrone seems to be a nice project!
i'm using a SH1106 oled, the 'big' brother of the 1306
The difference is minimal in hardware terms:
https://forum.arduino.cc/index.php?topic=256374.0

Any chance to get the sh1106 added?
regards,

dhrone, 2017-10-28:
@p1234321 you can pass the i2c address in as a command line parameter.  Documentation and examples are included inside the pydpiper.service file.  You could also modify the pydPiper_config.py file.  If you go that route, make sure to add the appropriate mount from the host file system into the docker container.  This is done by adding -v <host dir where pydPiper is installed>:/app:ro to the ExecStart command.  

example
ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  **-v /home/volumio/pydPiper:/app:ro** dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780_i2c  --width 100 --height 32 --i2caddress 0x27 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_20x4.py

Also, remember that when you make a change to the pydpiper.service file located in /etc/systemd/system that you also need to execute

sudo systemctl daemon-reload

dhrone, 2017-10-28:
@wimpie007 I am using Luma.OLED to support the SSD1306 which also supports the SH1106.  So, I suspect that getting it to work would be straight forward.  Unfortunately, I do not have any SH1106 displays to test with.  

As the original reason for this issue now appears resolved, I'm going to close it out.  I'll open a new issue though for your request.

closed 2017-10-28 by dhrone

spinko101, 2018-03-04:
Hi, and thank you for this package :-)

I'm using a Raspberry Pi3 with a Adafruit SSD1306 128x64 display and Volumio. It is wired as i2c . All seems to be working but the screen is garbled (white dots all over it). I have checked all the parameters and they look ok too. This is what i have in /etc/systemd/system/pydpiper.service

`ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio --driver ssd1306_i2c  --width 128 --height 64 --rs 7 --e 8 --d4 25 --d5 24 --d6 23 --d7 27 --timezone 'paris/france' --temperature celsius --pages pages_ssd1306.py --i2caddress 0x3D
`
any help or pointers would be greatly appreciated ! Thanks

drmabuse71, 2018-03-07:
Hi,
my OLED 1306 display doesn´t work @all :-(
I´m using RPi Model B (not B+ !) with the latest volumio ver. 2.368.  
i2cdetect -y 1 shows:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
Service file contains:
ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver ssd1306_i2c --width 128 --height 64 --i2caddress 0x3c --i2cport 1 --timezone 'Dortmund/Germany'  --temperature celsius --pages pages_ssd1306.py
Any suggestions what happened ? Or something I can check ?

Kind regards, Michael

drmabuse71, 2018-03-08:
Sorry guys,
a little correction: the Display shows up something  like "no Metadata" for a very short time , then displays No data H0 L0 for about 10 seconds and then turns off. Seems the mpd-database can´t be reached ?
Kind regards