dhrone/pydPiper #5: [Enhancement Request] Add support for Spotify Connect & Airplay Metadata

michaellunzer, 2017-09-05:
Hi @dhrone, good job on this project! 

I just discovered it last night while looking for something to drive a pi with a 4x20 LCD running MPD with [Iris ](https://github.com/jaedb/Iris) as the front-end and [Spotify Connect](https://github.com/dtcooper/raspotify) and Airplay.

I'm not sure if pydPiper can pull the track metadata from Volumio using the [Spotify Connect Plugin](https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect2) and Airplay Plugins and it would be under the Volumio "umbrella"

If that is the case, and Spotify Connect and Airplay display on the LCD because they are connected to Volumio, please disregard the text below....

## Airplay
I see that there is a [quick tutorial](http://acquisitionsyndrome.com/2014/08/using-a-usb-lcd-display-with-your-music-player/) which shows you how to get the Airplay metadata. Once we have that, I think it would be fairly straightforward to integrate into your existing project. Is this something you'd be interested in collaborating on?

## Spotify Connect
I've found an interesting section of the Spotify Connect Volumio Plugin which explains how they extract the track title [here](https://github.com/balbuze/volumio-plugins/blob/master/plugins/music_service/volspotconnect2/cutinfo)

Here is a relevant thread which discusses a few possible methods to extract [data](https://github.com/plietar/librespot/issues/154)

Also it looks like there has been some updates to the code since the discussion ended: https://github.com/plietar/librespot/blob/master/metadata/src/lib.rs

Sorry to overwhelm you with a bunch of links and random information. But, let me know if you're interested in collaborating and if you have the time to build out your product even more!

Best,
Michael

dhrone, 2017-09-07:
I have some good news and a bit of bad news.  The good news is that pydPiper supports a 4x20 display very well and can pull metadata directly from Volumio 2.  This works fine for Spotify content though I'm not sure if it works for Airplay.  If when you play Airplay content, it shows up in the Volumio web interface, it should be supported just fine with pydPiper.  

The bad news is that the current release of Volumio made some significant changes to the Python installation and many of the dependencies that pydPiper requires are missing.  I've gotten tired of chasing this particular problem so I'm working on placing pydPiper within a docker container.  Unfortunately, I have not had much time to work on this for the last few weeks but it is on my near term to-do list.

dhrone, 2017-10-21:
Latest version just released (v0.3-alpha).  It should work fine for Volumio/Spotify.  Airplay also works but Volumio doesn't seem to get the playlist and song position data so those screen elements may not be accurate.  Artist, album, and song title seem to be ok.  Check it out and see if it works for you.

dhrone, 2017-10-28:
I'm going to assume that this is resolved for you.  Please re-open if that is not the case.  FYI, there have been some bug fixes since v0.3-alpha was released.  I'm making a few more changes over the weekend so will likely issue a v0.31-alpha but in the mean time you may want to clone from the main branch instead of pulling down the release version for the moment.

closed 2017-10-28 by dhrone