dhrone/pydPiper #6: [Request] HD44780 i2c support

cvdbout, 2017-10-07:
Is there any way support can be implemented for HD44780 w/ i2c chip PCF8574?

cvdbout, 2017-10-08:
Moved from other thread:

Ideally, my set-up would be as follows:

HD44780 i2c, 20x4.
i2c address 0x27.

If i2c can't be added let me know and I can let you know the pin configuration (I've two different set-ups, the one with i2c is preferred).

dhrone, 2017-10-08:
The current version of the software does not support I2C on HD44780 displays but there really isn't any reason it can't.  The hard part of the code is the HD44780 support which is already written.  Changing the communications protocol is straight forward.  Just have to get a display to test with.  Probably not going to happen this week but might have some time for it next weekend.

cvdbout, 2017-10-09:
Thank you,

I've tried to get it to work in parallel tonight, but I keep getting this error:

volumio@volumio:/usr/local/bin/pydPiper-0.251-alpha$ sudo python pydPiper.py --volumio --pages pages_lcd_20x4.py
pydPiper starting
Loading font from /usr/local/bin/pydPiper-0.251-alpha/displays/fonts/latin1_5x8_fixed.fnt
Uncaught exception
Traceback (most recent call last):
  File "pydPiper.py", line 534, in <module>
    mc = music_controller(services_list, dc, showupdates)
  File "pydPiper.py", line 93, in __init__
    self.initservices()
  File "pydPiper.py", line 122, in initservices
    musicservice = sources.musicdata_volumio2.musicdata_volumio2(self.musicqueue, pydPiper_config.VOLUMIO_SERVER, pydPiper_config.VOLUMIO_PORT, exitapp )
AttributeError: 'module' object has no attribute 'musicdata_volumio2'
Traceback (most recent call last):
  File "pydPiper.py", line 534, in <module>
    mc = music_controller(services_list, dc, showupdates)
  File "pydPiper.py", line 93, in __init__
    self.initservices()
  File "pydPiper.py", line 122, in initservices
    musicservice = sources.musicdata_volumio2.musicdata_volumio2(self.musicqueue, pydPiper_config.VOLUMIO_SERVER, pydPiper_config.VOLUMIO_PORT, exitapp )
AttributeError: 'module' object has no attribute 'musicdata_volumio2'

If you could have a look, that would be great but there really is no rush.

dhrone, 2017-10-09:
Volumio substantially changed it's configuration a couple of releases ago, breaking several dependencies within pydPiper.  That's the reason I'm switching to a dockerized version of the program.  From the error you've posted, it looks like you are running the last released version of pydPiper.  To get it running on the current volumio distro, you need to pull the current development version (e.g. git clone https://github.com/dhrone/pydPiper).  Unfortunately, this is still under development.  This next weekend, I may have the time to pull together instructions for getting it to work with your specific configuration.

cvdbout, 2017-10-10:
Thanks, that'd be great if you could

dhrone, 2017-10-16:
I got the HD44780 I2C driver written over the weekend.  Need to test and then write up some instructions.

cvdbout, 2017-10-16:
That's brilliant, thank you!  Can't wait to try it out!

dhrone, 2017-10-21:
I finished the HD44780 I2C code and made some additional changes that should make the whole package easier to install.  The latest version is now v0.3-alpha and the instructions are on the home page.  Give it a shot and let me know if it works for you.  FYI, not heavily tested but I did perform a from scratch installation on Volumio 2.296.  Make sure you review the pydpiper.service file as it WILL NEED TO BE EDITED in order for the installer to work.

cvdbout, 2017-10-21:
Thank you so much!

I've installed it as per instruction, I'm getting the clock but nothing else (I also need to change the time zone, not sure if there is an 'official' source for time zones and what to put here?  I've tried a couple already to no avail, but will keep trying).

I'm just updating to 2.296 to see if this changes anything, if not I'll do a clean install.

cvdbout, 2017-10-21:
Clean install, and in the ExecStart the i2cport section was missing, so I've added that.

All working now, thanks again!  Its looking great!

closed 2017-10-21 by cvdbout