dhrone/pydPiper #7: Pages Modifications

cvdbout, 2017-10-22:
I think I've got my head round what each part of the pages file is doing, and I'm now attempting to edit.

This is my display right now (hd44780, i2c), however whenever I make changes to pages_lcd_20x4.py it isn't changing anything on the display.  Right now I'm trying to remove the TRK section at the bottom of the display from the canvases where its references, and the file shows that I've saved those changes, and this is the file in ExecStart.  Rebooting doesn't make a difference.

I assumed that as that's the file in execstart that it would be the one to edit?

[image]

dhrone, 2017-10-22:
Yes you are editing the right file.  The problem is that the software is running within a Docker container so it cannot see the version of the file that you have edited.  Your ExecStart probably looks something like...

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780_i2c  --width 100 --height 32 --i2caddress 0x27 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_20x4.py.

Try adding another -v command to it to allow the container to see the pydPiper directory.

ExecStart=/usr/bin/docker run --network=host --privileged **-v /home/volumio/pydPiper-v0.3-alpha:/app:ro** -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780_i2c  --width 100 --height 32 --i2caddress 0x27 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_20x4.py.

The format for -v is <dir_on_host>:<dir_in_container>:<permissions>.  So if you have placed pydPiper in a different directory, adjust <dir_on_host> to that directory.

I'd appreciate receiving any pages files you come up with.  I'm always looking to improve them.  I do have a specific question for you.  Right now, overly long lines scroll but the slow refresh of the LCDs makes this look less than perfect.  Do you prefer the scrolling, or would you prefer that scrolling be shut off?

cvdbout, 2017-10-22:
That's great - I did think that would be the case so attempted to change
the line myself but managed to make the whole thing unbootable instead.

I'll make the changes you suggested as soon as I do a clean install.

As soon as I get a page I'm happy with I'll share it here.

I like the scrolling - with very long song  titles it makes it easier to
read, but the wait time on the screen sometimes cuts off before the scroll
has finished, something I'm going to change. If possible though, it might
be easier to give people the option.  Even if that is possible though, I
understand that would mean writing an instruction file for how to make
changes and then allow docker to see the local file.

I've got a few ideas that I'd like to try and implement, if I can get them
to work I'll let you know.

The format for -v is <dir_on_host>:<dir_in_container>:. So if you have

<https://github.com/dhrone/pydPiper/issues/7#issuecomment-338471637>, or mute
the thread

cvdbout, 2017-10-22:
I've now made those changes, and got it to display a modified canvas once.  I made a further change to a canvas, saved the file, restarted docker and it's not loading the modified version now - changes still show in pydpiper.service and the pages file but are not being shown on the display.

I've commented out all sequences but one now, but its still scrolling through them all.

dhrone, 2017-10-22:
I think both of your problems (pages file and timezone) are related to the same issue.  I suspect you have not reloaded your systemd config.  Make sure that the copy of pydpiper.service located in /etc/systemd/system has the right settings.  Then you will need to force systemd to reload its configuration by issuing...

sudo systemctl daemon-reload

See...
[
https://serverfault.com/questions/700862/do-systemd-unit-files-have-to-be-reloaded-when-modified](url) 

Europe/London should be the correct setting.  A complete list of valid values is at...
[https://en.wikipedia.org/wiki/List_of_tz_database_time_zones](url)

cvdbout, 2017-10-22:
I've now got it showing the correct time zone, and this is the execstart in pydpiper.service

ExecStart=/usr/bin/docker run --network=host --privileged -v /home/volumio/**pydPiper-0.3-alpha**:/app:ro -v /var/log:/var/log:rw dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780_i2c --width 100 --height 32 --i2caddress 0x27 --timezone 'Europe/London' --temperature celcius --pages pages_lcd_20x4.py.

The part in bold may be of help to some people, as the default install location doesn't have v0.3, just 0.3.  Scratched my head over that one for a while!

dhrone, 2017-10-22:
Ahh.  Glad you caught the error.  Let me know how your page design goes.  I'll leave this issue open for the moment.

cvdbout, 2017-10-22:
Two last (hopefully anyway!) questions - within volumio there are 3 states - play, stop and pause.  Is that reflected anywhere on here?  I can only find play and stop, whereas if possible I'd like pause too (but no big deal if not, I've only had a quick glance through pydpiper.py, might be I'm not seeing it or I need to add that myself)

Also, the duration that the canvases are held for, what are those numbers?  I want to hold something for longer than just a few seconds but can't seem to find the magic number!

dhrone, 2017-10-23:
Currently the only states that pydPiper tracks are playing and stopped.  Paused == stopped.  Way back when I started this project, the distribution I was working with did not provide a paused status so it wasn't possible for me to distinguish between stopped and paused.   I will check to see if Volumio2 is exposing a paused status.  If yes, I'll see about adding it.

You may want to review the documentation for the pages file format in the docs directory.  Let me try to give a quick description here though.

Pages are displayed as part of a sequence of pages, sequence being composed of a set of canvases.  When a sequence is active, it starts displaying its canvases in order keeping each canvas on screen for <duration> seconds.  Each sequence has a Boolean value called a conditional that determines when the sequence should be active.   

Likewise, each canvas also has a conditional that determines whether that canvas should be displayed when its turn in the rotation comes up.  In the included pages_lcd_20x4.py file, there is a sequence called seqPlay that's conditional is db['state'] == 'play'.  So, this sequence becomes active when Volumio starts playing.  There are four canvases that are part of seqPlay.  The first three, playartist, playalbum, playtitle have conditionals that make them active when not playing from a webradio source.  The last, play_radio is active when Volumio is playing from a webradio source.  When a canvas is active and it's turn comes up, it is displayed for duration seconds.  So, if for instance you wanted to double the amount of time that the title is displayed, you could change duration for the playtitle canvas from 8 to 16.  

Hope this helps!

cvdbout, 2017-10-23:
That definitely helps, thank you!

I thought the number was in seconds, but it wasn't working correctly when I changed it.  I'll try again later.

I've got the canvases setup that I like, just looking into sequences now and what might work best for my needs.

The reason I asked about pause was I'd really like a pause sequence, that would display the pause canvas for x seconds then display the clock. Sort of like a timeout feature.  I've used the socketIO.emit feature with volumio2 which can toggle the states between play and pause - I can't remember if stop comes into it unless radio is playing though.

cvdbout, 2017-10-25:
Thanks for all your help, this is my display now. Very happy with it!

Play media screen, which holds for 30 seconds then volume, random (if enabled) and repeat (if enabled) screens scroll through 1 second each.
[image]
[image]
[image]
[image]

Starting splash screen
[image]

Clock with no weather data
[image]

Clock with weather data
[image]

Radio screen showing elapsed time and clock, dislays for 30 seconds with volume screen showing for 1 second.
[image]
[image]

It might be worth noting that Volumio2 doesn't have repeat single as an option, not sure if there are plans to reintroduce it.

I was also having issues with the random and repeat screens showing randomly on boot and whenever play was switched between radio and MPD.  I resolved it by adding an extra conditional to the seqAnnounceRandom and seqAnnounceRepeat so they look like this now:

{
		'name':'seqAnnounceRandom',
		'canvases': [ { 'name':'showrandom', 'duration':2 } ],
		'conditional': "db['random'] != dbp['random'] and not db['actPlayer']=='webradio'",
		'minimum':2,
},
{
		'name':'seqAnnounceRepeat',
		'canvases': [ { 'name':'showrepeatall', 'duration':2 } ],
		'conditional': "db['repeat'] != dbp['repeat'] and not db['actPlayer']=='webradio'",
		'minimum':2,
},

dhrone, 2017-10-28:
Yes, this is caused by spurious changes of the random and repeat variables from Volumio when changing playing state.

Great job on your screens.  Should we close this issue out?

closed 2017-10-28 by cvdbout