dhrone/pydPiper #8: Time only updates every 5 minutes?

cvdbout, 2017-10-24:
Is it a problem with my setup, or by design?

if it's by design, is it possible to change this so it keeps actual time?

dhrone, 2017-10-25:
Looks like a bug.  I've replicated it on my system.  Will need to chase it down.  Stay tuned...

dhrone, 2017-10-26:
Found the issue.  Don't know how I didn't notice before.  Time is one of the variables that is updated along with the rest of the systems variables.  As I don't want to hammer the system (or the weather service), I limit the call to once every five minutes.  To fix, I just need to move time updates out of the system variables update routine.

cvdbout, 2017-10-26:
Glad you found it! Looking forward to accurate timekeeping ;)

dhrone, 2017-10-27:
Ok.  It should be fixed in the main branch.  Give it a shot and let me know.  FYI, if you installed from the v0.3-alpha release, you'll want to clone from the main branch.

cvdbout, 2017-10-27:
Working perfectly now!  Thank you!

closed 2017-10-27 by cvdbout

gizegoiz, 2018-01-24:
First of all thank you for this software. You've done really good job.
However I have still the issue with time updating every 5 minutes. I have followed installation instructions from README.md. I use 24x2 hd44780 with weather informations. Everything is ok, except this time updating.
I have tried to do fresh installation:
sudo git clone https://github.com/dhrone/pydPiper.git
cd pydPiper
sudo ./install_docker.sh
but it doesn't work at all.