dhrone/pydPiper #9: Please add support for SH1106

dhrone, 2017-10-28:
Per @wimpie007 there is a need to add driver support for the SH1106 display.

wimpie007, 2017-10-28:
so how can i run it with sh1106 driver?
replace SSD1306  with SH1106? and pages_ssd1306.py with...?

dhrone, 2017-10-28:
This is completely untested, but I have just added a new display driver that allows all of the luma display types to be supported.  To use it you need to set display to 'luma_i2c' and devicetype to 'sh1106'.  Here's an example ExecStart...

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:ro dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver luma_i2c  --devicetype sh1106 --width 128 --height 64 --i2caddress 0x3D --timezone 'US/Eastern' --temperature fahrenheit --pages pages_ssd1306.py

This assumes you are using the volumio 2 distribution and that you have cloned pydPiper into /home/volumio.  E.g.

cd /home/volumio
sudo git clone https://github.com/dhrone/pydPiper

wimpie007, 2017-10-28:
will try tomorrow (have to leave now), and report back! thanks in advance!

wimpie007, 2017-10-29:
@dhrone It's working!!! great work man! Thanks a lot!
used your exec line et voila...
You should consider making a plugin package for volumio, i guess it would be used a lot!

closed 2017-10-30 by dhrone