dhrone/pydPiper #12: High CPU usage on RPI1B

wimpie007, 2017-10-30:
Hi, pydPiper takes 50-60% CPU of my RPI 1B.
Is there a way to lower this by for example lower the update speed? (and increasing scroll pixels per 'loop')
Not really a 'issue', but can't find a way to ask...

dhrone, 2017-10-30:
Yes, I'd imagine that a RPi 1 would have high CPU utilization.  You can try increasing the ANIMATION_SMOOTHING setting in pydPiper_config.py.

dhrone, 2017-12-02:
Did changing the ANIMATION_SMOOTHING setting have any impact on your CPU utilization?  

Another factor can be in play for CPU utilization.  Displays that use a parallel interface take much more GPIO updates to control them.  Displays with I2C interfaces are more efficient.  You may get better results with those style displays.  FYI, there are piggyback boards [https://www.amazon.com/Adafruit-i2c-SPI-character-backpack/dp/B00OKCON84/ref=sr_1_1?s=electronics&ie=UTF8&qid=1512235221&sr=1-1&keywords=i2c+backpack](url) that you can add to an LCD display.

wimpie007, 2017-12-18:
Hi, i did change the ANIMATION_SMOOTHING setting, but it didn't change much.
My display is a oled display with I2C interface already. Maybe the docker container is a bit of a overhead. (almost too much for the ole RPI1)