dhrone/pydPiper #13: Scroll speed

nicolaipeter, 2017-10-31:
Hi, 
First, thank you for this nice piece of software.
I installed it on a RPI2 with volumio and a 20x4 parallel display. 
It works with a little change in  pydpiper.service

# HD44780 LCD 20x4 display with I2C interface on address 0x27
ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780  --width 100 --height 32 --rs 7 --e 8 --d4 25 --d5 24 --d6 23 --d7 27 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_20x4.py

a re-install was required to make these changes work. 

I tried to change the scroll speed because it was much to fast. I found the scroll function in the pages_lcd_20x4.py

i changed the following line in pages_lcd_20x4.py
 'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',5,1,20,'onloop',3,100) },

to

 'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',5,8,20,'onloop',3,100) },

also to 
 'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',5,88,20,'onloop',3,100) },

but did not noticed any change in scroll speed when playing webradio or file from USB.
What am i doing wrong?

Please help me, 

Kind regards,
Peter

dhrone, 2017-10-31:
You've made the right type of change to the page file to cause it to slow down the scroll.  The problem is that the file you've edited was not referenced by the pydPiper.py program.  

In this latest version of the software, I've moved it into a docker container. This was to simplify management of all of the dependent software that pydPiper relies upon.  When you create a Docker container, all of the software (and dependencies) that you need to run the container are stored within it.  So, when you are running software within the container, unless you explicitly ask that files outside the container be available to the container. they will not be.  

Try modifying the ExecStart to mount the directory that you have the pydPiper software installed in. 
  It should look like the following:

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw **-v /home/volumio/pydPiper:/app:rw** dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio --driver hd44780 --width 100 --height 32 --rs 7 --e 8 --d4 25 --d5 24 --d6 23 --d7 27 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_20x4.py

This example assumes that you have cloned pydPiper into your /home/volumio directory.

nicolaipeter, 2017-10-31:
Hi dhrone,
Thank you for your answer. I am verry new to linux and learing every minute...

In the docs folder there is a file describing the volumio install. But it requires volumio_install.sh. This script is not available..

So i  installed it according to the description on the main page. 
This resulted in this directory contents: 

volumio@volumio:~$ ls
pydPiper-0.3-alpha  v0.3-alpha.tar.gz
volumio@volumio:~$

So i changed your ExecStart with this path..

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper-0.3-alpha:/app:rw dhrone/pydpiper:v0.3-alpha python /app/pydPiper.py --volumio --driver hd44780 --width 100 --height 32 --rs 7 --e 8 --d4 25 --d5 24 --d6 23 --d7 27 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_20x4.py

after this i did a 
sudo ./install.sh

But now the display does not init..
I just don't understand exacly what i am doing. But please help me.

Kind regards,
Peter

dhrone, 2017-11-01:
What is the full path to the pydpiper.service file that you edited?  The one that you need to change is located at /etc/systemd/system.  After editing, you need to cause the systemd daemon to reload its config file.

sudo systemctl daemon-reload
sudo systemctl restart pydpiper

nicolaipeter, 2017-11-01:
Thank you for your help!

This is the full path of the files i was editing

volumio@volumio:~$ ls
pydPiper-0.3-alpha  v0.3-alpha.tar.gz
volumio@volumio:~$ cd pydPiper-0.3-alpha
volumio@volumio:~/pydPiper-0.3-alpha$
volumio@volumio:~/pydPiper-0.3-alpha$ ls
Dockerfile  install.sh            pages_lcd_16x2.py       pages_weh_80x16.py  pydpiper.service.save
LICENSE     pages.py              pages_lcd_20x4.py       pydPiper.py         pydpiper.service.save.1
README.md   pages_fixed.py        pages_ssd1306.py        pydPiper_IR.py      requirements.txt
displays    pages_fixed_jcu.py    pages_ssd1306_nottf.py  pydPiper_config.py  sources
docs        pages_fixed_moode.py  pages_weg_100x16.py     pydpiper.service    utils
volumio@volumio:~/pydPiper-0.3-alpha$
volumio@volumio:~/pydPiper-0.3-alpha$ pwd
/home/volumio/pydPiper-0.3-alpha

Here i edited the pydpiper.service and the pages_lcd_20x4.py
but this did not work

Now i changed the pydpiper.service at /etc/systemd/system in to this:
ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper-0.3-alpha:/app:rw dhrone/pydpiper:v0.3-alpha python /app/pydPiper.py --volumio --driver hd44780 --width 100 --height 32 --rs 7 --e 8 --d4 25 --d5 24 --d6 23 --d7 27 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_20x4.py

please note that i am using version 0.3 (according to the folder name) 
So i changed 0.31 to 0.3 in the ExecStart line

After this change and deamon reload and restart of pydpiper the display did stay blank. (init was already done before i think)

After selecting the original ExecStart line in the service it did work again with the fast scroll. (after deamon reload an restart)

I think i am editting the pages_lcd_20x4.py on the wrong location or that the Execstart is not finding the correct file...

Any suggestion?

nicolaipeter, 2017-11-01:
Hi, 
It is working..

I changed:    dhrone/pydpiper:v0.3-alpha back to dhrone/pydpiper:v0.31-alpha  

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper-0.3-alpha:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio --driver hd44780 --width 100 --height 32 --rs 7 --e 8 --d4 25 --d5 24 --d6 23 --d7 27 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_20x4.py

The scroll speed is changed in: pages_lcd_20x4.py in this path:
/home/volumio/pydPiper-0.3-alpha

Can you explain why this must be version 0.31? Where can i find this file or see what version this should be?
It is probably installed when i used the install.sh script yesterday

Kind regards,
Peter

dhrone, 2017-11-02:
dhrone/pydpiper:v0.31-alpha is the most recent Docker container that I have posted to Docker's repository.  It is stored at https://hub.docker.com/r/dhrone/pydpiper/ but is retrieved automatically by the docker run command when it is included as the container to execute.   This container file includes a copy of the pydPiper software that is current as of this weekend.  I should have also issued a new release of the software on github simultaneously but ran out of time.

giannihsd, 2018-12-09:
Hi,

I have installed plugin in volumio  2.513 amenità work perfect.
I have 16x2 lcd display connectd. I want decrease the scroll time when play webradio , so I have modified page_lcd_16x2.py and set time parameter to’ 100 to all types of data because webradio sometimes send only title data. 
After I execute agami ./install, and when finish I reboot volumio.
Nothing chance, scroll si always fast. What I can do?