dhrone/pydPiper #14: Container vs actual

cvdbout, 2017-11-02:
I'm not sure if this is an issue with my set-up, or if its a bug -

I've been editing my files in a container, and they all display fine.
When I exit the container, everything displays fine, however after I reboot the pi one screen is refusing to display correctly.
To get this screen working it appears that I can go back into the container, run pydpiper from there to check it displays fine, exit the container, start system pydpiper and the screen displays as I'd expect again.

I've pulled from master and v0.3-alpha (the latter I am then pulling the musicdata volumio2 file from master for the pause status).

I am editing pydpiper.service prior to installation with this additional -v so it can see my modified files (when pulling from master, the folder is simply pydPiper)

-v /home/volumio/pydPiper-0.3-alpha:/app:ro 

My pages file has this (in addition to other things)widget:

'pausesymbol':{'type':'text', 'format':'|| PAUSED','font':'small','size':(45,8)},

Canvas:
'playpause':{'widgets':  [ ('artist',0,0), ('album',0,8), ('title',0,16),('pausesymbol', 0,24)], 'size':(100,32) },

This is my sequence:

	{
		'name': 'seqPause',
		'canvases': [
			{ 'name':'playpause', 'duration':180, 'conditional':"not db['actPlayer']=='webradio'"},
			{ 'name':'stoptime', 'duration':3000, 'conditional':"db['outside_conditions']=='No data'" },
			{ 'name':'stoptimeweather', 'duration':3000, 'conditional':"not db['outside_conditions']=='No data'" },
		],
		'conditional': "db['state']=='pause'"
	},

I am replacing the default pages file with this one, editing the permissions to 777 before doing so.  Everything except the pause sequence works on boot, the pause sequence will only work after I've gone into the container and exited again.

On boot, I only get the pause widget on the display, with no other information.  After exiting container, the whole canvas as defined is displaying and works as I'd expect it to until the next reboot of the pi.

I've done 4 clean installs now, all with the same result.  Am I missing something really glaringly obvious?

dhrone, 2017-11-02:
Where are you editing the file?  Within the docker container or within the host OS?  If the former, everytime you restart the container, the files are going to revert to their prior status.  You can either edit the files outside the container, or change your mount to read-write.  

-v /home/volumio/pydPiper-0.3-alpha:/app:rw

cvdbout, 2017-11-02:
Steps so far:

Clean install Volumio 2.296
sudo git clone https://github.com/dhrone/pydPiper.git
edit pydpiper.service, add to the execstart entry -v /home/volumio/pydPiper:/app:rw
enable SSH, chmod 777 pages file, upload full edited pages file.
install pydPiper

Display is working perfectly, except for the pause screen which displays only the pause widget, not the canvas..

sudo systemctl stop pydpiper
sudo docker run --network=host --privileged -ti -v /var/log:/var/log:rw -v `pwd`:/app:rw dhrone/pydpiper:v0.31-alpha /bin/bash
sudo python /app/pydPiper.py --volumio --driver hd44780_i2c --width 100 --height 32 --i2caddress 0x27 --timezone 'Europe/London' --temperature celsius --pages pages_lcd_20x4.py

Pause screen works perfectly - I've made no changes to any files other than prior to installing.

Exit container
sudo systemctl start pydpiper

Pause screen still works, shows full pause canvas.

Reboot the pi, and only the pause widget is displaying again.

Having written all that, it occurred to me to just stop and start pydPiper without going into the container to see what happens, and the pause canvas displays after restarting pydPiper until the pi is rebooted again wherein the pause widget only displays.

dhrone, 2017-11-03:
Given your last paragraph, I suspect the issue is the conditional evaluations, the state of the players variables on initial start-up and the length you've specified to display each of the pages during the seqPause sequence.  I'd focus on the specification of your conditionals and the length of the canvas displays.  FYI, you can get the system to show you every variable update by adding --showupdates to the command.

cvdbout, 2017-11-03:
Great, I'll try that and see what happens! Thank you

closed 2017-11-03 by cvdbout