dhrone/pydPiper #15: Conditionals

cvdbout, 2017-11-03:
Is there any way to 'reset' the conditional statement of the sequences?

Right now, it seems to be holding the time information e.g.

A sequence displays one canvas for 10 seconds, then another for 50 - if it's interrupted during that second canvas, when the sequence is called again it will resume where it left off before cycling back through the canvases again.

Is there a way to force it to always restart the sequence when it's being called? So always restart at the 10 second canvas, then moving to the 50 second canvas every time?

dhrone, 2017-11-03:
It should be possible.  I'll have to look at the logic that controls what canvas is current.  It contains a flag to determine whether is should reset the display timing but its been so long since I wrote that code I'm going to have to read through it to re-figure out how it works.

dhrone, 2017-12-02:
Re-reading your message, I'm not 100% what behavior you are looking for.  Under what conditions would you want the timer to be reset, and when would you want it to be stable?  For instance, if interrupted by a volume update, I'm assuming you'd want the timer to be held.  What scenarios would you want it to be reset?

cvdbout, 2017-12-02:
I was thinking mainly along the lines of a pause screen. My initial attempt
at it was the hold a pause screen for x seconds before it would timeout and
then show the clock - problem with this being that eventually it won't show
the pause screen and only the clock screen until it's completed the clock
screen time period.

My second attempt which works OK is to have a pause screen with cycles
through pause and the clock every x number of seconds. This however means
that it will never just display the clock after music has been paused, and
because of the way volumio works if it's playing from a local or network
drive it will only pause not stop using a Web or app interface.

It's works OK, but still doesn't feel right. If the unit is turned on all
the time, but music is paused it will always be apparent every x seconds
that music is paused, however I don't want to only show the clock screen as
again, if the music is paused the user may want to see the info.

In my head, I'd want to show a pause screen for 5 minutes, then the clock.
When music is unpaused, then paused again it would start again with the
pause screen for 5 minutes.

Stevemoode, 2020-09-14:
Hi dhrone, first of all thanks for this brilliant piece of code. 

I'm totally new to coding etc. and have hit what I think is the same issue so did you get chance to look at this? 

In my case I want to display the full screen clock for 10mins when I stop or pause Moode, then clear the screen completely until I begin a new action (play etc), this is to protct against oled screen burn. I have managed to get this working by creating a new widget with a blank screen output and having it in the sequence after the stop/pause which activates the clock, but it obviously only works until the initial time hasn't passed the 10 mins. After that each time I stop/pause the screen goes blank which I believe is correct after looking through the code.  Having the timer 'reset' each time it is accessed would solve this issue so is there anything I can do to make this happen?

Once again thanks for your work and sorry if my explination is not very clear, this is my first steps into code.

dhrone, 2020-09-14:
Could you share your pages file?  You can attach it here but will need to rename it to ``something.txt`` for it to be accepted.  Do not paste it into the input field as the formatting gets completely destroyed.

Stevemoode, 2020-09-15:
As requested. 

[pages.txt](https://github.com/dhrone/pydPiper/files/5223242/pages.txt)

Thanks

dhrone, 2020-09-15:
I made a small change to the code which will have the effect of expiring the current widget when the sequence it is contained within expires.  This will cause the next widget in that sequence to be rendered the next time the sequence is active.  Give it a try and see if it resolves your problem.

You need to pull it from the seq_reset branch.

Stevemoode, 2020-09-15:
I've copied your change and as expected it is working exactly as I had hoped.

Many thanks for both your time and help, it really is appreciated.

dhrone, 2020-09-15:
Glad it worked for you.  I'll leave the branch up in case anyone else wants to pull from it.

closed 2020-09-15 by dhrone