dhrone/pydPiper #17: problems under volumio 2.323

p1234321, 2017-11-24:
Hello,
First again many thanks for this awesome software! 
I just made a clean install of volumio 2.323 and installed pidpiper again and i cant get it to work!
when i downgrad to volumio 2.285 averything works.
any glue what the problem could be?

best regards

paul

dhrone, 2017-11-24:
Did you receive any errors?  Check the log file at /var/log/pydPiper.log and see if anything is reported there.

dhrone, 2017-11-26:
It looks like their was a version change for docker that is broken on the RPi.  Try issuing the following to downgrade to the prior version.

`sudo apt-get install -y --force-yes docker-ce=17.09.0~ce-0~raspbian`

Then reboot and see if it comes back up successfully.

paudio25, 2017-11-27:
Hello,
I ran into the same issue with broken docker. After downgrading and reboot, my pydPiper.log gave this:

2017-11-27 21:24:28,803:INFO:pydPiper starting
2017-11-27 21:24:29,020:ERROR:Uncaught exception
Traceback (most recent call last):
  File "/app/pydPiper.py", line 615, in <module>
    lcd = displays.hd44780_i2c.hd44780_i2c(rows, cols, i2c_address, i2c_port)
  File "/app/displays/hd44780_i2c.py", line 134, in __init__
    self.write4bits(0x33,False)
  File "/app/displays/hd44780_i2c.py", line 160, in write4bits
    self.bus.write_byte(self.i2c_addr, bits_high)
IOError: [Errno 121] Remote I/O error
2017-11-27 21:24:40,635:INFO:pydPiper starting
2017-11-27 21:24:40,747:ERROR:Uncaught exception
Traceback (most recent call last):
  File "/app/pydPiper.py", line 615, in <module>
    lcd = displays.hd44780_i2c.hd44780_i2c(rows, cols, i2c_address, i2c_port)
  File "/app/displays/hd44780_i2c.py", line 134, in __init__
    self.write4bits(0x33,False)
  File "/app/displays/hd44780_i2c.py", line 160, in write4bits
    self.bus.write_byte(self.i2c_addr, bits_high)
IOError: [Errno 121] Remote I/O error

Does it mean that I have chosen the wrong display ? If so, how do I perform a re-install with the correct display. I had some hesitation between the hd4778 and the Winstar WEH display. I bought it at Audiophonics.fr at january.
Thanks in advance and keep on the good work.
PB

dhrone, 2017-11-28:
Yes, this error would be caused by the selection of the wrong display type.  If you have the Raspdac from Audiophonics, you have a Winstar WEH display.  There is an example of the correct ExecStart for it in the pydpiper.service file.  If not, let me know what you did purchase and I’ll give you a specific example for your system.

paudio25, 2017-11-28:
hi dhrone,

Thank you for your reply. I can only find an example for Winstar WEG display only in the pydpiper.service file, not for WEH display. I bought this at Audiophonics:

https://www.audiophonics.fr/fr/dac-diy/audiophonics-i-sabre-dac-es9023-v3-tcxo-raspberry-pi-20-a-b-i2s-p-10657.html

My DAC is the V3 Generation 2. I also have a problem with docker, it does not start after a second reboot. Do I have to re-run install.sh.

Regard,

PB

paudio25, 2017-11-29:
Hi dhrone,
I managed to solve the problems. It's all working now. You were right about the Winstar WEH driver, I use: ExecStart of "# RaspdacV3 without weather"
Here how I solved it by making a fresh install of docker:

1- Remove docker : 
sudo apt-get purge docker-ce
sudo apt-get autoremove 
rm -rf /var/lib/docker # This deletes all images, containers, and volumes

2- Re-install docker:
sudo apt-get install -y --force-yes docker-ce=17.09.0~ce-0~raspbian

3- Check if it works:
docker run armhf/hello-world

4- Make volumio member of docker group"
sudo usermod -aG docker volumio

5- Then the rest from install.sh:

sudo systemctl enable docker
sudo systemctl start docker
sudo docker pull dhrone/pydpiper:v0.31-alpha
sudo cp pydpiper.service /etc/systemd/system
sudo systemctl enable pydpiper
sudo systemctl start pydpiper

I hope this can help another people who ran in the same corrupted installation.
Thanks for your tip about the driver.

dhrone, 2017-12-02:
Excellent work.  I'll revise the instructions and the install script after a few tests.

p1234321, 2017-12-07:
i have it running now without docker, since I am experimenting a lot with settings and displays and it works perfectly!
thanks a lot!

manolomalele, 2019-12-22:
That's interesting, I've installed pydpiper on the latest version of volumio. Smooth installation and clock on the display but, playing music, no info are not displayed. 
There's something I have to check?