dhrone/pydPiper #18: PydPiper with LCD

Alien8ions, 2017-12-03:
Hi
I'm having fun getting your script to work with a LCD display.
 I did have to copy the files out of the docker container just to be able to tweak it to run fine. As the lcd is quite slow (some overly cheap Chinese hd44780 controller) the most important  is to introduce the ms delay before the data bus setup and to lower the hesitate time to get the scrolling readable.
In the config file the displaytype was locked to the OLED display. So uncommmenting the right line in the service file didn't work as said in the manual.

I don't know if you plan on getting the lcd support fully working, but I'd like to add some backlight control. Since I'm not so well versed in python it's hard to find the correct place to put it.
Any thoughts?
TIA
Diego

dhrone, 2017-12-08:
I’ll have to look into it.  The original HD44780 display I was using didn’t have software controllable backlight but I think one of the ones I have that has an I2C interface does.  What interfaces does your use?

Alien8ions, 2017-12-09:
Thx for looking into this.

The interface I'm using is an arduino lcd shield for the moment. Patched the wires from an audiophonics dac to it and use the 'bare' HD44780 interface. The shield has a switching transistor for the backlight, directly from the 5V line. Handy since the 3.3V from the pi won't probably even light it up nor have enough current to drive it (about 20 mA when fully lit). (I do have quite a bit of experience with basic electronics)
I believe it should be possible to switch the backlight on and off and preferably use some (rather slow, 125 Hz?) pwm to control it depending on the status.
E.g. Dim when in stop mode, full on when a change to the system occurs (start playing, change track, source...) and some thing in between when just playing...
So basically just one pin as control should do it....

I looked in the files, made some tweaks here and there but I can't think of a suitable way to add this as such...

Would be happy to do some experimentation. Nothing is urgent, I have learned a lot of python along the way so thanks for all you've done with this!

Diego

dhrone, 2018-02-05:
Did you ever make any progress adding the backlight support to the code?  If you can provide a pointer to documentation on controlling the backlight I can at least point you in the right direction.

Alien8ions, 2018-02-14:
Sorry for the long wait, work and a small kid limit my playtime...

I've managed to get the software pwm running in another python  thread, next to the pydpiper script.
It is working but the soft pwm is glitchy at best, no matter which frequency I use the backlight flickers more or less when the duty cycle is less than 100 %.

I believe I can solve this with some added electronics or try hardware pwm. From what I've found the pi has a few hardware pwm pins. But that might make it harder to make your script universal.
I'll see if it's uberhaupt possible with the hardware (RPi3 and audiophonics dac amp.

Even if it's working  I'm still a bit unsure where the states are being evaluated.
Eg when in stop mode,  after the text 'stop' is displayed and the clock shows I'd like to dim the display 

My simple code is this
` import RPi.GPIO as GPIO                                                                 
 GPIO.setmode(GPIO.BCM)                                                                  
 GPIO.setup(15, GPIO.OUT)                                                                
 pwm.start(50)                                                                           
 pwm.ChangeDutyCycle(80)                                                                 
`

simple enough but it would be interesting to only include it if a backlight pin is present 

Diego

Alien8ions, 2018-06-27:
As a follow-up 

I've succeeded in getting this to work reasonably well.
Probably not good programming practice, but the workaround was to start another python script from within the main pydPiper script.
This just polls the player state occasionally and sets the pwm duty cycle accordingly.

`volumio@volumio:~/pydPiper$ cat backlight.py                                         
#!/usr/bin/python                                                                    
                                                                                     
import subprocess                                                                    
import RPi.GPIO as GPIO                                                              
import time                                                                          
                                                                                     
GPIO.setmode(GPIO.BCM)                                                               
backlight=15                                                                         
GPIO.setup(backlight,GPIO.OUT)                                                       
pwm=GPIO.PWM(backlight,220)                                                          
pwm.start(10)                                                                        
                                                                                     
while True:                                                                          
        process = subprocess.Popen('volumio status', shell=True, stdout=subprocess.PI
PE)                                                                                  
        status = process.communicate()[0]                                            
        statusLines = status.split('\n')                                             
        if "play" in statusLines[1]:                                                 
                pwm.ChangeDutyCycle(75)                                              
        else:                                                                        
                pwm.ChangeDutyCycle(10)                                              
                                                                                     
        time.sleep(20)`

Maybe it's a bit crude, but it works.
Also the software pwm is a bit jiffy, the backlight flashes a bit, I'll try to make it look neater with some  extra filtering.

HTH 
Diego

closed 2018-06-27 by Alien8ions