dhrone/pydPiper #19: timezone problems

p1234321, 2017-12-07:
whatever timezone I set, it has no influence on the displayed time. the utc timezone of the raspberry is correct, but when I start the script the display shows an minus 6 hour offset. I tried editing the timezone in pydPiper.config but it seems not to have an impact.
any glues?

paudio25, 2017-12-07:
I used the ExecStart in pydPiper.service with "--timezone France/Paris" it did NOT worked.
It work with "Europe/Amsterdam" . I hope this helps.

p1234321, 2017-12-07:
i trieb the following:
`sudo python pydPiper.py --volumio --driver ssd1306_i2c --width 128 --height 64 --i2caddress 0x3C --timezone "Europe/Amsterdam" --temperature celsius --pages pages_ssd1306.py `
but it has no effect.

p1234321, 2017-12-08:
it also seems to pass the right arguments, since it notices if I pass wrong values, but the time doesn't change.

paudio25, 2017-12-08:
Did you check the log?
Also see this https://github.com/volumio/Volumio2/issues/741 .

dhrone, 2017-12-08:
I found the problem.  Inside the pages_ssd1306.py file, the text widget that displays the time is not using the localtime variable.  Instead it is using utc variable with the strftime conversion set to US/Eastern.  I added the localtime variable relatively recently and apparently did not convert all of my sample page files to use it.  Check out the example from pages_weg_100x16.py and you can see the change that needs to be made.  When I get on a real computer (working from my iPad right now) I’ll make the change to the file myself.

p1234321, 2018-01-05:
thanks that did the trick!!!!
really awesome piece of software

closed 2018-01-05 by p1234321