dhrone/pydPiper #20: PidPiper with Moode 4 beta 12

Wadonge, 2017-12-15:
Good evening,
I try to run your python program "pydPiper" with moode 4B12. I tried but it does not work. Is it compatibel with MoOde 4b12?
I own a RaspDAC with rpi3 + [AUDIOPHONICS I-Saber V3 DAC ES9023 TCXO](https://www.audiophonics.fr/en/diy-dac/audiophonics-i-sabre-dac-es9023-v2-tcxo-raspberry-pi-20-a-b-i2s-p-10657.html) + [Screen Winstar OLED 16x2 White HD44780](https://www.audiophonics.fr/en/screens-vu-meters/winstar-weh001602a-ecran-oled-16x2-blanc-hd44780-p-8512.html).

Do you have any idea how to make it work?

Thank you

dwhacks, 2018-01-20:
I am also quite interested in this!

dhrone, 2018-02-05:
I'll have to look into it.  The code has been significantly modified since I did any Moode testing but it shouldn't be that hard to get it working.  The big issue will be whether I can get docker running on the distribution easily as pydPiper is now run out of a container to reduce dependencies on the underlying OS.

dwhacks, 2018-02-05:
Well moode is now installed on a standard stretch image, there is no longer a precompiled .img to download. 

I would really like to have this on my player, as I find it looks much nicer then the other display setups available. I hope it will work with the display directly connected to the pi without an i2c backpack as well.

rikardo1979, 2018-04-27:
I also looking into moOde support. Hope this become available. At the moment I using your old one @dhrone

dhrone, 2018-05-28:
I've got Moode working again but only for content that uses the MPD daemon.  Instructions for installing from fresh Moode 4.1

`cd /home/pi`
`git clone https://github.com/dhrone/pydPiper`
`cd /home/pi/pydPiper`
`sudo install_docker.sh`
Make sure that docker reports that it has installed correctly!

Next, run the new configure script.
`sudo python configure.py`

Configure.py will produce two files when finished.  pydpiper.service to be used by systemd to autostart pydPiper and pydPiper.cfg which is where all of the configuration information for pydPiper now resides.

To test the installation you can open up a shell within the docker container and manually run pydPiper...
`sudo /usr/bin/docker run --network=host --privileged -ti -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha /bin/bash`
`python pydPiper.cfg`

If pydPiper doesn't start up successfully, you may want to change the logging level to debug to see if any messages show up that help determine what the problem is.

If it is working correctly, you can enable it within systemd using the following commands.
`sudo cp /home/pi/pydpiper.service /etc/systemd/system`
`sudo systemctl enable pydpiper`
`sudo systemctl start pydpiper`

Note, I have not had much time to test this so my apologizes in advance if it fails.

Good luck!

rikardo1979, 2018-05-30:
Nice one @dhrone 
I am currently on Europe holiday so will test it when we are back home sometime end of the next week.
Will report back here

dwhacks, 2018-05-31:
I've been waiting for this to update my Moode install. I'll try it out this weekend hopefully and report back!

MikkelLau, 2018-06-09:
Thanks again for your great code:-) 

I have been trying to install this but with some errors.
First I had to use this to install docker:
`sudo ./install_docker.sh` 
Next i run the new configuration file and that works great by the way:-)
I do get a syntax error in line 15 regarding the log file when I test the installation. I added cd in front of the log path:
`logfile = cd /var/log/pydPiper.log`
I think this solved the problem and I do not get any syntax errors. Is this correct?
Next when testing the installation file I get an error regarding the [STARTUP]:
`Traceback (most recent call last):
  File "pydPiper.cfg", line 1, in <module>
    [STARTUP]
NameError: name 'STARTUP' is not defined`

Could you please help with this? :-) 

Regards MIkkel

dhrone, 2018-06-09:
A small typo on my part.  the test line should read...
`sudo /usr/bin/docker run --network=host --privileged -ti -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha /bin/bash
python pydPiper.py`

The difference is at the very end of the line.  I accidentally used cfg as the extension when the correct file to run has the py extension.

You will need to revert the changes you made to the pydPiper.cfg file.  You could also remove it and then re-run the configure program.

MikkelLau, 2018-06-09:
Thanks for your reply:-)
And it works!!! 👍 :-) 
Your last correction do not work. But using your code from the above install description and replacing `.cfg` with `.py`
For others: `sudo /usr/bin/docker run --network=host --privileged -ti -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha /bin/bash
python pydPiper.py`
What file is to be enable in the systemd? Your code says it is the `sudo cp /home/pi/pydpiper.service /etc/systemd/system`. Is that also for the new code? Error: no such file or directory...

I was not able to make the time work with "France/Paris". US/Eastern works

Thanks again :-)

dhrone, 2018-06-09:
The configure program creates the service file which is named pydpiper.service.  It needs to be moved into the appropriate directory for Moode and then enabled (and then started) using the systemctl command.  Unfortunately, I don't have a Moode instance running today so can't verify the directory.

The correct TZ value for Paris is Europe/Paris 
See https://en.wikipedia.org/wiki/List_of_tz_database_time_zones for list of possible values.

No, the LCD update engine works using a completely different mechanism.  At the moment, it does not need to be activated for pydPiper.

MikkelLau, 2018-06-10:
That sounds great:-) Let me know if I should test something. 

I do also get a "no active widgets" when playing from the MPD. Would that also be related to the correct moode directory? I have used default settings from the pydPiper.cfg...

I am using a 24x2 display and have used 120 width and 16 in height. The text and clock is off center and sometimes the weather text is rolling and not using the entire length of the display. Should this automatically be adjusted when i specify the dimensions or should I start to change all dimensions in the pages_lcd_16x2.py? 

Thanks again :-)

MikkelLau, 2018-06-10:
Regarding the pages I have also tried the pages_fixed_moode.py. There is a lot of question marks at a 1/4 of the bottom left of 24x2 HD44780 i2c display. I have tried to change the width to 80 but the same.

dhrone, 2018-06-10:
Configuring the page files for an HD44780 is a challenge as the placement of characters has to exactly match the pixel matrix of the underlying display and you can only use the latin1_5x8_lcd.fnt and BigFont_10x16_fixed.fnt fonts.  The reason is that pydPiper treats all displays as graphics displays as opposed to character displays.  HD44780 displays are character based with each character being represented by a 5x8 pixel sprite that is stored in the display's character generator.  pydPiper's HD44780 driver converts from its graphical representation into the HD44780s characters by looking at the display as a set of 5x8 blocks and then seeing if it can match what is inside the block to a character in the HD44780 displays character map.  This only works if the pages file is set to draw the characters using a font that exactly matches the HD44780s and that the characters are drawn in alignment with that 5x8 grid.  When there is a mismatch between a grid value and what is within the character generator, a custom character is created on the fly, stored in the HD44780s limited custom character memory and then displayed.  Unfortunately, you can only have 8 custom characters loaded at a time which means that if you have say a few foreign characters you need to display it works great but if you are trying to create an entirely custom display, you'll run out of custom font memory.  At this point you get the ? marks that you are seeing in the display.

I have not run across a 24x2 display before.  Please send me a link so I can take a look and see if there are any other issues that need to be addressed.  

FYI, the "no active widgets" message occurs when none of the sequence conditions are being met.  You may want to run pydPiper with the --showupdates flag and then see what the values are when you are getting the "no active widgets" message.  Most of the time it is a problem with the "state" variable.

MikkelLau, 2018-06-11:
Thanks for that detailed description of how you write to a display:-) That was very useful.

I have bought this display:
http://www.buydisplay.com/download/manual/ERM2402-1_Series_Datasheet.pdf
From www.buydisplay.com:
https://www.buydisplay.com/default/3-3v-5v-24x2-character-lcd-display-module-hd44780-white-on-black
It is running a SPLC780C controller:
http://www.buydisplay.com/download/ic/SPLC780.pdf

I have tried to use the --showupdates in the shell code but unsuccessful. I have both added in the steel line and when the python (--showupdates) pydPiper.py is executed. Also tried to state "True" in the self.showupdates=showupdates but this isen't displaying anything...
Sorry but I am still learning and this coding is still new to me :-)

MikkelLau, 2018-06-11:
I figured out the --showupdates. Sorry :-) 
For other:
`sudo /usr/bin/docker run --network=host --privileged -ti -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha /bin/bash
python pydPiper.py --showupdates`

MikkelLau, 2018-06-11:
I figured out the --showupdates. Sorry :-) 
For other:
`sudo /usr/bin/docker run --network=host --privileged -ti -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha /bin/bash
python pydPiper.py --showupdates`

Could you give me a hint on what to look after with the showupdates please? I can see that it does read the radiostream I am playing with url and station name under artist. It also read the freq ans so on. 
is the <type 'unicode'> an error? I can see this, <type 'float'> and <type 'int'> next to the different info...

dhrone, 2018-06-11:
If you look at the pages file, you'll see condition statements associated with each sequence and also attached to canvases within a sequence.  To use the showupdates information, get your player into the state that is displaying the no active widget message and then look at the showupdates variables.  See if there is a sequence (and a canvas) that should be displaying, but is not.

The <var type> labels are not errors.  It is just showing what type of variable is being reported.  FYI, unicode is a type of string.

MikkelLau, 2018-06-16:
Hi again and once again thanks for helping.

I tried to look at the showupdates and noticed the artist line looked different:
`Status at time 8:22:22 PM
    [encoding]=u'webradio' <type 'unicode'>
    [ip]=u'192.168.1.4' <type 'unicode'>
    [outside_temp_max_formatted]=u'22\xb0C' <type 'unicode'>
    [random]=0 <type 'int'>
    [outside_temp_min]=12.0 <type 'float'>
    [channels]=2 <type 'int'>
    [single]=0 <type 'int'>
    [duration]=320 <type 'int'>
    [samplerate]=u'44.1 kHz' <type 'unicode'>
    [system_tempf]=155.8 <type 'float'>
    [random_onoff]=u'Off' <type 'unicode'>
    [system_tempc]=68.8 <type 'float'>
    [album]=u'Live from the Rebel Hearts Club' <type 'unicode'>
    [current_time]=u'20:22' <type 'unicode'>
    [system_temp]=68.8 <type 'float'>
    [bitdepth]=u'16 bits' <type 'unicode'>
    [musicdatasource]=u'MPD' <type 'unicode'>
    [disk_availp]=77 <type 'int'>
    [title]=u'Rebel Heart (Live)' <type 'unicode'>
    [playlist_count]=2 <type 'int'>
    [bitrate]=u'884 kbps' <type 'unicode'>
    [current_tempc]=68.8 <type 'float'>
    [current]=183 <type 'int'>
    [outside_temp_max]=22.0 <type 'float'>
    [state]=u'play' <type 'unicode'>
    [time_formatted]=u'20:22' <type 'unicode'>
    [type]=u'' <type 'unicode'>
    [remaining]=u'2:01' <type 'unicode'>
    [localtime]=<Moment(2018-06-16T20:22:22)> <class 'moment.core.Moment'>
    [time_ampm]=u'PM' <type 'unicode'>
    [actPlayer]=u'MPD' <type 'unicode'>
    [repeat]=0 <type 'int'>
    [time]=u'20:22' <type 'unicode'>
    [outside_temp_min_formatted]=u'12\xb0C' <type 'unicode'>
    [elapsed]=199 <type 'int'>
    [volume]=-1 <type 'int'>
    [playlist_display]=u'Radio' <type 'unicode'>
    [disk_avail]=11543310336L <type 'long'>
    [playlist_length]=2 <type 'int'>
    [outside_conditions]=u'Rain' <type 'unicode'>
    [single_onoff]=u'Off' <type 'unicode'>
    [system_temp_formatted]=u'68\xb0c' <type 'unicode'>
    [outside_temp_formatted]=u'16\xb0C' <type 'unicode'>
    [elapsed_formatted]=u'3:19/5:20' <type 'unicode'>
    [utc]=<Moment(2018-06-16T18:22:22)> <class 'moment.core.Moment'>
    [disk_usedp]=23 <type 'int'>
    [repeat_onoff]=u'Off' <type 'unicode'>
    [artist]=u'' <type 'unicode'>
    [disk_used]=3395751936L <type 'long'>
    [current_time_sec]=u'20:22' <type 'unicode'>
    [uri]=u'http://192.168.1.4:58050/stream/audio/9241f02b61c1e5cbcbc68894c347e943.flac' <type 'unicode'>
    [outside_temp]=16.3 <type 'float'>
    [current_ip]=u'192.168.1.4' <type 'unicode'>
    [playlist_position]=1 <type 'int'>
    [length]=320 <type 'int'>
    [current_tempf]=155.8 <type 'float'>
    [position]=u'3:19/5:20' <type 'unicode'>
    [current_time_formatted]=u'20:22' <type 'unicode'>
    [tracktype]=u'webradio Stereo 16 bit 44.1 kHz' <type 'unicode'>`
So I comment out the artist, playartist and playartist_radio in the widget, sequences and canvases but still the same problem with no active widgets. I also tried with the random sequences, still the same. Sorry but I am not that good at coding:-)

I did mange to change some of the canvases to display at the 120x16(weather info), but the time canvas is somehow locked to the 70x16 size and I can't shift it further than 80 even through I have made the size 120x16 in the canvas...

Thanks again :-)

MikkelLau, 2018-06-18:
I did manage to shift the time further to the right on the display. Now it is in the center. I had forgot to more the AM/PM... :-)

MikkelLau, 2018-06-19:
Thanks! That worked for now! :-) 
I will try to make a 24x2 LCD I2C page file. I can send it to you when i am finished if your are interested:-)  
If i should test something regarding the auto start on boot up running moode audio please let me know.

MikkelLau, 2018-06-20:
Hi again. 
Sorry for all the question:-) 
I jsut had a error when trying to run the docker deamon. It suddenly stopped working after a reboot. Have you heard of this before?
I have tried to run the docker install again but with an error:
`Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
Warning: failed to get default registry endpoint from daemon (error during connect: Get http://%2Fvar%2Frun%2Fdocker.sock/v1.37/info: read unix @->/var/run/docker.sock: read: connection reset by peer). Using system default: https://index.docker.io/v1/
error during connect: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.37/images/create?fromImage=dhrone%2Fpydpiper&tag=v0.31-alpha: read unix @->/var/run/docker.sock: read: connection reset by peer
Testing docker

docker: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?.
See 'docker run --help'.`

If this is just an error on my raspberry pi I will make a total reinstall, but it is just a lot of work:-) 
Thanks again:-)

MikkelLau, 2018-06-20:
Hi again. 
Sorry for all the question:-) 
I jsut had a error when trying to run the docker deamon. It suddenly stopped working after a reboot. Have you heard of this before?
I have tried to run the docker install again but with an error:

`Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
Warning: failed to get default registry endpoint from daemon (error during connect: Get http://%2Fvar%2Frun%2Fdocker.sock/v1.37/info: read unix @->/var/run/docker.sock: read: connection reset by peer). Using system default: https://index.docker.io/v1/
error during connect: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.37/images/create?fromImage=dhrone%2Fpydpiper&tag=v0.31-alpha: read unix @->/var/run/docker.sock: read: connection reset by peer
Testing docker

docker: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?.
See 'docker run --help'.`

I ran the systemctl and journalctl -xe but not sure what to do with this other than the failed in loading:
`pi@moode:~ $ systemctl --state=failed
  UNIT           LOAD   ACTIVE SUB    DESCRIPTION                        
● docker.service loaded failed failed Docker Application Container Engine
● docker.socket  loaded failed failed Docker Socket for the API          

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

2 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.`

I have also made an update and upgrade command...
If this is just an error on my raspberry pi I will make a total reinstall, but it is just a lot of work:-) 
Thanks again:-)

dhrone, 2018-06-20:
I'm assuming a reboot didn't resolve the issue.  The problem is some kind of failure of the docker system.  Unfortunately, a reinstall is likely the easiest resolution.

MikkelLau, 2018-06-20:
Thanks once again for helping and you are assuming right that a reboot didn't solve the problem :-) 
Removing the docker only was fine and now it is running again:-) Reinstalled pydPiper :-)

I have been messing a bit around with the "artist" section but i can't make it work. I have tried to change it to actPlayer in the sequences but no success. The [artist]=u'' <type 'unicode'> is empty when I use the --showupdates ... it's the same when running the MPD and Radio. Do you have any good ideas :-)? 

For others with same issue(backup customized page file first!):
`cd /home/pi/pydPiper
sudo apt-get purge docker-ce
sudo rm -rf /var/lib/docker
cd
sudo rm -rf cd pydPiper`
Reboot and install everthing again as described above by pydPiper.

rikardo1979, 2018-06-21:
OK, so I may have some time finally to try this.
Now reading all the comments, what is the final and right way to install this???
Which steps to follow to install it on latest moOde?

MikkelLau, 2018-06-21:
Hi Rikardo 
The second comment by pydpiper should be fine. Just use this for install:
$ sudo ./install_docker.sh
I made a comment later with a correct line for executing the docket with the pydpiper. (Can’t copy on my phone)

There isn’t a final solution for the auto start of pydpiper yet. Maybe you have a good idea? :-)

Regards Mikkel

MikkelLau, 2018-07-04:
Hi Drone

I managed to figure out to autostart the pydPiper in Moode. I used some of your code from the volumio install file ;-) And it worked!
Autostart on boot:
`$ cd pydPiper`
`$ sudo cp pydpiper.service /etc/systemd/system/pydpiper.service`
`$ sudo systemctl enable pydpiper`

The display of Artist work when playing from HDD. I can still not get it to work in webradio. 
--showupdates display an empty field but using the Moode UI in my Browser does show artist. Do you have any good ideas on how to solve this please?

I am using Spotify Connect made by RafaPolit: https://github.com/RafaPolit/moode-spotify-connect-web
He has made a UI using Node red. I think he is using a small server (not sure!) to store what is playing using this URL: http://localhost:4000/api/info/status. Would it be possible to collect this info and somehow display this in pydPiper. I am no programmer so sorry if this is a stupid question:-)

Thanks again for this great code:-)

MikkelLau, 2018-12-17:
My display didn't work after updating to MoOde 4.4. The docker was not running. 
I tried to uninstall but was not able to uninstall the Docker. After failing I had to issue the following command:
$ sudo dpkg --configure -a
After this it is possible to uninstall the docker(and all pydPiper) bu using the purge command as I have commented on the 20 jun in this thread.
When installing docker once again there will be a message displaying that the docker is already installed. Please ignore that and install the everything once again as previously described.