dhrone/pydPiper #21: Adding SH1106 driver compatibility?

MikkelLau, 2018-01-17:
Hi pydPiper

Is it possible to add support for displays using the SH1106 driver please? 
I have my display running using the Luma examples, but I do not have the skills to collect the text from volumio,:-) 

Thanks 
Best regards Mikkel

dhrone, 2018-02-05:
It is supported but not tested as I do not have any SH1106 displays.  
Just use pydPiper -d luma_i2c and --devicetype sh1106 ...

MikkelLau, 2018-02-05:
Hi There

Thanks for your kind reply and thanks for the great work down with this script of different display:-)

I have read some of your code in the pydpiper.py file at line 530 and 659 and then tried to make this exec start file according to your recommendations:

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py -d --volumio --driver luma_i2c --width 128 --height 64 --i2caddress 0x3c --i2cport 1 —-devicetype sh1106 --timezone 'France/Paris' --temperature celcius --wapi 4584f00e8061268d --wlocale 'Denmark/Odense'  --pages pages_ssd1306.py

I2c address: 0x3c

But i can’t make this work. Is this all wrong? 

I have only used the  sudo -/install.sh to make the pydpiper run, but not sure it is running…

Thanks again

Best Regards / Med Venlig Hilsen

Mikkel Lauridsen

Email:  miklau.dk@gmail.com <mailto:milau08@student.sdu.dk>
Email:  m.lauridsen@icloud.com <mailto:milau08@student.sdu.dk>
Phone:  +45 25 32 43 93

Uffesvej 34 ST. TH.
5200 Odense V
Denmark

MikkelLau, 2018-02-05:
Hi Again.

Sorry it was this file i used:

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio —-driver luma_i2c --width 128 --height 64 --i2caddress 0x3c --i2cport 1 —-devicetype sh1106 --timezone 'France/Paris' --temperature celcius --wapi 4584f00e8061268d --wlocale 'Denmark/Odense'  --pages pages_ssd1306.py

thanks again

Best Regards / Med Venlig Hilsen

Mikkel Lauridsen

Email:  miklau.dk@gmail.com <mailto:milau08@student.sdu.dk>
Email:  m.lauridsen@icloud.com <mailto:milau08@student.sdu.dk>
Phone:  +45 25 32 43 93

Uffesvej 34 ST. TH.
5200 Odense V
Denmark