dhrone/pydPiper #22: Time updates every 5 minutes

gizegoiz, 2018-01-24:
First of all thank you for this software. You've done really good job.
However I have still the issue with time updating every 5 minutes. I have followed installation instructions from README.md. I use 24x2 hd44780 with weather informations. Everything is ok, except this time updating.
I have tried to do fresh installation:
sudo git clone https://github.com/dhrone/pydPiper.git
cd pydPiper
sudo ./install_docker.sh
but it doesn't work at all.
Am I doing something wrong?

Alien8ions, 2018-01-30:
Ran into this too
Changed this line in the pydpiper.py file
`		# Read environmentals every 5 minutes
			time.sleep(300)`
To 
`		# Read environmentals every 1 minutes
			time.sleep(60)`
Works ok like this
Btw you can't use the docker container for this, or dhrone has to make these changes in the github files

gizegoiz, 2018-01-30:
Thank you, Alien8ions. It really works.

closed 2018-01-30 by gizegoiz