dhrone/pydPiper #23: tiny font doesn't allow for non-capitalized letters

Saiyato, 2018-02-24:
First of all, great work! I've been following your progress on the line displays for some time and I must say: kudos to you sir!

I've always wanted to do thing myself and understand why things work the way they do. That's the reason I only recently turned to your python solution, which is both robust and elegant.

That being said, I remain curious about how stuff works, so I played around with the configs and decided to go for the suggested solution about editing the mounting point for the app.py (and pages), this allowed me to change things.

TLDR;

When editing a pages file and using font tiny, question marks are shown for non-capitalized letters. For example if I want to show bit depth (24 ???) or sample rate (192 KH?). 

WIDGET example:
`'bitdepth_tiny': { 'type':'text', 'format':'{0}', 'variables':['bitdepth'], 'font':'tiny','varwidth':True},`

Also a dot in tiny seems to show a single (extra) pixel in the right bottom and four in the left.

dhrone, 2018-02-26:
This response assumes that you are using 'upperascii_3x5.fnt' as tiny.  This is configured in the FONTS section of the pages file.

'upperascii_3x5.fnt' does not contain lower-case characters.  This bit-mapped font is simply too small to allow lower-case characters to be represented.  If you want to use this font for content that contains lower-case characters, use the '| upper' transformation on the variable you are displaying to force it to all upper characters before use.  An example...

In pages_weh_80x16.py inside WIDGETS, there is a widget called nowplaying which uses the variable actPlayer but converts actPlayer to upper case by using the '| upper' transform operator.  So that line reads:
'nowplaying': { 'type':'text', 'format':'{0}', 'variables':['actPlayer**|upper**'], 'font':'tiny', 'varwidth':True}

Saiyato, 2018-02-26:
Hi,

I'm sorry, I should've provided you more info.

I'm using the Audiophonics case with the Winstar WEH OLED display connected to the Sabre V3 DAC.
Also I'm indeed using the upperascii font (upperasciiwide_3x5_fixed.fnt), the name should've triggered me that it's only upper case. 

As for the pixel showing, that was when a dot (for example in 2.82) was printed to the screen.

Saiyato, 2018-03-29:
Not an issue, the font itself doesn't allow for non-capitalized letters. Case closed ;)

closed 2018-03-29 by Saiyato