dhrone/pydPiper #24: elapsed_formatted only option to show progress

Saiyato, 2018-02-25:
Hi,

It seems like elapsed_formatted is the only way to show progression in minutes and seconds, but since most functionality in Volumio 2 provides a duration it's always formatted "elapsed/total".

Is there a way to incorporate an extra elapsed that only shows progression, always?

I've not locally edited the pydpiper.py and musicdata_volumio2.py; should I change both? I'm not too Python savvy to say the least. Else I'll just create a PR with my suggested change for all sources (I think I need edit all of them).

My change in pydpiper.py is as follows
```
		'elapsed_formatted':u"",
		'elapsed_simple':u"",
		'volume':-1,
.....
					if self.musicdata[u'length'] > 0:
						timepos = time.strftime("%-M:%S", time.gmtime(self.musicdata[u'elapsed'])) + "/" + time.strftime("%-M:%S", time.gmtime(self.musicdata[u'length']))
						remaining = time.strftime("%-M:%S", time.gmtime(self.musicdata[u'length'] - self.musicdata[u'elapsed'] ) )
						timepos_simple = time.strftime("%-M:%S", time.gmtime(self.musicdata[u'elapsed']))
					else:
						timepos = time.strftime("%-M:%S", time.gmtime(self.musicdata[u'elapsed']))
						remaining = timepos

					self.musicdata[u'remaining'] = remaining.decode()
					self.musicdata[u'elapsed_formatted'] = self.musicdata[u'position'] = timepos.decode()
					self.musicdata[u'elapsed_simple'] = timepos_simple.decode()
```

Saiyato, 2018-02-25:
Just another thing, I'd like a total time as well, if you can tell me which files I need to update for you, I will :)

Why would I want this? Well I was inspired by the CXN streamer and Pioneer N-50. Ideally I want an LCD 
that looks just like that for Volumio, but I'm far from it. It's full of interesting information and I like that it shows artwork as well.

TLDR;

I want to create a canvas that has the following lines (which I did locally, but I', not sure it works for all your users):

Seq 1 (15s)
LINE 1: SMALL (8x5 font) | Artist - Title
LINE 2: TINY (5x5 font) | {elapsed}  {filetype} {total} + progessbar on the last three pixels

Seq 2 (5s)
LINE 1: SMALL (8x5 font) | {sample rate} | {bit depth}
LINE 2: TINY (5x5 font) | {actPlayer} + progessbar on the last three pixels

I may need to play with the duration of the canvases, but that the easy part.

dhrone, 2018-03-13:
I'm not 100% sure if I understand what you are looking for.  Are you asking for a way to reformat the elapsed and length variables?  Probably the most general solution would be to write another transform to convert from an integer to a formatted string.  Let me take a look.  That shouldn't be very difficult.  If you want to try to add that feature yourself, the file is display.py in function transformvariable at line 110.

Saiyato, 2018-03-29:
I just created PR #27 for the above, thanks for the pointer :)

closed 2018-03-29 by Saiyato