dhrone/pydPiper #25: Garbled screen

spinko101, 2018-03-05:
Hi, as far as I can see, I have everything working on the software side, i.e. the screen comes on when a song is played but unfortunately it is full of white dots 8-P (totally garbled). I have got the screen to work properly 

I have done a clean install (Volumio, Docker, pydPiper as stated in another post here) with no effect. The screen is OK because I got it to work normally with a python script provided by Adafruit.

Any help would be muuuuuch appreciated. Thanks.

Configuration is:

Raspberry Pi3, Oled SSD1306 via i2c on 0x3D, Volumio 2.368, Allo Kali, Allo Piano2.1 DAC.

pydpiper.service

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio --driver ssd1306_i2c  --width 128 --height 64 --timezone 'paris/france' --temperature celsius --pages pages_ssd1306.py --i2caddress 0x3D --i2cport 1

[image]

dhrone, 2018-03-06:
The library that pydPiper uses to display to the ssd1306 is Luma.  https://pypi.python.org/pypi/luma.oled.  See if you can get any of their demo programs running on the display.

spinko101, 2018-03-06:
I have successfully tested my SSD1301 Oled with the Adafuit scripts which are based on the luma scripts I think. Looking though the luma-oled.readthedocs.io site I found this article describing the issue I'm having. It seems to be related to the display initiation.

https://luma-oled.readthedocs.io/en/latest/troubleshooting.html

Basically, it says to use the SH1106 driver instead of the SSD1301 driver. I did try to install luma but got multiple errors and gave up. I'm not a developer, so for me its pretty hard to navigate and comprehend all this information. But as said, the display works with the Adafruit packages.

Where is the snippet that initiates the display ? Maybe I could try and get it out of one of the Adafruit scripts and paste it somewhere in a pydpiper script ?

thanks

spinko101, 2018-03-06:
update: finally got luma to work. All demos I tried work with the sh1106 driver ;-)))

dhrone, 2018-03-06:
The SH1106 driver is already integrated into pydPiper.  Try the following start line...

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio --driver luma_i2c --devicetype sh1106  --width 128 --height 64 --timezone 'paris/france' --temperature celsius --pages pages_ssd1306.py --i2caddress 0x3D --i2cport 1

spinko101, 2018-03-06:
hi, no unfortunately the screen doesn't even come on.

spinko101, 2018-03-06:
hmm, not sure what's going on but it only says "NONE". I restarted the daemon before testing and cut the DC to the screen before restarting pydpiper.

spinko101, 2018-03-06:
hmm, unfortunately nothing or snow on the screen. I will reinstall everything and try later.

spinko101, 2018-03-06:
after clean install, still not working. Reinstalled luma as well and it works.

dhrone, 2018-03-07:
It's not clear to me from your message.  Are you saying that the Luma demo programs are working but that using pydPiper with --driver luma_i2c --devicetype sh1106 does not?

spinko101, 2018-03-07:
yes, correct. It works with luma but not with the pydPiper and luma_i2c driver

dhrone, 2018-03-09:
Interesting.  You can run the luma_i2c driver in a test mode without running the rest of pydPiper.  

Try...
sudo /usr/bin/docker run -ti --network=host --privileged -v /var/log:/var/log:rw dhrone/pydpiper:v0.31-alpha /bin/bash

This will start a bash shell inside the docker container.  Once you are at the bash prompt inside the container...

cd displays
python luma_i2c.py -r 64 -c 128 --devicetype sh1106 --i2c_address 61 --i2c_port 1

Note that the command line argument parser is slightly different than what pydPiper.py uses with the biggest potential confusion being that the i2c_address needs to be specified in decimal.  This example should work though if your address is 0x3D.

spinko101, 2018-03-09:
hi dhrone, sorry, all I get is "snow" (same as in my first pic) and a lengthly output.

> root@volumio:/app/displays# python luma_i2c.py -r 64 -c 128 --devicetype sh1106 --i2c_address 61 --i2c_port 1

LUMA OLED Display Test
ROWS=64, COLS=128, DEVICETYPE=sh1106, I2C_ADDRESS=61, I2C_PORT=1
Loading font from /app/displays/fonts/latin1_5x8_fixed.fnt
2018-03-09 22:19:13,519:DEBUG:STREAM 'IHDR' 16 13
2018-03-09 22:19:13,521:DEBUG:STREAM 'gAMA' 41 4
2018-03-09 22:19:13,523:DEBUG:STREAM 'pHYs' 57 9
2018-03-09 22:19:13,524:DEBUG:STREAM 'iTXt' 78 1061
2018-03-09 22:19:13,526:DEBUG:STREAM 'IDAT' 1151 3193
2018-03-09 22:19:13,596:DEBUG:Error closing: 'NoneType' object has no attribute 'close'
2018-03-09 22:19:16,395:DEBUG:Loading ../pages_ssd1306.py as page file
2018-03-09 22:19:16,397:DEBUG:Loading font small
Loading font from /app/displays/fonts/latin1_5x8.fnt
2018-03-09 22:19:16,436:DEBUG:STREAM 'IHDR' 16 13
2018-03-09 22:19:16,437:DEBUG:STREAM 'gAMA' 41 4
2018-03-09 22:19:16,438:DEBUG:STREAM 'pHYs' 57 9
2018-03-09 22:19:16,439:DEBUG:STREAM 'iTXt' 78 1061
2018-03-09 22:19:16,440:DEBUG:STREAM 'IDAT' 1151 3193
2018-03-09 22:19:16,510:DEBUG:Error closing: 'NoneType' object has no attribute 'close'
2018-03-09 22:19:16,511:DEBUG:Loading font large
Loading font from /app/displays/fonts/Vintl01_10x16.fnt
2018-03-09 22:19:16,551:DEBUG:STREAM 'IHDR' 16 13
2018-03-09 22:19:16,552:DEBUG:STREAM 'gAMA' 41 4
2018-03-09 22:19:16,553:DEBUG:STREAM 'pHYs' 57 9
2018-03-09 22:19:16,554:DEBUG:STREAM 'iTXt' 78 1062
2018-03-09 22:19:16,555:DEBUG:STREAM 'IDAT' 1152 7058
2018-03-09 22:19:16,659:DEBUG:Error closing: 'NoneType' object has no attribute 'close'
2018-03-09 22:19:16,661:DEBUG:Loading font tiny
Loading font from /app/displays/fonts/upperasciiwide_3x5_fixed.fnt
2018-03-09 22:19:16,677:DEBUG:STREAM 'IHDR' 16 13
2018-03-09 22:19:16,679:DEBUG:STREAM 'gAMA' 41 4
2018-03-09 22:19:16,680:DEBUG:STREAM 'pHYs' 57 9
2018-03-09 22:19:16,681:DEBUG:STREAM 'iTXt' 78 1058
2018-03-09 22:19:16,682:DEBUG:STREAM 'IDAT' 1148 726
2018-03-09 22:19:16,708:DEBUG:Error closing: 'NoneType' object has no attribute 'close'
2018-03-09 22:19:16,709:DEBUG:Loading font DejaVuSans28
2018-03-09 22:19:16,717:DEBUG:Loading image splash
2018-03-09 22:19:16,718:DEBUG:STREAM 'IHDR' 16 13
2018-03-09 22:19:16,719:DEBUG:STREAM 'gAMA' 41 4
2018-03-09 22:19:16,721:DEBUG:STREAM 'pHYs' 57 9
2018-03-09 22:19:16,722:DEBUG:STREAM 'iTXt' 78 1060
2018-03-09 22:19:16,723:DEBUG:STREAM 'IDAT' 1150 295
2018-03-09 22:19:16,724:DEBUG:Loading image progbar
2018-03-09 22:19:16,726:DEBUG:STREAM 'IHDR' 16 13
2018-03-09 22:19:16,727:DEBUG:STREAM 'gAMA' 41 4
2018-03-09 22:19:16,728:DEBUG:STREAM 'pHYs' 57 9
2018-03-09 22:19:16,729:DEBUG:STREAM 'iTXt' 78 1057
2018-03-09 22:19:16,730:DEBUG:STREAM 'IDAT' 1147 107
2018-03-09 22:19:16,732:DEBUG:Loading widget xofy
2018-03-09 22:19:16,734:DEBUG:Loading widget weather
2018-03-09 22:19:16,737:DEBUG:Adding effect scroll to weather which is currently size (40, 16)
2018-03-09 22:19:16,739:DEBUG:Scroll added with widget size (60, 16)
2018-03-09 22:19:16,740:DEBUG:Loading widget temp
2018-03-09 22:19:16,743:DEBUG:Loading widget artist
2018-03-09 22:19:16,747:DEBUG:Adding effect scroll to artist which is currently size (34, 16)
2018-03-09 22:19:16,749:DEBUG:Scroll added with widget size (54, 16)
2018-03-09 22:19:16,750:DEBUG:Loading widget songprogress
2018-03-09 22:19:16,754:DEBUG:Loading widget title
2018-03-09 22:19:16,760:DEBUG:Adding effect scroll to title which is currently size (86, 8)
2018-03-09 22:19:16,762:DEBUG:Scroll added with widget size (106, 8)
2018-03-09 22:19:16,763:DEBUG:Loading widget elapsed
2018-03-09 22:19:16,766:DEBUG:Loading widget volume
2018-03-09 22:19:16,771:DEBUG:Loading widget length
2018-03-09 22:19:16,774:DEBUG:Loading widget ampm
2018-03-09 22:19:16,781:DEBUG:Loading widget volumelarge
2018-03-09 22:19:16,786:DEBUG:Loading widget temphilow
2018-03-09 22:19:16,789:DEBUG:Loading widget time
2018-03-09 22:19:16,797:DEBUG:Loading widget volumebar
2018-03-09 22:19:16,802:DEBUG:Loading widget bitdepth
2018-03-09 22:19:16,805:DEBUG:Loading widget samplerate
2018-03-09 22:19:16,809:DEBUG:Loading widget ttime
2018-03-09 22:19:16,821:DEBUG:Loading widget tempsmall
2018-03-09 22:19:16,823:DEBUG:Loading widget stoptimetemp_popup
2018-03-09 22:19:16,826:DEBUG:Loading widget volume_changed
2018-03-09 22:19:16,828:DEBUG:Loading widget stoptime
2018-03-09 22:19:16,829:DEBUG:Loading widget playing
2018-03-09 22:19:16,831:DEBUG:Loading sequence seqPlay
2018-03-09 22:19:16,832:DEBUG:Loading sequence seqStop
2018-03-09 22:19:16,833:DEBUG:Loading sequence seqVolume

spinko101, 2018-03-11:
any idea why the screen works with lumina but not with pydpiper/docker ? I have tried all sorts of combinations in the pydpiper.service file with no success. The only thing that works is that the screen comes on (covered in random dots) when pydpiper is activated.

dhrone, 2018-03-13:
Still looks like a driver issue.  I don't have a SH1106 to test with but placed one on order so should be able to test it soon.

spinko101, 2018-03-13:
Hi, thanks. I’ve ordered a couple of screens from different manufacturers.
all the best
rupert

spinko101, 2018-03-14:
I'm beginning to think it is a "reset" issue. I am taking the signals from a Piano 2.1 DAC header and ma not using RST. Just wondering out loud after having read this ..

https://github.com/adafruit/Adafruit_SSD1306/issues/111

dhrone, 2018-05-25:
Are you still experiencing display problems?

spinko101, 2018-05-29:
hi Dhrone,
sorry, but I gave up and havn’t had time to revive the project… if / when I will (revive it), I’ll def let you know via github..

all the best and greetings

rupert

spinko101, 2019-02-25:
Hi, rebooted this issue with a 3$ sh1106 display from Aliexpress. I got this display to work with cava/antiprism package. But with your solution, I get horizontal lines with garbled text. But at least, the display seems to work :-) I used the service parameters referenced in for the sh1106 driver.

_ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio --driver luma_i2c --devicetype sh1106 --width 128 --height 64 --timezone 'paris/france' --temperature celsius --pages pages_ssd1306.py --i2caddress 0x3D --i2cport 1_