dhrone/pydPiper #28: Time and weather info not updating

gizegoiz, 2018-04-10:
Hello,
I Am really satisfied using pydpiper and volumio. I have only one problem. After some random time (sometimes one hour, sometimes 24 hours) time and weather info are not updated anymore. In the same time pydpiper still works - i mean all the rest functions are ok. Where can I find reasons?

dhrone, 2018-04-10:
Not sure.   Has anything shown up in the log file?  Failures calling to the weather service are logged at WARN level.

gizegoiz, 2018-04-10:
Pydpiper log is almost empty:
2018-04-09 03:55:39,623:INFO:pydPiper starting
2018-04-09 13:35:05,686:WARNING:The Weather Underground server couldn't fulfill the request and responded with error code 404
2018-04-09 20:57:17,860:WARNING:The Weather Underground server couldn't fulfill the request and responded with error code 504

Time stuck at 3:47 a.m. (10-04-2018).
P.S. I don't know if it's important but I have changed one line in the pydpiper.service file (/etc/systemd/system) to:
ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper-0.3-alpha:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio.....

dhrone, 2018-04-13:
Looks like a problem on the Weather Underground side.  I'll have to run some tests to see what is going on.

dhrone, 2018-05-25:
Is this problem still occurring?