dhrone/pydPiper #29: Problem with Volumio v2.389

rikardo1979, 2018-05-02:
Trying to make this working on Volumio v2.389 with no luck. 
All I can see on display is the pyPiper start logo and then 'STOP' permanently displayed
[image]

Nothing in log apart from 

`2018-05-02 10:42:42,807:INFO:pydPiper starting
2018-05-02 10:42:46,657:WARNING:Weather Underground API key or location are mis$
2018-05-02 10:47:47,019:WARNING:Weather Underground API key or location are mis$
2018-05-02 10:52:47,512:WARNING:Weather Underground API key or location are mis$
2018-05-02 10:57:47,915:WARNING:Weather Underground API key or location are mis$
`

MikkelLau, 2018-05-03:
Hi there.
Thanks again for this very nice piece of code for different mediaplayers :-) 

I have the same problem running a HD44780 16x2 display. It is displaying song and start/stop logo. The funny things is that the display work even though nothing is changed in the pydpiper.service file. Not any ExecStart line is uncomment. only ExecStop file is in use...
It will not respond to any change in a uncomment line regarding time, wheather, driver, pages etc...
Please let me know if i can do any trouble shooting...
Regards Mikkel

dhrone, 2018-05-14:
Sorry for everyone that's been having trouble.  I've had another project in flight that has eaten all of my development time.  I'll take a look at the installation on the current version of Volumio this week.

W1tch3d, 2018-05-14:
No soucy, it's OK now for me, i had a problem with docker.

Saiyato, 2018-05-14:
This problem comes and goes, I haven't been able to pinpoint it. But most of the times it works for me

dhrone, 2018-05-25:
If you get a chance, try the latest build and see if it resolves your problem. Starting from a fresh install of Volumio 2.389 issue the following commands from within the /home/volumio directory.

`git clone https://github.com/dhrone/pydPiper`
`cd pydPiper`
`sudo ./install_docker.sh`
Make sure that you get a message indicating that Docker has installed successfully
`sudo ./install_volumio.sh`
This will run a new configure program to set up the system for your display and distribution and then create and install a systemd service file for pydPiper.

If the problem continues, shut down the service and then run pydPiper manually adding the --showupdates flag to see if you are receiving updates from Volumio correctly.

So...
`sudo systemctl stop pydpiper`
`sudo systemctl disable pydpiper`

`sudo /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --showupdates`

With the showupdates flag set, every time pydPiper receives new data, it will print a message. 

NOTE: I have only tested this version with Volumio 2.389 and an HD44780 20x4 line display.

Saiyato, 2018-05-26:
I think I found why I experienced the problem, I created a plugin for Volumio (https://github.com/Saiyato/volumio-pydpiper-plugin) and the systemd unit started pydPiper, but when Volumio was started, I restarted the pydPiper service.

What I did now, was remove the systemd unit file, symlink to the one in the plugin directory (so I could write without sudo) and disable the service. In which case Volumio controls the service, plugin disabled = no start, and it only starts after Volumio has been initialized.

MikkelLau, 2018-10-03:
The display is working great and it is easy to install with the new config :)

Is it some how possible to retrieve the 'bitrate' info (kbps). It is not present when using the --showupdates. It is present when running this with MoOde audio but not here in Volumio. kHz and Bit are retrieved from 'tracktype'

I am trying to run a 'conditional' in the sequences section of the pages file. I am trying to read the 'stream' state when playing flac but i can't make it work. Is this all wrong: `{ 'name':'playartist_flac', 'duration':14, 'conditional':"db['stream']=='flac' " },`  I have a canvas called 'play artist_flac'...
Thanks again

dhrone, 2018-10-03:
pydPiper does look for the variable samplerate from Volumio.  If it is not showing up in --showupdates though it is not being sent by Volumio.  You can verify what Volumio is producing by going to the /dev page of the web interface (see https://volumio.github.io/docs/User_Manual/SSH.html).  

For your conditional, you are testing against the wrong variable.  I think you want tracktype.  Something like...

``` 'conditional':"db['tracktype'][:4].lower()=='flac' " ```

Saiyato, 2018-10-03:
Correct, the bitrate is not being sent by Volumio, mpd however does have this info:

```
volumio@volumioboss:~$ telnet localhost 6600
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
OK MPD 0.20.0
status
volume: -1
repeat: 0
random: 0
single: 0
consume: 1
playlist: 716
playlistlength: 1
mixrampdb: 0.000000
state: play
song: 0
songid: 342
time: 52:179
elapsed: 51.547
bitrate: 1047
duration: 178.867
audio: 44100:16:2
OK
```

It's been a while since I dug into this code, but maybe this can be concatenated to the Volumio results that do exist?

/dev gives me the following:

```
{
	"status": "play",
	"position": 11,
	"title": "Path [Vol. 1]",
	"artist": "Apocalyptica",
	"album": "Cult (Special Edition)",
	"albumart": "/albumart?cacheid=307&web=Apocalyptica/Cult%20(Special%20Edition)/extralarge&path=%2FNAS%2FiTunes%2FApocalyptica%2FCult%20(Special%20Edition)&metadata=false",
	"uri": "mnt/NAS/iTunes/Apocalyptica/Cult (Special Edition)/1-01 Path [Vol. 1].m4a",
	"trackType": "m4a",
	"seek": 8577,
	"duration": 186,
	"samplerate": "44.1 KHz",
	"bitdepth": "16 bit",
	"channels": 2,
	"random": null,
	"repeat": null,
	"repeatSingle": false,
	"consume": false,
	"volume": 0,
	"mute": false,
	"stream": "m4a",
	"updatedb": false,
	"volatile": false,
	"service": "mpd"
}
```

I think you should be able to calculate kbps, but easiest would be to ask MPD for the value.

dhrone, 2018-10-03:
If what you are looking for is instantaneous bitrate then I think MPD is your only source.  Unfortunately, it is not currently possible to have two sources active if both are producing information at the same time.  

I have been thinking of extending the metadata subsection of pydPiper but do not know when I'm going to get to it.  This is an area of the code that has been difficult to deal with as each of the distributions handles metadata somewhat differently.  It has needed clean-up for awhile.   FYI, if I do upgrade that subsystem, it will likely require that the page files be modified.

Jozin7, 2021-11-20:
Hello everyone and thank you Saiyato for a great plug.
Unfortunately, after the last volumio update, I cannot install the plugin. During installation I have the following errors in the console:

volumio@volumio:~/pydPiper$ sudo ./install_docker.sh
[sudo] password for volumio: 
Installing Docker from the convenience script

# Executing docker install script, commit: 93d2499759296ac1f9c510605fef85052a2c32be

DEPRECATION WARNING
    This Linux distribution (raspbian jessie) reached end-of-life and is no longer supported by this script.
    No updates or security fixes will be released for this distribution, and users are recommended
    to upgrade to a currently maintained version of raspbian.

Press Ctrl+C now to abort this script, or wait for the installation to continue.
+ sh -c apt-get update -qq >/dev/null
+ sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "cs_CZ.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
E: Sub-process /usr/bin/dpkg returned an error code (1)
Enabling docker

Failed to execute operation: No such file or directory
Failed to start docker.service: Unit docker.service failed to load: No such file or directory.
sudo: docker: command not found
Testing docker

sudo: docker: command not found

Please let me know if it is still possible to install the plugin on the Raspberry. I have a Raspberry PI 3+ with volumio 2.917, LCD HD44780 16x2 installed.

Thank you for your answer and best regards

Jozin7

Saiyato, 2022-02-08:
Hi @Jozin7 ,

Unfortunately this is not something @dhrone or I can fix, I've briefly tried investigating another way to run containers (podman, or crun or containerd), but I haven't been able to make it work on Jessie yet. Your best bet is to upgrade to Volumio 3.1xx and up, that way Docker will install fine again :smile:

Jozin7, 2022-02-27:
Hello @Saiyato and thank you again for the great plugin :-)

After upgrading Volumio to version 3.198, I installed the pydPiper plugin without any problems. I added a GPIO buttons plugin to handle basic commands and my old DSR tuner REVOX starts its second life :-)

I greet you cordially and wish you all the best

Jozin7

[image]
[image]