dhrone/pydPiper #30: Unwanted random characters

hopek1986, 2018-05-13:
Many Thanks for great software, i'm really like it. It;s not so hard to configure when you spend few hours on it 💃  ;) But in my case i have a little problem.  Unwanted  random characters displaying on random positions on a display. Im using a Noritake Itron 2x20 CU20029ECPB-W1J Fluorescent Vaccum Dispaly wich is a HD44780 compatibile. In my case i need to connect a RW pin to the Ground to start display working.  I was tested on a i2c backpack and parallel connection. On i2c connection was worse than in parallel mode. I think that is something wrong with a timings, but i don't know where to start find a problem. I was start to searching and at this moment i found a solution. I've change a "ANIMATION_SMOTTHING" parameter and slows down it to .050 and its better, but not a perfect ;) If i'm slowing down the page refresh rate, the scrooling speed slows down too.

dhrone, 2018-05-14:
Hmm.  If it really is a timing issue, you might try adding some additional delay in the pulseEnable function in lcd_display_driver.py.  Try uncommenting line 77 and changing the delay on lines 77 and 79 to 1 microsecond instead of .1

dhrone, 2018-05-25:
If you get a chance, try the latest build and see if it resolves your problem.  Starting from a fresh install of Volumio 2.389 issue the following commands from within the /home/volumio directory.

`git clone https://github.com/dhrone/pydPiper`
`cd pydPiper`
`sudo ./install_docker.sh`
Make sure that you get a message indicating that Docker has installed successfully
`sudo ./install_volumio.sh`
This will run a new configure program to set up the system for your display and distribution and then create and install a systemd service file for pydPiper.

hopek1986, 2018-05-25:
Thanks for reply! I was start to seek any solution with timings but sometime is better or worse but in both cases far from ideal. Fresh installation dos not help. There is a problem with VFD module. In this display need to check busy flag (RW Pin - as i do this on my simple microcontroller projects) but in this case is connected to ground. I think that lcd driver dose not support this feature. I dont have knownledge to write up this function. It must stay like is.

Joeydd2, 2020-03-17:
I think I am having the same problem: using a 20x2 VFD display, I'm getting random characters. Did you manage to get it to work?
Thanks

hopek1986, 2020-03-17:
Hi. At this moment i leave it as is. The problem is in VFD because Your display is almost compatible with HD44780 driver. I have 2 different VFD's (2x16 - PT6314 and 2x20 - Noritake Itron driver) and at PT6314 everythink is ok. I think the software driver sends data to the display too quickly for this display and VFD cannot support the Enable pin properly or software cannot read bussy flag. The solution may be to add a function to send data e.g. every half second or when the display changes display data.

Joeydd2, 2020-03-18:
Hi, I somehow managed to get the display to work perfectly, although I'm not sure what I did to get it to work, to be quite honest. 
Now I'm just trying to figure out how to change the pages file to customize the information displayed. It's easy to figure out the pages file, but Im unsure how to update the pages for the modifications to take effect.

uncle-benz, 2021-02-21:
Hello @ all
has someone already found the solution? :
"using a 20x2 VFD display, I'm getting random characters"
does someone have a finished file for me for such a "20x2 VFD display"?
does anyone have "pages_lcd_20 x2.py"?
greetings
Rafael

dhrone, 2021-02-22:
There is currently no driver available for the Noritake Itron 2x20 CU20029ECPB-W1J.  If the HD44780 functionality is all that is required, then it is likely just a minor modification of the existing HD44780 driver to make it work.  If you are experienced in Python, you might look to see if their are modifications to the initialization you need to make in order to get the display properly set up.

As to the 20x2 page file, I'd start with one of the 16x2 files and modify based upon your preferences.

hopek1986, 2021-03-01:
To fix this problem driver need to send data to display in longer intervals like 250ms or more but this has impact on scrooling speed.  In this case driver sends data to display to fast and that make unwanted (ghost) characters.

av71966, 2021-10-19:
> To fix this problem driver need to send data to display in longer intervals like 250ms or more but this has impact on scrooling speed. In this case driver sends data to display to fast and that make unwanted (ghost) characters.

Hi! Did you fix blinking charecters?  I have simillar problem....