dhrone/pydPiper #33: Time

superjasiek, 2018-08-07:
Hi! First of all thank you very much for this grate piece of software. Everything goes smooth with dock installer and volumio installer (I use raspberry pi zero w with hifiberry dac + and lcd hd44780 i2c, on adress 3F) . Setting accuweather was a little tricky about location code but it works fine.  There is one thing i can't figure out - how to change clock format - I set time zone to Europe/Warsaw and set true to 24h clock but lcd still display 12h clock.
One more time thank's for your time for developing this soft!

dhrone, 2018-08-09:
Not sure what would cause that exception except possibly a bad pages file.  Which one are you using?  As for the 12 vs 24 hour issue.  The 24h setting is no longer enforced.  That has moved into the pages file.  If you look in the file you are using examine the definition for 'time' in the set of widgets.  You'll notice the variables list has a single value which is 'localtime|strftime+%-I:%M'.  Just change this to 'localtime|strftime+%-H:%M' and you'll get a 24 hour clock.

superjasiek, 2018-08-09:
Thank's for reply - made changes and everything works fine. I think that this exception was made by my code in python for rotary encoders (two scripts for two encoders run by system service) - is it possible? Maybe you have some hints to improve it?
I use page file 16x2 

'#!/usr/bin/env python
import sys
import time
import os
from rotary_class import RotaryEncoder

PIN_A = 13              # Pin 16
PIN_B = 26             # Pin 18
BUTTON = 24     # Pin 22

# This is the event callback routine to handle events
def switch_event(event):
        if event == RotaryEncoder.CLOCKWISE:
              os.system ('volumio volume minus')
              time.sleep(.2)
        elif event == RotaryEncoder.ANTICLOCKWISE:
              os.system ('volumio volume plus')
              time.sleep(.2)
	elif event == RotaryEncoder.BUTTONDOWN:
              os.system ('volumio volume mute')
        #elif event == RotaryEncoder.BUTTONUP:
             # os.system ('volumio volume unmute') 
        return

# Define the switch
rswitch = RotaryEncoder(PIN_A,PIN_B,BUTTON,switch_event)

while True:
        time.sleep(0.5)'
and rotary class looks like this:
`#!/usr/bin/env python

import RPi.GPIO as GPIO

class RotaryEncoder:

	CLOCKWISE=1
	ANTICLOCKWISE=2
	BUTTONDOWN=3
	BUTTONUP=4

	rotary_a = 0
	rotary_b = 0
	rotary_c = 0
	last_state = 0
	direction = 0

	# Initialise rotary encoder object
	def __init__(self,pinA,pinB,button,callback):
		self.pinA = pinA
		self.pinB = pinB
		self.button = button
		self.callback = callback

		GPIO.setmode(GPIO.BCM)
		
		# The following lines enable the internal pull-up resistors
		# on version 2 (latest) boards
		GPIO.setwarnings(False)
		GPIO.setup(self.pinA, GPIO.IN, pull_up_down=GPIO.PUD_UP)
		GPIO.setup(self.pinB, GPIO.IN, pull_up_down=GPIO.PUD_UP)
		GPIO.setup(self.button, GPIO.IN, pull_up_down=GPIO.PUD_UP)

		# For version 1 (old) boards comment out the above four lines
		# and un-comment the following 3 lines
		#GPIO.setup(self.pinA, GPIO.IN)
		#GPIO.setup(self.pinB, GPIO.IN)
		#GPIO.setup(self.button, GPIO.IN)

		# Add event detection to the GPIO inputs
		GPIO.add_event_detect(self.pinA, GPIO.FALLING, callback=self.switch_event)
		GPIO.add_event_detect(self.pinB, GPIO.FALLING, callback=self.switch_event)
		GPIO.add_event_detect(self.button, GPIO.BOTH, callback=self.button_event, bouncetime=200)
		return

	# Call back routine called by switch events
	def switch_event(self,switch):
		if GPIO.input(self.pinA):
			self.rotary_a = 1
		else:
			self.rotary_a = 0

		if GPIO.input(self.pinB):
			self.rotary_b = 1
		else:
			self.rotary_b = 0

		self.rotary_c = self.rotary_a ^ self.rotary_b
		new_state = self.rotary_a * 4 + self.rotary_b * 2 + self.rotary_c * 1
		delta = (new_state - self.last_state) % 4
		self.last_state = new_state
		event = 0

		if delta == 1:
			if self.direction == self.CLOCKWISE:
				# print "Clockwise"
				event = self.direction
			else:
				self.direction = self.CLOCKWISE
		elif delta == 3:
			if self.direction == self.ANTICLOCKWISE:
				# print "Anticlockwise"
				event = self.direction
			else:
				self.direction = self.ANTICLOCKWISE
		if event > 0:
			self.callback(event)
		return

	# Push button up event
	def button_event(self,button):
		if GPIO.input(button): 
			event = self.BUTTONUP 
		else:
			event = self.BUTTONDOWN 
		self.callback(event)
		return

	# Get a switch state
	def getSwitchState(self, switch):
		return  GPIO.input(switch)

`
And last question: is it possible to change weather info provider? for example for openweathermaps or sth? Where is code that is responsible for this part (weather)?

superjasiek, 2019-05-19:
Hi again :) Just updated pydPiper (using instruction on github) and have same issue - how to change time from 12h to 24h? is there any way to force 24h clock? I edited page file - 16x2 'localtime|strftime+%-H:%M' but it does not work. Also question about weather service is it possible to run accuweather like in previous version?
Thanks!