dhrone/pydPiper #35: configuration for ssd1306 i2c oled display

remy1961, 2018-08-22:
I have been using pydPiper with a winstar 16x2 oled display without a problem and i wrote a tutorial on audiophonics forum. What i like about pydPiper is the control it gives over what is displayed and how. Now i am trying to use pydPiper with a [ssd1306 128x32 i2c oled display (0.96inch)](https://abra-electronics.com/opto-illumination/lcds-and-displays/oled/lcd-oled-0.96-0.96-inches-oled-i2c-white-text.html) and a [ssd1306 128x64 i2c oled display (2.42inch)](https://www.ebay.com/itm/I2C-2-42-OLED-128x64-Graphic-OLED-White-Display-Arduino-PIC-Multi-wii/191835417966?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649). Here is the config file for the 128x32 pixel one, i will be gratefull if you could tell me what is wrong in the configuration because i can't have it to work. Are there other changes i need to perform? 

[STARTUP]
startup_msg_duration = 5

[DISPLAY]
display_driver = hd44780_i2c
display_i2c_port = 1
display_i2c_address = 0x3d
display_width = 128
display_height = 32
pagefile = pages_ssd1306.py
animation_smoothing = 0.15
display_enable_duration = 1

[SYSTEM]
logfile = /var/log/pydPiper.log
loglevel = info
timezone = US/Eastern
time24hour = false
temperature = celsius

[SOURCE]
source_type = moode
mpd_server = localhost
mpd_port = 6600
mpd_password = moodeaudio

Thank you.
Remy

dhrone, 2018-08-22:
The display_driver is incorrect.  The SSD1306 uses the luma_i2c driver.

remy1961, 2018-08-23:
Thanks dhrone. The 128x64 display worked without a problem. The 128x32 did not work. I think the pages files were made for the 128x64 displays and need to be edited in order to work with the 128x32.If that's the case, which files do i need to edit?
Remy

dhrone, 2018-08-23:
Which page file are you using?  You may find the pages_lcd_20x4 a good one to start with for testing but the text is going to be tiny on a 0.96" screen.  It would at least allow you to test.  The pages_lcd_20x4 screen was built up using a resolution of 100x32.  FYI, most LCDs using a character structure that is 5x8 so that's why a 20x4 LCD is represented in pydPiper as a 100x32 pixel display.

remy1961, 2018-08-24:
Thank you. I will try that.

remy1961, 2018-09-05:
Worked. Thank you.

closed 2018-09-05 by remy1961