dhrone/pydPiper #37: How to center icons on large screens

remy1961, 2018-09-16:
[image]
Hi dhrone. I am editing my page file for ssd1306 to incorporate icons (play, stop, repeat, shuffle). Is it possible to have this play icon right at the center of the screen? (128x64 oled). I have set the right size and justification center but it is still at the top of the screen. Any idea?
Remy

dhrone, 2018-09-17:
There is currently no vertical justification implemented.  You can move the widgets lower in the Y dimension within the CANVASES section.

```
CANVASES = {
    # ...
    'showplay_canvas': { 'widgets': [ ('showplay',0,8) ], 'size':(100,32) }
}
```

remy1961, 2018-09-19:
Thank you. Will try that this week end.

remy1961, 2018-09-22:
It worked. Creating a canvas did the trick. Thank you very much.
Remy

closed 2018-09-22 by remy1961