dhrone/pydPiper #39: Control 2 displays on the same pi?

remy1961, 2018-09-25:
Hi dhrone. Thanks for this great software. I am getting very good in customizing my page file as you can see here.
[image]
A quick small question: is it possible to use pydPiper to control 2 displays on the same pi?
I have 2 displays using 2 different scripts, would like to control both using pydPiper
[image]
Remy

dhrone, 2018-09-27:
Your system looks very cool.  Yes, you can run multiple instances of pydPiper on a single Pi.  You just need to configure each instance to reference the correct display.  With I2C displays this is as simple as making sure each display is at a different I2C address and setting the pydPiper config file appropriately for each.  With parallel interface displays you will need to find 6 additional free GPIO pins for the second display and set the individual pin assignments in pydPiper appropriately based upon how you wired up the display.  Also, parallel displays hit the GPIO subsystem much harder and you may experience performance issues because of this.  I think a RPI3 should have sufficient horsepower to handle two displays but you'll have to test to confirm.

Another option you might consider is using a dedicated Raspberry Pi Zero for each additional display.  All of the music data sources are exposed as network services so you can have the Pi Zero connect back to music data source running on the main Pi.  This will require you to add a network interface to the PiZero or that you are using a PiZeroW.

remy1961, 2018-09-27:
Hi dhrone. Thank you for the answer. I need a guidance as to how to run 2 instances of pydPiper.
I have a 16x2 oled using parallel interface and a 128x64 oled using i2c interface. Both are working using 2 different script: your old raspdac script for the 16x2 and pydPiper for the 128x64. Now i want to use pydPiper to run both, since it allows easy customisation of the display.
Here are how my config and service files look like:
For the 128x64:
[DISPLAY]
display_driver = luma_i2c
display_i2c_port = 1
display_i2c_address = 0x3C
display_devicetype = ssd1306
display_width = 128
display_height = 64
pagefile = pages_ssd1306.py
animation_smoothing = 0.15

For the 16x2:
[DISPLAY]
display_driver = winstar_weg
display_pin_rs = 7
display_pin_e = 8
display_pin_d4 = 25
display_pin_d5 = 24
display_pin_d6 = 23
display_pin_d7 = 27
display_width = 80
display_height = 16
display_enable_duration = 0.1
pagefile = pages_weh_80x16.py
animation_smoothing = 0.15

pydpiper.service:
Unit]
Description = pydPiper container
Requires = docker.service
After = mpd.service docker.service

[Service]
Restart = always
ExecStart = /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py

[Install]
WantedBy = multi-user.target

Remy

dhrone, 2018-09-27:
You can run distinct instances of pydPiper without problem as long as they do not attempt to control the same GPIO ports or attempt to send data to the same I2C address. 

Basic steps:

1.  Create two installations of pydPiper into two different directories.  Note: docker only needs to get installed once.
```
cd /home/pi
mkdir ssd1306
mkdir winstar_weg
cd ssd1306
git clone https://github.com/dhrone/pydPiper
cd /home/pi/winstar_weg
git clone https://github.com/dhrone/pydPiper
```

2.  Run the configure program in each directory. Answering the questions identically except for the display related questions. 
```
cd /home/pi/ssd1306
sudo python configure.py
cd /home/pi/winstar_weg
sudo python configure.py
```

3.  Create two different services files that have the ExecStarts pointed to each installation.  
```
vi /home/pi/ssd1306/pydPiper/pydpiper.service
vi /home/pi/winstar_weg/pydPiper/pydpiper.service
```
You will need to point the ExecStart line to the appropriate directory for each instance.

In /home/pi/ssd1306/pydPiper/pydpiper.service the ExecStart would be:
`ExecStart = /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/pi/ssd1306/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py`

and in /home/pi/winstar_weg/pydPiper/pydpiper.service the ExecStart would be:
`ExecStart = /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/pi/winstar_weg/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py`

4.  Move the service files into the systemd directory and use systemctl to activate each of them.  Something like...

```
cp /home/pi/ssd1306/pydpiper.service /etc/systemd/system/pydpiper_ssd1306.service
cp /home/pi/winstar_weg/pydpiper.service /etc/systemd/system/pydpiper_winstar_weg.service

sudo systemctl enable pydpiper_ssd1306
sudo systemctl enable pydpiper_winstar_weg
```

That should do it.  Hopefully I got this all right.  I'm typing it from memory without the time to test it.

remy1961, 2018-09-27:
HI dhrone. I will test it and let you know. Thank you very much.
Remy

remy1961, 2018-09-27:
IT WORKED!
pydPiper is a fantastic piece of software. It worked and i am busy customising the 2 displays. I had to correct a few typographic errors in the instructions (pydPiper directory missing here and there in the code), but it worked on the first try. And the 2 displays start exactly at the same time.
Thank you dhrone.

closed 2018-09-28 by remy1961