dhrone/pydPiper #40: Shairport-sync data source (non-Volumio)

davidjroos, 2018-09-27:
Volumio works to display airplay content (not sure exactly how), and pydPiper works with it subsequently.

But volumio has a variety of options, and moOde, max2olay etc users are out in the cold.

Shairplay-sync has the ability to provide metadata:
https://github.com/mikebrady/shairport-sync-metadata-reader

An example project here:
https://github.com/luckydonald/shairport-decoder

Any chance? :)

dhrone, 2018-09-27:
Producing a driver that pulls data from Shairplay-sync would be possible.  That said, there is already support for Mo0de and max2play.  Is the issue specifically metadata for Spotify?

davidjroos, 2018-09-28:
Sure:

Device is [this guy](https://www.audiophonics.fr/en/network-audio-players-raspdac/audiophonics-raspdac-i-sabre-es9038q2m-v2-streamer-raspberry-pi-3-b-p-12664.html)

It's basically a fancy box with a fancy dac and a pretty screen attached to a RPI 3B+

There are a variety sold by audiophonics using the same screen. They provide a prebuilt volumio instance as well as a [GitHub repo](https://github.com/audiophonics/Pydpiper-Raspdac) with details on installing pydPiper and configuring it for the device.

All works as expected, but after ~15 to 45 minutes the screen just goes blank. Restarting the pydPiper service is required to wake it up again, but it's not a software bug (as far as can be determined, I have no way of knowing if this issue can be caught and rectified in software) and the pydPiper service does not crash or log any issue in debug mode.

I contacted audiophonics and they indicated that the RPI (3B+) onboard wifi may causes 'perturbation on display wires' which I'm guessing translates to an electrical level issue that causes the display to zonk out. 

Disabling the onboard wifi adapter (I also disabled bluetooth, just in case) allow the screen to run for (24+ hours so far) without issue. I'm currently testing with a USB wifi dongle to ensure the same issue doesn't crop up that way, but ahve been using USB devices all laong without issue, so I don't expect a similar situation.

Wifi It can be disabled adding this line to /boot/config.txt and rebooting:
`dtoverlay=pi3-disable-wifi`

You can disable bluetooth in the same way with:
`dtoverlay=pi3-disable-bt`

Hope that helps anyone else who might be having a similat issue

davidjroos, 2018-09-28:
The support exists for when moOde and max2play are playing 'their' songs via MPD only tough, right? There is no way for the pydPiper script to show the track details from the moOde Airplay instance (or did I miss that somehow? It doesn't appear to do anything when I try)

Thinking though, an shairport-sync only input wouldn't really help me, as then pydPiper would need to be set to that, and not MPD, meaning that it would only show track details from airplay and not moOde - is that correct?

remy1961, 2018-09-28:
Hi. Looks like we have the same concern. [I asked the question to Tim](http://moodeaudio.org/forum/showthread.php?tid=510), the creator of moode audio.
The link given in your post contains interesting information.
Remy

davidjroos, 2018-09-28:
Right so it has been removed at some point. I've +1'd your thread

dhrone, 2018-09-28:
Under the covers, pydPiper is able to pull information from multiple metadata sources so accessing MPD and shairport-sync data would not be a problem.  I'd just need to code a new source for shairport-sync.  I haven't looked that deeply into Moode though.  Does it in fact use shairport-sync for it's airport support and if yes is it configured to produce metadata?

dhrone, 2018-09-28:
I forgot to mention, Moode can provide metadata by writing out a file every three seconds with the current information.  Unfortunately, the file does not contain the data that would be needed to calculate the elapsed, elapsed_formatted and remaining data fields.  This is why pydPiper currently pulls the data directly from MPD for Moode installations.

davidjroos, 2018-09-28:
Yes I found that option and have enabled, but it doesn't produce anything when airplay is active -  moOde owner indicated he would just set it to say "airplay" in v4.3 coming out (shortly?)

Having pydPiper able to pick up mod through moOde as well as shairport-sync simultaneously would be amazing!

Everything I see indicates Shairport-sync, it's running in the processes and listening on port 5000

```
pi@moode:~ $ shairport-sync -v
Value read for output rate is 44100.
alsa output device name is "default".
Version: "3.1.7-OpenSSL-Avahi-ALSA-stdout-soxr-metadata-sysconfdir:/usr/local/etc"
statistics_requester status is 0.
daemon status is 0.
deamon pid file is "/var/run/shairport-sync/shairport-sync.pid".
rtsp listening port is 5000.
udp base port is 6001.
udp port range is 100.
player name is "Moode".
```