dhrone/pydPiper #41: Tidal streams info

rikardo1979, 2018-10-05:
Not sure if the 'issue' rely in within pydPiper or elsewhere but I have this -

When streming web radio which injects metadata in streams I see message 'WEB RADIO' and second line showing scrolling Track - Album title, so all good.

But when I sending Tidal streams using UPnP (Bubbleupnp) I see 'WEB RADIO' and just Track title in second line. So the information about Album is missing.

This is using the latest moOde distribution 

All the metadata are shown and done correctly in moOde(webgui, system, ..)

Any help or clue where to look please?

dhrone, 2018-10-05:
For Moode distributions, pydPiper gets its data from MPD.  This should work for UPnP as according to Moode,  "its actually proxying tracks into MPD playlist".  If album is not displaying it is because it is not being provided.  You can confirm that by running pydPiper with the --showupdates flag.

rikardo1979, 2018-10-06:
Hmm, not sure what is going on but on display I see only WEB RADIO and track title and switching to play time. Nothing else.

If I SSH to moode and do 

`pi@moode:~ $ mpc current
Fever Ray - If I Had a Heart
pi@moode:~ $ telnet localhost 6600
Trying ::1...
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
OK MPD 0.20.0
currentsong
file: http://192.168.0.3:57645/proxy/tidal/5DA5EF88A2650290687FCD4E52D70681.flac
Album: Fever Ray (Deluxe Edition)
Artist: Fever Ray
Date: 2009-10-19
Disc: 1
Genre: Alternative
Title: If I Had a Heart
Track: 1
Pos: 13
Id: 14
OK
`
[image]

So the moode seems to have all the info needed

MikkelLau, 2018-10-07:
Hi @rikardo1979 
I have experienced the same. Info at web UI and not in --showupdates
I just upgraded to MoOde Audio 4.3 and unfortunately i cannot make the docker run. Have you updated to 4.3? I did a fresh instal since i could't update from the web UI...

rikardo1979, 2018-10-07:
> Hi @rikardo1979
> I have experienced the same. Info at web UI and not in --showupdates
> I just upgraded to MoOde Audio 4.3 and unfortunately i cannot make the docker run. Have you updated to 4.3? I did a fresh instal since i could't update from the web UI...

Yes I've got it working on the latest moOde 4.3  Please have a look into [my thread in our forum](http://moodeaudio.org/forum/showthread.php?tid=550)

dhrone, 2018-10-12:
I found the error in the mpd source code.  Should be fixed now if you want to pull it.

rikardo1979, 2018-10-13:
Perfect work @dhrone All works now. So this can be closed and marked as resolved.
Thank you for your hard work and extended support