dhrone/pydPiper #42: Install on runeaudio

Alien8ions, 2018-10-23:
Hi

I'm trying to get the script to run on the latest runeaudio 0.5b image.
The install instructions are far behind....

I've figured out the install sequence for this version of archlinux

```
pacman -Syy

pacman -S docker

systemctl enable docker

systemctl start docker

echo 'Testing docker\n'

docker run hello-world
```

That part works fine...

Getting the actual pydPiper script to run was fun.

```
python configure.py
cp pydpiper.service /etc/systemd/system/pydpiper.service
systemctl enable pydpiper
systemctl start pydpiper
```
This is where it won't work out of the box... The only edit I did was to get the dhrone\pydpiper:v031... to :latest.

Starting the container into a bash prompt  `docker run --network=host --privileged -ti -v /var/log:/var/log:rw -v `pwd`:/app:rw dhrone/pydpiper:latest /bin/bash`
and running the python gives the following : 

```
root@runeaudio:/app# python pydPiper.py                                                     
pydPiper starting                                                                           
Loading font from /app/displays/fonts/latin1_5x8_fixed.fnt                                  
Loading display controller                                                                  
Loading music controller                                                                    
RUNE musicdata service starting                                                             
Connecting to Rune Redis service on localhost:6379                                          
Connected to Rune Redis service                                                             
Music Controller Starting                                                                   
Loading pages_lcd_16x2.py as page file                                                      
Initializing current system status update process.  System status will update every f       
ive minutes                                                                                 
Loading font from /app/displays/fonts/latin1_5x8_lcd.fnt                                    
Weather service not enabled                                                                 
Weather service not enabled                                                                 
System status: Temp 41°c, Disk space remaining 5%, IP address 192.168.7.170                 
Loading font from /app/displays/fonts/BigFont_10x16_fixed.fnt                               
Loading font from /app/displays/fonts/upperasciiwide_3x5_fixed.fnt                          
Loading image progbar                                                                       
v out of range with value -1.  Should have been between 0 and 100                           
v out of range with value 1.  Should have been between -1 and 0                             
v out of range with value 2.  Should have been between -1 and 0                             
v out of range with value 3.  Should have been between -1 and 0                             
v out of range with value 4.  Should have been between -1 and 0                             
v out of range with value 5.  Should have been between -1 and 0                             
v out of range with value 6.  Should have been between -1 and 0                             
v out of range with value 7.  Should have been between -1 and 0                             
v out of range with value 8.  Should have been between -1 and 0                             
v out of range with value 9.  Should have been between -1 and 0                             
v out of range with value 10.  Should have been between -1 and 0                            
v out of range with value 11.  Should have been between -1 and 0                            
v out of range with value 12.  Should have been between -1 and 0                            
v out of range with value 13.  Should have been between -1 and`

```

No idea where to go from here...

Btw I've took the latest version from github

Hope to get this sorted as the current volumio version doesn't run stably for a long time, rune can handle to be up much longer it seems...

Diego

dhrone, 2018-10-24:
Can you provide additional detail on what the problem is?  It looks like you've successfully started the application.  The out of range message can usually be ignored but it is interesting that the range of one of the values is showing -1 to 0.

Alien8ions, 2018-10-24:
I was just firing things up, to see if it can actually run.

I do have debug mode on...
I haven't got a clue what v is used for... Nor if I should do anything about it.

I've taken the most recent versions, which probably omit the changes I had to make earlier to get the display ungarbled.

If the script runs fine on Rune, I have to see if I can include backlight support again.

So it would be fine as is for you ?
Diego

Maybe the install notes above are useful for rune users...

closed 2018-11-01 by Alien8ions