dhrone/pydPiper #44: new myvolumio Tidal and Qobuz streaming options not displayed when played

cortegedusage, 2018-11-20:
Hi dhrone,

first I have to say you did a very good job coding this piece of SW. 
It is running 24/7 on my volumio system and gives it a real high end look.

but now I encountared a slight flaw. In the new Volumio release, which includes
myvolumio which is a way to stream both Tidal and Qobuz Native from within
the UI of Volumio. 
It won't display Tidal or Qobuz as one would expect, sometimes it displays MPD
but most of the time it says nothing.

is this btw resolved by issue #41? if that is the case I'll try and update pydpiper

thanks!

dhrone, 2018-11-30:
I'll have to pull the latest version of Volumio.  It's likely they have made changes to their metadata to support the new streams.

cortegedusage, 2018-12-10:
both Tidal and Qobuz are displaying mpd from time to time, which makes me think they use mpd as backend to play all stuff. 
that said, I decided to move away from myvolumio and qobuz so for me it is no issue anymore.
But I'll leave it open as it is still there.

thanks
Dennis