dhrone/pydPiper #45: Help on what I have done wrong on the install & setup.

bibmuseli, 2018-11-29:
I get this on my log;

2018-11-29 15:24:09,976:INFO:pydPiper starting
2018-11-29 15:24:11,254:ERROR:Uncaught exception
Traceback (most recent call last):
  File "/app/pydPiper.py", line 615, in <module>
    lcd = displays.hd44780_i2c.hd44780_i2c(rows, cols, i2c_address, i2c_port)
  File "/app/displays/hd44780_i2c.py", line 134, in __init__
    self.write4bits(0x33,False)
  File "/app/displays/hd44780_i2c.py", line 160, in write4bits
    self.bus.write_byte(self.i2c_addr, bits_high)
IOError: [Errno 121] Remote I/O error

this is my  execstart from pypdpiper.service, which I ran as ;

volumio@volumio:/etc/systemd$ sudo systemctl start pydpiper.service

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780_i2c  --width 100 --height 32 --i2caddress 0x37 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_20x4.py

I'm running Volumio 2.502 on a pi zerow with a PhatDAC.

regards.

dhrone, 2018-11-30:
This looks like a misconfigured display.  What type of display do you have?

bibmuseli, 2018-11-30:
Thank you for the reply, it was this display and Ic 2 controller.

https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.co.uk%2Fulk%2Fitm%2F272626779592
Regards
Bob Moseley

bibmuseli, 2018-12-01:
here's what i have;

volumio@volumio:/$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 3f
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

bibmuseli, 2018-12-01:
from the pydpier.service file;

# HD44780 LCD 20x4 display with I2C interface on address 0x27
ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780_i2c  --width 100 --height 32 --i2caddress 0x37 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_20x4.py

ignore the comment

bibmuseli, 2018-12-01:
one other thing i downloaded and installed a couple of demo programs from the Raspberry Pi guy and installed them and they work  fine. I had to change the hex adress to 0x37 in them before running and they gave the predicted displays.

bibmuseli, 2018-12-01:
[image]
[image]

bibmuseli, 2018-12-01:
OK I have sorted it out, it was the wrong hex address, I kept putting in 0x37 and it should have been 0x3f.

Where to I re-configure the information to display I  want to swap a few things around. and also update the time to UK GMT?

dhrone, 2018-12-03:
That's done within the page file.  You can get some idea of how to do this by referring to the various page files that are included and the info in the docs folder.

bibmuseli, 2018-12-03:
ok thanks for that.

closed 2018-12-03 by bibmuseli