dhrone/pydPiper #46: Newhaven OLED random characters

PMu1, 2018-12-02:
Hi, 

I have been trying to get pydPiper to work with a 16x2 OLED display, specifically this one: https://www.adafruit.com/product/823 , but I cannot seem to get the display to work. 

I have it connected with 4-bit parallel to the RPi3B+, but if I configure pydPiper with HD44780 I get random characters on the display, and with winter_weg I don't get anything. 
I tried to play with the pulseEnable time, but still got nowhere. 

Is there a specific configuration I'm missing with this type of OLED? 

Thanks

dhrone, 2018-12-03:
Have you gotten any other sample code to work?  Usually when displays don't work correctly its a wiring problem.  FYI, the adafruit display appears to be the WEH001602a which is definitely supported.  I'm not sure about the Newhaven.

dhrone, 2018-12-03:
Yes, it does appear to be compatible.  Can you confirm whether you've gotten the display to work with an alternate program?

PMu1, 2018-12-10:
I just tried a basic python script to print "Hello World" on it using the RPLCD library and it works.

dhrone, 2018-12-11:
Well that rules out wiring problems.  

I took a quick look at RPLCD and it looks like it is purely HD44780.  Why don't you try running with the hd44780 driver to see if that works.  FYI, you can run the display drivers as stand alone python programs for testing. 

`python hd44780.py`

Normally you would need to supply all of the pin assignments on the command line but you've used the same values I have for defaults.   If you did need to enter them, it would look like this.

`python hd44780.py -r 80 -c 16 --rs 7 --e 8 --d4 25 --d5 24 --d6 23 --d7 27`

PMu1, 2018-12-11:
I tried setting it up as hd44780 previously, but I would only see “pydPiper starting” (or something similar) for about 1 second and then random characters over all the screen.