dhrone/pydPiper #49: artist, album and title not shown with Tidal via bubbleupnp server

Dual01, 2018-12-13:
I installed pydPiper and use it with a Winstar LCD screen. It works very nice, thanks to dhrone, you created a great piece of software. On the raspberry I run the mediaplayer software by Pete Manchester, which is a Java-based openhome renderer which uses mpd for playback. As control point, I use LINN Kazoo or the LUMIN software running on an ipad. Everthing works perfect when I stream music from my Music Server. In order to have tidal access, I run the bubbleupnp server software on the pi, which provides tidal access for the mediaplayer.  When I use tidal in this setting, no artist, album and title information is available although Kazoo or Lumin shows all the information
Do you have any idea what could be the reason?

Dual01, 2018-12-13:
yes, I use the MPD source driver. 
When I use the --showupdates  parameter when I run the program, I get the message db[encoding]'webradio'. Is that normal?

Dual01, 2018-12-16:
interestingly, sometimes it shows title and artist but no album with tidal via bubbleupnp server and sometimes all three items are missing. I have no idea what could be the reason for this differences.

dhrone, 2018-12-31:
db[encoding] will equal 'webradio' whenever the file that MPD is playing is specified as a webserver address.  In other words if the current file starts with 'http' the driver assumes that the source is a web server that is streaming the contents to the player.