dhrone/pydPiper #50: Wish to talk to another microcontroller - where to start ?

PaoloBi, 2018-12-31:
Hi, I am developing a STM32 custom display for industrial purposes, and I'd like to use it with Volumio. To do this i'd like to send raw infos on tune played, (possibly) sample rate do my processor, and to get back commands like Play, Stop etc.
I can invest some time in modifying pydPiper sources in order to do that, some infos/initial help would be welcome
Thank you
Paolo

dhrone, 2018-12-31:
I'm not completely clear on what you are trying to accomplish.  Do you want to run a program on the STM32 microcontroller that would drive the contents of an LCD or OLED display?  Would Volumio be running on a separate Raspberry Pi?  If yes to both, as long as the STM32 has the resources to run pydPiper, it should be pretty straight forward to get your use case to work.   A systems diagram for what you are proposing would be very helpful in understanding how to guide you.

Cheers,
Dhrone

PaoloBi, 2019-01-01:
Hi, yes I want to develop my own GUI on my STM32 micro controller (+ display + touchscreen). Volumio is running on a raspberry pi3. I would like to modify/run pydPiper only on Raspberry, and to open a serial communication channel to my STM32 (implementing my own protocol based e.g. AT commands, or Modbus). Alternatively I could talk to my STM32 with bidirectional I2C. So, to begin, I'd like to have some details on "where to put my hands" into pydPiper.
My GUI will show tracks/Stream details, and will send commands to Volumio (start, stop etc).
Cheers,
Paolo

dhrone, 2019-01-02:
A large part of the functionality of pydPiper is the rendering of the screen that is output on the display.  If you are going to implement that functionality yourself, the only sections of the project you may be interested in are the source modules and the display drivers.  These are in the source and the display folders.

dhrone, 2019-01-26:
Closing.  Please re-open if you need further assistance.

closed 2019-01-26 by dhrone