dhrone/pydPiper #51: Support for UTF8 characters

dikodahan, 2019-01-04:
Hey, first off, AWESOME plugin, LOVE IT!
really makes a difference with a rotary encoder added as well to have almost full local control on the device.

I did notice one thing, when I play songs where the name of the song is not in a standard language, in my case Hebrew, it only shows a bunch of ??????? instead of the characters. Is there any chance to provide support for either additional languages or just use a universal character code, maybe UTF8 since I know most use that, but not really sure.

Thanks!!!

dhrone, 2019-01-05:
What display are you using?  If it is one of the character-based display (e.g. HD44780) there is only limited support for alternate character sets.  With HD44780 based devices, pydPiper can create up to eight special characters.  It does this on the fly as the screen is being rendered so you can use fonts with characters not included in your displays character data ROM as long as you do not attempt to display more than eight unique custom characters on the same screen.  If you are using a graphical display like an SSD1306 or a winstar WEH/WEG display, then the eight character limitation goes away.  

For HD44780 (and the Winstar WEH/WEG) you also need to be aware of their low resolution.  Each character has to be rendered inside a 5x8 pixel grid.  This creates some significant challenges in font design!

Take a look at the font image files at pydPiper/displays/fonts/*.png.  They contain the bitmap representation of the fonts that pydPiper uses.  If you can find the characters you need in one of those files, chances are we can get it to work.  I can also walk you through how to add custom characters to a font file if you want to create your own characters.

It's probably worth mentioning that pydPiper supports Trutype fonts so if you are using a graphical display and can find a font that has the character-set you want to use, you'd be set.  This is really only useful for the SSD11xx style displays though as the other displays pydPiper supports do not have sufficient resolution to display Trutype fonts intelligibly.

dhrone, 2019-01-26:
Closing this issue.  Please re-open if you need further help to get your Hebrew content displayed.

closed 2019-01-26 by dhrone