dhrone/pydPiper #52: font is ？？？？？

arthurj58, 2019-01-18:
HI
I need help
Asian font display is ???

[http://www.bamboogames.net/webdir/moode/IMG_6650.JPG](url)

WAV file 
[http://www.bamboogames.net/webdir/moode/%E5%A4%A7%E3%81%8D%E3%81%AA%E5%8F%A4%E6%99%82%E8%A8%88.wav](url)

my pydPiper.cfg
[STARTUP]
startup_msg_duration = 5

[DISPLAY]
display_driver = luma_i2c
display_i2c_port = 1
display_i2c_address = 0x3C
display_devicetype = sh1106
display_width = 128
display_height = 64
pagefile = pages_ssd1306.py
animation_smoothing = 0.15

[SYSTEM]
logfile = /var/log/pydPiper.log
loglevel = info
timezone = Asia/Taipei
time24hour = true
temperature = celsius

[SOURCE]
source_type = lms
lms_server = 192.168.0.28
lms_port = 9090
lms_player = b8:27:eb:19:7d:1c

I also tried to replace the font 
[http://www.bamboogames.net/webdir/moode/DejaVuSansYuanTiMono-Regular.ttf](url)
to /usr/share/fonts/truetype/dejavu/DejaVuSans.ttf
Still invalid

Thanks

dhrone, 2019-01-22:
You're heading in the right direction but still have work to do.  You do need to add the DejaVuSansTuanTiMono-Regular.ttf font into the page file.  You do that by adding a lines within the TRUTYPE font section...

```
TRUETYPE_FONTS = {
	'DejaVuSans28': { 'file':'DejaVuSans.ttf', 'size':28 }
	'DejaVuSansYTM8': { 'file':'DejaVuSans.ttf', 'size':8 }
	'DejaVuSansYTM10': { 'file':'DejaVuSans.ttf', 'size':10 }
}
```
Note: You need a definition for each size of the font that you need to support.

You then need to change the widgets that will contain the non-Latin characters to use the new font that you have defined.

`WIDGETS = {
	'artist': { 'type':'ttext', 'format':'{0}', 'variables':['artist'], 'font':'DejaVuSansYTM10','varwidth':True,'effect':('scroll','left',1,1,20,'onloop',3,125)},
	'title': { 'type':'ttext', 'format':'{0}', 'variables':['title'], 'font':'DejaVuSansYTM8','varwidth':True,'effect':('scroll','left',1,1,20,'onloop',3,125)}
}
`
Note: Trutype fonts use a different widget type (ttext). So if you are changing from using one of the built-in fonts to a Trutype font, make sure to change 'type':'text' to 'type':'ttext'.

Last item to mention: low resolution displays make it difficult to legibly display very small trutype fonts.  You may need to increase the font size of the widget you replace which will have an impact on the widgets overall dimensions.  This will likely require a redesign of the layout of the page file.

arthurj58, 2019-01-24:
Hi dhrone
thank you very much Fonts display correctly
I want to buy this NHD-2.8-25664 OLED

http://www.newhavendisplay.com/specs/NHD-2.8-25664UCY2.pdf

I don't know if it supports

Thanks again

dhrone, 2019-01-24:
I haven't tested it but it should work.  Choose luma_i2c as the display type and ssd1322 as the type of display.

dhrone, 2019-01-26:
Looks like this is resolved.  Re-open if you are still having problems.

closed 2019-01-26 by dhrone