dhrone/pydPiper #53: i2c Display with mcp23008 wrong pins

PRDGYSA, 2019-02-06:
Hi there,
I have a i2c backpack from Adafruit that seems to have a slightly different pinout to what is written in your code as i can't get it for the life of me to display any characters but i can't find the pin definitions within the code. attached is a circuit schematic that has the pinout of the backpack.

kind regards
Tiaan Giliomee
[image]

dhrone, 2019-02-06:
Hmm.  pydPiper does support i2c displays.  Do you have i2c enabled correctly?  What address does your display show up as?  (e.g. what does i2cdetect report)?  

Take a look at https://learn.adafruit.com/adafruits-raspberry-pi-lesson-4-gpio-setup/configuring-i2c

PRDGYSA, 2019-02-06:
As I ran the script for pydPiper the sreens backlight started flickering at regular intervals with just a series of blocks on the screens, this led me to believe that my i2c board might be malfunctioning. I then proceeded to use a quick test script from the Adafruit libraries that is supposedly written for the backpackfor which I got the same results, backlight still flickers. After editing the pins that correspond to the above attached pcb diagram I had success with the messages being displayed clear as daylight which lead me to belive that there is possibly a fault with the pin mapping for this specific backpack within the code. 

this is the backpack in question.
https://www.adafruit.com/product/292

dhrone, 2019-02-07:
I'm not sure what pins you are referring to.  When using an i2c device interfaced with a Pi, you will always be connected to pins 3 and 5.  Can you share the code you used for the test that worked?

PRDGYSA, 2019-02-07:
This is the code that has worked, the i2c from the mcp23008 is connected to those pins, its the pins on the mcp23008 to the display that need to be mapped. 

import Adafruit_CharLCD as LCD
import Adafruit_GPIO.MCP230xx as MCP

GPIO = MCP.MCP23008(address=0x27, busnum=1)

# Define PCF pins connected to the LCD.
lcd_rs        = 1
lcd_en        = 2
d4,d5,d6,d7   = 3,4,5,6
lcd_backlight     = 7
cols,lines    = 16,2

# Instantiate LCD Display
lcd = LCD.Adafruit_CharLCD(lcd_rs, lcd_en, d4, d5, d6, d7,
                       cols, lines, lcd_backlight, gpio=GPIO)

lcd.set_backlight(1.0)
lcd.clear()
lcd.message('  Raspberry Pi\nI2C LCD MCP23008')

PRDGYSA, 2019-02-08:
If it helps at all.

This library works perfectly by using the simple initialize for the mcp23008.

https://github.com/dbrgn/RPLCD/tree/master/RPLCD

dhrone, 2019-02-10:
The problem is that the MCP23008 uses different pin assignments than a standard I2C hd44780 display.  Using the code you provided as an example, I have created a new driver specific to the MCP23008 backpack on an HD44780 LCD.  Unfortunately, I do not have an MCP23008 so I am not in a position to test the new driver.  Pull the changes from the repo and try to run the driver standalone.

`python hd44780_mcp23008.py -r 16 -c 80`

PRDGYSA, 2019-02-11:
Just updated libraries, etc. ran the code and still getting a flashing backlight with blocks on the screen.
the frequency seems to shift as it runs through the code so the mapping isn't quite right

dhrone, 2019-02-11:
Not sure I can take it much further as I lack the hardware to test the solution.

PRDGYSA, 2019-02-11:
I'm quite new to python as i mainly code in Java, would it help if we scheduled a time to try various solutions?

dhrone, 2019-03-11:
The code for the driver is reasonably well contained so feel free to experiment.  It can be run separate from the rest of the pydPiper code for testing.

oisteink, 2020-01-24:
I've made a driver using RPLCD, but it's horribly ugly as I know next to nothing about python, and I've not coded much in years.

I had to edit the docker image to add the RPLCD library (I could not manage to re-build the image as the base is gone).

I'll update my fork with the code once I read up on how to do that...

oisteink, 2020-01-30:
I ended up fixing the mcp23008 driver instead. At least it works for me.