dhrone/pydPiper #54: Last version of Rune 0.5

RP250258, 2019-02-09:
Can you inform me how to install this for rune audio 0.5

RP250258, 2019-02-09:
cd /home/rune
git clone https://github.com/dhrone/pydPiper.git
cd pydPiper
./install_docker.sh

wget https://raw.githubusercontent.com/audiophonics/Pydpiper-Raspdac/master/pydPiper.cfg
wget https://raw.githubusercontent.com/audiophonics/Pydpiper-

Raspdac/master/pages_raspdac_16x2.py
You can test display with this command :
Take care to replace directory with the one where PydPiper is located

sudo docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py
Service Installation :

wget https://raw.githubusercontent.com/audiophonics/Pydpiper-Raspdac/master/pydpiper.service
Then activate service :

sudo cp pydpiper.service /etc/systemd/system
sudo systemctl enable pydpiper
sudo systemctl start pydpiper

dhrone, 2019-02-10:
Try the following using a fresh install of Rune v0.5

Login in to rune as root
username: root
password: rune

Enter the following commands...
git clone https://github.com/dhrone/pydPiper
cd pydPiper
chmod 755 rune_version5.0_install.sh
./rune_version5.0_install.sh

RP250258, 2019-02-10:
Hi drone,

thanks will try that

git clone https://github.com/dhrone/pydPiper
cd pydPiper
chmod 755 rune_version5.0_install.sh
./rune_version5.0_install.sh

only this or must i add other script?

now i want to use the winstar oled WEH001602BWPP5N00000

thanks

dhrone, 2019-02-11:
Yes, those commands should be sufficient.  The winstar weh is supported.  Just choose winstar as your display type.  I'd recommend pages_weh_80x16.py as your page file.

RP250258, 2019-02-12:
hi Drone,

do you mean

sudo python configure.py

And set all as followed:
Code:
Startup mode duration (in seconds)? [5]: 5 

DISPLAY CONFIGURATION 
Display type ?: winstar_weg  ******** winstar_weh?
Register select pin? [7]: 7 
Enable pine? [8]: 8 
Data 4 pin? [25]: 25 
Data 5 pin? [24]: 24 
Data 6 pin? [23]: 23 
Data 7 pin? [27]: 27 
Width of display (in pixels)? [80]: 80 
Height of display (in pixels)? [16]: 16 
Enable pulse duration (in microseconds)? [0.1]: 0.1 
Location of the pagefile? [pages_lcd_16x2.py]: pages_raspdac_16x2.py
Animation Smoothing (in seconds)? [0.15]: 0.15 

SYSTEM CONFIGURATION 
Location of log file? [/var/log/pydPiper.log]: /var/log/pydPiper.log 
Logging Level? [info]: info 
Time Zone? [Europe/London]: Europe/London 
24-hour clock? [True]: true 
Temperature Scale? [fahrenheit]: celsius 
Enable weather? [n]: n 

MUSIC DISTRIBUTION 
Name of distribution? [moode]: moode 
Server address? [localhost]: localhost 
Port? [3000]: 6600
Password:

change this !!

RP250258, 2019-02-12:
i cannot change winstar_weg into winstar_weh

RP250258, 2019-02-12:
I have connected the WH1602L the same as WEH1602A but the L does not startup
i use the raspberry setup from audiophonics with moode audio

RP250258, 2019-02-14:
Hi Drone,

Did you already had the time to see why the L display does not work.

Many thanks.....

RP250258, 2019-02-14:
Drone,

This is the datasheet
https://www.winstar.com.tw/products/oled-module/oled-character-display/oled-16x2-character.html

RP250258, 2019-02-14:
I cannot see any difference between the both oled displays. I won't start up i have check the wires several times.....

RP250258, 2019-02-21:
Sorry i make an mistake.
The oled large is an oled
WEH001602BL, im using it with moode audio at this moment, but the display stays blank off
Could you please help me with this..

mihneanu, 2020-02-17:
Hello RP250258! Did you eventually solve the issues? 
I'm struggling to set up the Display on Rune 0.5 but I am constantly blocked by the missing Docker thing...
If you managed to have it work on your Rune, or if you find another working solution, please let me know.
Thank you,
Mihnea

RP250258, 2020-02-17:
Mihnea, yes for 0.5

mihneanu, 2020-02-17:
That is really good news! Can you please give me some details? I got stuck on the Docker installation/enabling part. Thank you.

mihneanu, 2020-02-19:
Hello RP250258! Sorry for bothering you again. I made it work eventually, on a fresh Rune install. I guess the issue was that I first installed the power management pack and I guess that one messed something up. 
One more question. Do you know if there is a way to restart the service without restarting Rune? It would be useful when iterating on the page file.
Thank you,
Mihnea