dhrone/pydPiper #55: Big clock edit and customize

elpaul, 2019-03-11:
Hello everybody.
dhrone - good job!
I changed the time zone and clock format for 24 hours. It works perfectly on Volumio 2.555
What do I have to do to remove "AM/PM" text behind the clock?
And the second question, in which file can I edit the appearance of a large clock (font, etc.)?
BR

dhrone, 2019-03-23:
Remove the ampm widget from the stoptime and stoptimeweather canvases.

So change...
`
     'stoptime':        { 'widgets': [ ('time',10,8), ('ampm',75,8) ], 'size':(100,32) },
     'stoptimeweather': { 'widgets': [ ('time',0,0), ('ampm',65,0), ('temphilow',75,0), ('conditions2',0,24), ('temp',75,24) ], 'size':(100,32) },
`

to...
`
     'stoptime':        { 'widgets': [ ('time',10,8) ], 'size':(100,32) },
     'stoptimeweather': { 'widgets': [ ('time',0,0), ('temphilow',75,0), ('conditions2',0,24), ('temp',75,24) ], 'size':(100,32) },
`

elpaul, 2019-03-23:
Thank You so much!

[image]
Solved.

dhrone, 2019-03-23:
Glad to help.  What display are you using?  It looks interesting.  Could you recommend a part # and supplier?

elpaul, 2019-03-23:
Winstar WEH002004BL I bought in Poland at a local store for the equivalent 25 $.
At the beginning I wanted to run it on the winstar_weg driver, but the display did not work at all.
 But working good with i2c converter + HD44780i2c driver.
One problem is initialization after soft reboot. I have to turn off power each time.
In the manual I found information that I should send 5 times 0x00 with delay of 100us and omit the busy flag, but I do not know how to do it. (I'm not a programmer.)

dhrone, 2019-03-23:
It's interesting that the winstar_weg driver did not work for you.  I just double checked that which driver chipset the WEH002004BL uses and it is identical to the WEH001602 (WS0010).  The winstar_weg driver has more features than the HD44780 especially with custom characters.  Which i2c converter are you using?

elpaul, 2019-03-23:
Exactly, exactly. I bought it for it. It puzzles me too. I spent two weekends over this oled, but without success.  WEH2004 has two WS0010 controllers, maybe this is the reason?
In the end I took the i2c converter from the old LCD, soldered and ... the display runed right away,( but as 44780).
The converter is on PCF8574 [link](https://allegro.pl/oferta/konwerter-lcd-hd44780-i2c-iic-twi-arduino-avr-arm-5900520984?snapshot=MjAxOS0wMy0xNlQxNDowNzoyOS42NDNaO2J1eWVyOzgwZGM0ZDc4MWQ3M2NiYzE5YTYzZmNiMzI3ZjlkM2QzZjljNGVlMzQ0NThiNGI1OGU1MDNkZmIzNDEzNzUyMTc%3D)

elpaul, 2019-03-24:
Which software do you use to edit the font bitmap file?

closed 2019-03-29 by elpaul

Prfacia, 2019-04-28:
Hi,
how can I change the time zone and clock format for 24 hours?

hcanning87, 2024-04-28:
Hi @elpaul , I know this thread is ancient, sorry! I am having exactly the same problem as you did where the OLED display works perfectly until you soft reboot or restart the pydPiper service - a hard power off/on is required to get it to initialise properly.  I have tried getting it to send 0x00 5 times in the hd44780 driver file but this just results in a blank display.  Did you ever get to the bottom of it?