dhrone/pydPiper #57: Creating custom characterc for LCD

elpaul, 2019-03-21:
I am asking for instructions on how to create and where to save the definition of your own custom character for the LCD.
Or how to make Polish letters: "ą, ć, ó, ż, etc." be displayed as "a, c, o, l, z"?
Now, where there is a letter I have a sign "?".
I'm using 20x4 HD44780

dhrone, 2019-03-23:
Good effort but the font system has evolved over time and the file you are editing does not directly get used any longer. 

I'm assuming given that you are using the HD44780 driver with a 20x4 display that you are using the pages_lcd_20x4.py page file.  That page file uses three fonts (latin1_5x8_lcd.fnt, BigFont_10x16_fixed.fnt, and upperasciiwide_3x5_fixed.fnt).  We could add the polish characters to any of these fonts but the only one that is likely to matter is the latin1_5x8_lcd as this is the font used for all of the song metadata.

There are two files that make up a normal font in pydPiper.  These are the .fnt file and the .png file.  The .png file contains a bitmap that contains glyphs for every character contained within the font.  The .fnt file provides the metadata for each character so that pydPiper knows what part of the bitmap to use to extract the glyph for the character from the bitmap and the size of the character's glyph.  Each character in the *.fnt file is described using the following format.

`char id=99 x=18 y=48 width=5 height=8 xoffset=0 yoffset=0 xadvance=5 page=0 chnl=0`

char id is the unicode value in decimal of the character that is being added.  In this case unicode 99 is the 'c' character.
x, and y are the coordinates within the .png file where the character's glyph is located
width and height are the size of the glyph in pixels
The rest of the values can be ignored (for the purposes of this issue) though xadvance should equal width in most cases.

So to add a new character to the font, you need to add a glyph to the .png file and add a character to the .fnt file. Unfortunately, drawing characters that look acceptable in a 5x8 pixel grid is not that easy so if all you'd like to do is replace the correct orthography with a similar character that lacks the diacritics, you can copy the line for the character you want to leverage and just change the char id to the appropriate unicode value.  As example, if you want to replace the Ą with an A...

The 'A' character is added to the latin1_5x8_lcd.fnt file using the following line:
'char id=66 x=12 y=32 width=5 height=8 xoffset=0 yoffset=0 xadvance=5 page=0 chnl=0'

To use the 'A' as the 'Ą' add the following line to the latin1_5x8_lcd.fnt file.
'char id=260 x=12 y=32 width=5 height=8 xoffset=0 yoffset=0 xadvance=5 page=0 chnl=0'

Note: The unicode value for Ą is 0x0104.  However char id's are decimal values so the char id for Ą is 260.

Once you have added the characters that you want to the .fnt file, you also need to correct the chars count on the third line of the .fnt file.  The chars count value should equal the total number of char definitions contained within the file.

I've added a new font file named latin1_5x8_lcd_polish.fnt to the fonts directory.  This file uses the closest ascii character to replace all of the Polish orthography characters.  To give it a try pull the latest copy of the pydpiper from the repo and modify the FONTS definition for the small font in the pages file you are using.  E.g. where you see latin1_5x8_lcd.fnt, replace with latin1_5x8_lcd_polish.fnt.

elpaul, 2019-03-23:
dhrone... You are my Guru!
I have no words to thank you!
[image]
That's exactly what I expected.
Now I have instructions, so I have time to create my own characters.
Thank you again.
Best regards!

elpaul, 2019-03-24:
Full success! Thank you again for the instruction!
[image]

dhrone, 2019-03-24:
You work fast.  Congratulations.  You seem to have a talent for small resolution font editing.  Would you be willing to provide your updated font back to the repo?  It may be useful for other pydPiper users.  

Could you also confirm that you are using the hd44780_i2c.py driver?  I've been meaning to refactor the display driver code to allow more display types and i2c backpacks to work.  I could probably give you a rough hack of the winstar_weg code pretty quickly while I work on a more elegant solution for all of the current display types.

elpaul, 2019-03-24:
Today, I made only a few letters. When I do all I will send.
Yes, I am using the hd44780_i2c.py driver for the WEH002004BL display (initialize problem after soft restart). The converter (very popular & low cost) is on the PCF8574 chip.
[image]

dhrone, 2019-03-24:
There is one issue that you may want to be aware of when using the hd44780 driver with custom characters. As you are aware, the hd44780 controller supports the creation of custom characters.  pydPiper leverages this feature to auto-create new characters as needed to render whatever screen is requested.  However, the hd44780 only supports a maximum of 8 custom characters at a time.  If you try to display a screen with more than 8 custom characters, the remaining custom characters will be displayed as a '?'.

elpaul, 2019-03-24:
> only supports a maximum of 8 custom characters at a time. If you try to display a screen with more than 8 custom characters, the remaining custom characters will be displayed as a '?  

That problem has just arisen. Look at the 4th line: [video](https://youtu.be/m5r7uIktmQc)

> If I can get the winstar_weg driver to support the PCF8574/I2C backpack, that limitation goes  away. 

This is very good news. I will be waiting impatiently.

dhrone, 2020-09-15:
I've been working with the Luma.OLED project to move all of my driver code over to them.  We've got the ws0010 driver supported and there is a pcf8574 interface available.  I do not have a pcf8574 available to wire to a winstar display.  If you'd be interested trying.  Let me know and I'll send some instructions.

WayneTho, 2023-12-01:
@dhrone , did you ever find a way to overcome the 8 character limitation on the hd44780_i2c displays that use the pcf8574 chip?

If so, would you mind pointing me (us) in the right direction?

dhrone, 2023-12-03:
There is no way to get beyond the 8 character limit on any of the hd44780 devices that I am aware of.  The pcf8574 interface is not relevant to that limitation.  It exists regardless of which interface style you are using.

closed 2023-12-03 by dhrone