dhrone/pydPiper #58: Weather response language

elpaul, 2019-03-27:
accuweather.com allows you to set the response in a specific language:
`example: http://api.accuweather.com/currentconditions/v1/335315.json?apikey={your key}&language=es&details=true`
Where and how to put it in the code?

dhrone, 2019-03-28:
The lines you'd need to play with are at 353 and 354 of pydPiper.py.

elpaul, 2019-03-28:
> querystr = 'http://dataservice.accuweather.com/forecasts/v1/daily/1day/' + pydPiper_config.WEATHER_LOCATION
>                 r = requests.get(querystr, { 'apikey': pydPiper_config.WEATHER_API,  }), &language=pl

It's will be correct?

dhrone, 2019-03-28:
Close.  It's probably more like...

```
querystr = 'http://dataservice.accuweather.com/forecasts/v1/daily/1day/' + pydPiper_config.WEATHER_LOCATION
r = requests.get(querystr, { 'apikey': pydPiper_config.WEATHER_API, , language=es-es })
```

You'll want to look up the appropriate localization code from https://developer.accuweather.com/localizations-by-language.  In this example I've used Spain.

elpaul, 2019-03-29:
Unfortunately, after adding this parameter, the display is permanently dark.

closed 2019-04-29 by elpaul