dhrone/pydPiper #59: strange characters after installation

slawek125, 2019-03-27:
I have the following problem, after installing the plugin the display shows strange characters. These characters are movable. It looks like the plug works only strange characters are shown.
display is hd44780 connection via gpio. Volumio 2 system.
thank you for your help because I do not know what to do about it anymore.

Link to photo of screen:

https://drive.google.com/file/d/106ub3Dm5ACxu8BrvsuU89A_60QgPEYU_/view?usp=drivesdk

dhrone, 2019-03-28:
Quick questions needed to respond intelligently...

How is the display wired to the Pi?  Are you sure you have the pin settings correct?
Which Pi are you using?
What version of Volumio?
Have you gotten any other programs to run successfully?  If so, which programs?  

If you haven't, try instructions from  (e.g. https://cdn-learn.adafruit.com/downloads/pdf/drive-a-16x2-lcd-directly-with-a-raspberry-pi.pdf).

slawek125, 2019-03-28:
Hello thanks for Your reply.
Display is wired like in your link.
checked pin wiring and its correct.
Volumio version latest.
Raspberry pi3 b V1.2

In old version of Volumio LCD works with plugin from Saiyato.

slawek125, 2019-03-28:
the problem was solved, after replacing the second power supply, the display functions properly.
The problem is in the power supply.

dhrone, 2019-03-28:
Glad to hear that you solved your problem.

closed 2019-03-28 by dhrone