dhrone/pydPiper #60: [max2play / lms] problem screen : no indication of music

GG700, 2019-04-13:
Dear Dhrone,

i am unable to install pydPiper in max2play (with lms).
I used the command below to test :
 sudo docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py

And the answer is :
pydPiper starting
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/app/sources/musicdata_lms.py", line 136, in run
    self.connect()
  File "/app/sources/musicdata_lms.py", line 96, in connect
    self.dataserver.connect()
  File "/usr/local/lib/python2.7/dist-packages/pylms/server.py", line 60, in connect
    self.get_players(update=update)
  File "/usr/local/lib/python2.7/dist-packages/pylms/server.py", line 151, in get_players
    player_count = self.get_player_count()
  File "/usr/local/lib/python2.7/dist-packages/pylms/server.py", line 181, in get_player_count
    return int(self.player_count)
ValueError: invalid literal for int() with base 10: 'hod Not Allowed: </BODY></HTML>'

The RASPDAC (audiophonics) indicates the hour, but not the music I play !

Do you have an idea ?

Jerome

dhrone, 2019-04-22:
It looks like the LMS service is returning an unexpected value to the pylms get_player_count function.  I don't have much time for debugging at the moment but when I do I'm going to need to know the version of max2play, the version of LMS, and whether you have any other LMS servers running on your local network besides the one running on the raspberryPi that is running max2play.

GG700, 2020-03-21:
I have installed pydPiper.
But my screen don't refresh my I change a music, change the volume, etc...
systemctl status of pydpiper.service returns the values :+1: 

Do you have an idea of the problem ?

Thanks

pi@ZIKsalon:~ $ systemctl status pydpiper.service
● pydpiper.service - pydPiper container
   Loaded: loaded (/etc/systemd/system/pydpiper.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-03-21 14:38:12 CET; 3min 50s ago
 Main PID: 339 (docker)
      CPU: 456ms
   CGroup: /system.slice/pydpiper.service
           └─339 /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw dhrone/pydpiper:latest python /app/py

mars 21 14:39:40 ZIKsalon docker[339]: Traceback (most recent call last):
mars 21 14:39:40 ZIKsalon docker[339]:   File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
mars 21 14:39:40 ZIKsalon docker[339]:     self.run()
mars 21 14:39:40 ZIKsalon docker[339]:   File "/app/pydPiper.py", line 218, in run
mars 21 14:39:40 ZIKsalon docker[339]:     self.musicdata[u'time_formatted'] = moment.utcnow().timezone(pydPiper_config.TIMEZO
mars 21 14:39:40 ZIKsalon docker[339]:   File "/usr/local/lib/python2.7/dist-packages/moment/core.py", line 81, in timezone
mars 21 14:39:40 ZIKsalon docker[339]:     date = times.to_local(date, zone)
mars 21 14:39:40 ZIKsalon docker[339]:   File "/usr/local/lib/python2.7/dist-packages/times/__init__.py", line 61, in to_local
mars 21 14:39:40 ZIKsalon docker[339]:     'First argument to to_local() should be a universal time.'
mars 21 14:39:40 ZIKsalon docker[339]: ValueError: First argument to to_local() should be a universal time.
lines 1-18/18 (END)

GG700, 2020-03-21:
I have one server lms and my config for the installation is in pydpiper.service 

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --lms --driver winstar_weg --width 80 --height 16 --rs 7 --e 8 --d4 25 --d5 24 --d6 23 --d7 27 --timezone 'France/Paris' --temperature celsius --pages pages_weh_80x16.py

GG700, 2020-04-22:
Hi dhrone,

My version of max2play is : Version 2.51.
My version of LMS : Logitech Media Server Version : 7.9.2 

I have only one LMS running on my local network.

Thanks for your answer !

dhrone, 2020-04-22:
Ok,  I'll take a look at that combination when I have some spare time.

GG700, 2020-10-18:
Hi,

I think that the problem is that I have a bad parameter when I run : <pre><font color="#8AE234"><b>pi@ZIKsalon</b></font>:<font color="#729FCF"><b>~/pydPiper $</b></font> sudo python configure.py
</pre>

I don't know what kind of parameter I have to use for the username and the password : 
<pre>MUSIC DISTRIBUTION

Name of distribution? [lms]: 
Server address? [localhost]: 
Port? [9000]: 
Username? []: LMS_USER
Password? []: LMS_PASSWORD
LMS Player MAC address? [B8:27:EB:61:A2:E5]: 
</pre>

I think I have no login / password in lms ??

Can you help me ?

dhrone, 2020-10-19:
Typically, no log in details are required.  So, I do not think this is a likely reason for your problem.  I still haven't had the time to look into the fault but will try to get to it this week.

GG700, 2021-01-10:
Hi,

Can you help me to configure pydpiper ?
The LMS server is installed in the same raspberry than the squeezebox client.
1-I apply the readme file instructions : 
```
sudo wget https://github.com/dhrone/pydPiper/archive/v0.3-alpha.tar.gz
sudo tar zxvf v0.3-alpha.tar.gz
cd pydPiper-0.3-alpha
```

2-I edit the pydpiper.service file. The hardware is a raspdac v3 (audiophonics DAC).

3-Last, I run the install script:
`sudo ./install.sh`

My problem is that the screen don't indicates anything.

[pydpiper-service.txt](https://github.com/dhrone/pydPiper/files/5793297/pydpiper-service.txt)

dhrone, 2021-01-11:
When dealing with display related issues, you should attempt to get the display running separately from pydPiper.  From the beginning of this issue, it appears you have had it working in the past.  Do you know what you have changed?

Looking at the config and service files you have shared (which was very useful) it appears that you are specifying the hd44780 driver in the service file but the winstar_weg driver in the config file.  Since the command line options take precedent, these settings would mean you are attempting to run your display as a HD44780 using an I2C style interface.  There is a different example within the service file that you could uncomment for the RaspdacV3.  I've attached a sample file that is configured for the RaspDac V3.

[pydpiper-service-example.txt](https://github.com/dhrone/pydPiper/files/5796031/pydpiper-service-example.txt)

GG700, 2021-01-11:
Hi,

Exact.
I tried with your file but I had the same problem : the screen indicates "stop" but don't refresh when I play a song.

I resolved the problem by editing file  pydPiper_config.py  : 

I writed LMS_PLAYER = u"B8-27-EB-61-A2-E5" instead of : LMS_PLAYER = u"B8:27:EB:61:A2:E5".
I writed TIMEZONE=u"Europe/Paris" instead of : TIMEZONE=u"FR/Mulhouse"

I have a last question : I try to use a 24 hour format.

I edited the file pages_lcd_16x2.py

And I replaced 
'time': { 'type':'text', 'format':'{0}', `'variables':['localtime|strftime+%-I:%M'],...`

By

'time': { 'type':'text', 'format':'{0}', `'variables':['localtime|strftime+%-H:%M'],...`

But it is not good...
Can you help me ?

thanks

dhrone, 2021-01-12:
Not without understanding what is expected and what the system is actually doing.  I'm going to make a guess though.  You are looking for a 24 hour clock that looks like...

`09:32`
instead of
`9:32`

If yes then your strftime should be %H:%M not %-H:%M.

See https://strftime.org/ for more info on strftime.

GG700, 2021-01-12:
I would like that the clock looks like : 
20:32
instead of 
8:32 PM

But the screen indicates always x:xx PM.

I modified the file pages_weh_80x16.py because it is the parameters used in the file pydpiper.service.

[pages_lcd_16x2.txt](https://github.com/dhrone/pydPiper/files/5804619/pages_lcd_16x2.txt)
[pages_weh_80x16.txt](https://github.com/dhrone/pydPiper/files/5804620/pages_weh_80x16.txt)
[pydpiper.service.txt](https://github.com/dhrone/pydPiper/files/5804621/pydpiper.service.txt)
[pydPiper_config.txt](https://github.com/dhrone/pydPiper/files/5804622/pydPiper_config.txt)