dhrone/pydPiper #62: 24hr Time format

Prfacia, 2019-04-28:
Hi,
how can I change the time zone and clock format for 24 hours?

dhrone, 2019-04-29:
You would make those changes in the page file.  Which one are you currently using?

Prfacia, 2019-04-29:
V0.3-alpha and lastest volumio. Thanks for answer.

dhrone, 2019-04-29:
Actually, I was referring to which page file you are using (e.g. pages_lcd_16x2.py).  That said, I can generically describe what you need to change.  There should be a line in the file you are using that looks similar to this...

`	'time': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%-I:%M'], 'font':'large', 'just':'right', 'varwidth':True, 'size':(70,16) },`

You need to modify the values that get passed to the strftime transformation.  General format for this can be found at [strftime.org](url).  The current value '+%-I:%M' produces a 12 hour clock.  If you switch it to '+%-H:%M' it will give you a 24 hour clock.

Prfacia, 2019-04-29:
I use Hd44780_i2c page.

dhrone, 2019-05-02:
Can you post your service file?  I suspect that you are not referencing the local copy of the page file from within the docker container.

Prfacia, 2019-05-02:
ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780_i2c  --width 100 --height 32 --i2caddress 0x27 --timezone 'Europe/Istanbul' --temperature celcius --pages pages_lcd_20x4.py

Prfacia, 2019-05-02:
Hi,

I've edited the wrong pydpiper.service file.
I have edited the pydpiper.service file under the /etc/systemd/system

My execstart, 

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper-0.3-alpha:/app:rw dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780_i2c --width 100 --height 32 --i2caddress 0x27 --timezone 'Europe/Istanbul' --temperature celcius --pages pages_lcd_20x4.py

reload daemon and restart pydpiper.

Everything OK.

Thanks for your support.

closed 2019-05-03 by dhrone