dhrone/pydPiper #63: Updating pages_lcd_16x2.py

dydep, 2019-04-29:
Hi,

I have a test build on a Pi3 with a 16x2 i2c display. the display works and I get the artist/track/time etc. So all good so far.

I am trying to edit the info displayed when pydPiper is running, but cannot see how to get the changes into the docker container. I have edited the file in the directory /home/volumio/pydPiper-0.3-alpha, stopped and started the service. However the pages file always seems to be the same with the default config

I know I'll be doing something stupid, but any pointers on how to get an updated pages file into the docker container running the display would be helpful

Thanks Phil

dhrone, 2019-04-29:
Hmm.  It should be referencing the version of the code that is outside of the container.  Can you share what your ExecStart line looks like in the service file?

dydep, 2019-04-30:
Hi,

Thanks for the quick response. 

Info below. I am using the 16x2 display i2c config and the only change I have made, other than uncommenting the line is to the timezone.

Directory listings for the local directory and the one in the container. The pages file in the container has the Oct 2017 date and not the modified one from 29/04. 

What am I doing wrong for the modified pages not to be pulled into the container, or is it stored somewhere else.

# HD44780 LCD 16x2 display with I2C interface on address 0x27
ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780_i2c  --width 80 --height 16 --i2caddress 0x27 --timezone 'GB' --temperature fahrenheit --pages pages_lcd_16x2.py

pydPiper directory
volumio@volumio:~/pydPiper-0.3-alpha$ ls -l
total 196
-rwxrwxr-x 1 root root   356 Oct 21  2017 Dockerfile
-rw-rw-r-- 1 root root  1073 Oct 21  2017 LICENSE
-rwxrwxr-x 1 root root 12238 Oct 21  2017 README.md
drwxrwxr-x 4 root root  4096 Oct 21  2017 displays
drwxrwxr-x 2 root root  4096 Oct 21  2017 docs
-rwxrwxr-x 1 root root   241 Oct 21  2017 install.sh
-rwxrwxr-x 1 root root  8776 Oct 21  2017 pages.py
-rwxrwxr-x 1 root root  8805 Oct 21  2017 pages_fixed.py
-rwxrwxr-x 1 root root  9766 Oct 21  2017 pages_fixed_jcu.py
-rwxrwxr-x 1 root root  8817 Oct 21  2017 pages_fixed_moode.py
-rwxrwxr-x 1 root root  8808 Apr 29 09:19 pages_lcd_16x2.py
-rwxrwxr-x 1 root root  8647 Oct 21  2017 pages_lcd_20x4.py
-rwxrwxr-x 1 root root  5237 Oct 21  2017 pages_ssd1306.py
-rwxrwxr-x 1 root root  5152 Oct 21  2017 pages_ssd1306_nottf.py
-rwxrwxr-x 1 root root  8347 Oct 21  2017 pages_weg_100x16.py
-rwxrwxr-x 1 root root  8776 Oct 21  2017 pages_weh_80x16.py
-rwxrwxr-x 1 root root 23765 Oct 21  2017 pydPiper.py
-rw-rw-r-- 1 root root  3649 Oct 21  2017 pydPiper_IR.py
-rwxrwxr-x 1 root root  2711 Oct 21  2017 pydPiper_config.py
-rwxrwxr-x 1 root root  4548 Apr 29 09:20 pydpiper.service
-rw-rw-r-- 1 root root    63 Oct 21  2017 requirements.txt
drwxrwxr-x 2 root root  4096 Oct 21  2017 sources
drwxrwxr-x 2 root root  4096 Oct 21  2017 utils

Inside the docker container
root@volumio:/app# ls -l
total 216
-rwxr--r-- 1 root root   356 Oct 21  2017 Dockerfile
-rw-r--r-- 1 root root  1073 Oct  1  2017 LICENSE
-rw-r--r-- 1 root root 11621 Oct  1  2017 README.md
drwxr-xr-x 4 root root  4096 Apr 30 07:04 displays
drwxr-xr-x 2 root root  4096 Oct 13  2017 docs
-rw-r--r-- 1 root root  3466 Oct 13  2017 lcd_i2c.py
-rwxr-xr-x 1 root root  8776 Oct 21  2017 pages.py
-rw-r--r-- 1 root root  7534 Oct 21  2017 pages.pyc
-rwxr--r-- 1 root root  8805 Oct 21  2017 pages_fixed.py
-rwxr--r-- 1 root root  9766 Oct 21  2017 pages_fixed_jcu.py
-rwxr--r-- 1 root root  8817 Oct 21  2017 pages_fixed_moode.py
-rwxr--r-- 1 root root  8140 Oct 21  2017 pages_lcd_16x2.py
-rw-r--r-- 1 root root  7084 Apr 30 07:04 pages_lcd_16x2.pyc
-rwxr--r-- 1 root root  8647 Oct 21  2017 pages_lcd_20x4.py
-rwxr--r-- 1 root root  5237 Oct 21  2017 pages_ssd1306.py
-rwxr--r-- 1 root root  5152 Oct 21  2017 pages_ssd1306_nottf.py
-rwxr--r-- 1 root root  8347 Oct 21  2017 pages_weg_100x16.py
-rwxr--r-- 1 root root  8776 Oct 21  2017 pages_weh_80x16.py
-rwxr-xr-x 1 root root 23765 Oct 21  2017 pydPiper.py
-rw-r--r-- 1 root root  3649 Oct  1  2017 pydPiper_IR.py
-rwxr-xr-x 1 root root  2711 Oct 21  2017 pydPiper_config.py
-rw-r--r-- 1 root root  1297 Apr 30 07:04 pydPiper_config.pyc
-rwxr-xr-x 1 root root  4429 Oct 21  2017 pydpipervolumio.service
-rw-r--r-- 1 root root    63 Oct  1  2017 requirements.txt
drwxr-xr-x 2 root root  4096 Apr 30 07:04 sources
drwxr-xr-x 2 root root  4096 Oct  1  2017 utils
-rwxr-xr-x 1 root root   306 Oct 13  2017 volumio_install.sh

dhrone, 2019-04-30:
The problem is you have not granted the program permission to access the files outside of the container.  Change your ExecStart to be...

`ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:rw dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver hd44780_i2c --width 80 --height 16 --i2caddress 0x27 --timezone 'GB' --temperature fahrenheit --pages pages_lcd_16x2.py`

Notice the second -v in the command?  This tells docker to replace the /app directory in the container with the /home/volumio/pydPiper directory from the local filesystem.  Now any changes you make within /home/volumio/pydPiper will be referenced when the program is run.

dhrone, 2019-04-30:
FYI, I'm considering removing the dependency on docker in a future release.  It was an attempt to avoid some occasionally painful dependencies on the various distributions that pydPiper supports but it has turned out to be more trouble than it was worth.

dydep, 2019-04-30:
Hi,

I have tried the suggestion above, unfortunately no luck. The docker container just starts and stops. I have tried the following
Clean rebuild of Volumio, using the command suggested above, but changing the sub-directory to be pydPiper-0.3-alpha
Using the directory from the updated ExecStart above and just copying the pages file in to it
In both these scenarios, docker just bounces and won't stay up. If I swap out to use the template example, everything starts Ok
I am copying the pydpiper.service file to /etc/systemd/system between each change and stopping and starting the service, basically running the last three lines of the install.sh file each time after stopping the service
sudo cp pydpiper.service /etc/systemd/system
sudo systemctl enable pydpiper
sudo systemctl start pydpiper

dydep, 2019-04-30:
Ok, so getting somewhere now, thankfully. Looking at /var/logs pydPiper was giving a syntax error when starting. I had copied the example script in to give a simple display, but there are a couple of typos
'time': { 'type':'text', 'format:Time\n{0}',
should be
'time': { 'type':'text', 'format':'Time\n{0}',
I now get "No active widget". But hopefully, I can work out the pages file from your docs and docker is running fine.

Many thanks for the quick replies and pointing me in the right direction. I'll close the ticket as my initial query is now resolved

Phil

closed 2019-04-30 by dydep

dydep, 2019-04-30:
Sorted my simple display by using following
New canvas - phil
'phil': { 'widgets': [ ('artist',0,0), ('title',0,8), ('songprogress',65,8) ], 'size':(80,16) }
Updated seqPlay
        {
                'name': 'seqPlay',
                'canvases': [
                        { 'name':'phil', 'duration':8, 'conditional':"not db['stream']=='webradio'" },
                ],
                'conditional': "db['state']=='play'"
        },

May play further, but for now I have what my aim was. 

Thanks again