dhrone/pydPiper #64: Music info via I2c

Prfacia, 2019-04-30:
Hi,

Can you add all music info (album, artist, bitrate,player status etc.) via i2c , so different screens can be used.It can be activated from the pydpiper.service file like a new screen page.

For example,I want to use color display.

would be perfect.

thanks for your efforts.

dhrone, 2019-05-03:
Color is not supported currently but you can certainly redesign the page file to display whatever music data that your distribution is supplying.  Instructions are provided in the docs folder.

Prfacia, 2019-05-03:
Sorry ,my english is not very good so, couldn't tell you exactly what I mean.

I just want all info datas via i2c.I will design other jobs myself with another hardware.

dhrone, 2019-05-04:
I'm not sure I understand your question.  pydPiper supports some i2c displays but it is sending commands over the i2c interface in the protocol of the display itself.  It does not send the music information itself over the i2c interface but rather the commands necessary to get the display to render the information to the display.

Perhaps it would be easier to ask your question in your native language and then use google translate to convert it into English.

Prfacia, 2019-05-04:
Hi,

My question is the same as this title.

https://github.com/dhrone/pydPiper/issues/50

Thanks...

Prfacia, 2019-06-06:
Hi,
Last situation,thanks dhrone...

https://www.youtube.com/watch?v=oVoHwCJgBYw&t=21s

manolomalele, 2020-01-30:
Hi Prfacia, congrats for your excellent work!
I've built a media player looking like your and I'm using pydpiper at its basic with just a couple of customizations.
I would like to ask you about your page file in order to get myself the info you have on your display. I'm struggling with file type and bitrate. How did you add them on the file?
I've tried but my system fails both on volumio and moode audio.
Can you please give me some hints?
That would be really appreciated 
Thank you

Prfacia, 2020-02-07:
Hi manolomalele,thanks....
you will make the changes in the lcd file, for me pages_lcd_20x4.py

Prfacia, 2020-02-07:
**my lcd file for example**

#!/usr/bin/python.pydPiper
# coding: UTF-8

from __future__ import unicode_literals

# Page Definitions
# See Page Format.txt for instructions and examples on how to modify your display settings

# Load the fonts needed for this system
FONTS = {
	'small': { 'default':True, 'file':'latin1_5x8_lcd.fnt','size':(5,8) },
	'large': { 'file':'BigFont_10x16_fixed.fnt', 'size':(10,16) },
	'tiny': { 'file':'upperasciiwide_3x5_fixed.fnt', 'size':(5,5) },
}

IMAGES = {
	'progbar': {'file':'progressbar_100x8.png' },
}

# Load the Widgets that will be used to produce the display pages
WIDGETS = {
	'splash': { 'type':'text', 'format':'   V O L U M I O\n      v 2.575\npydPiper plugin v0.3\n IP : ', 'font':'small' },
	'iptext': { 'type':'text', 'format':'{0}', 'variables':['ip'], 'font':'small', 'varwidth':True},
	'nowplaying': { 'type':'text', 'format':'{0}', 'variables':['actPlayer|upper'], 'font':'small', 'varwidth':True},
	#'nowplayingdata': { 'type':'text', 'format':'{0} OF {1}', 'variables':['playlist_position', 'playlist_length'], 'font':'small', 'just':'right','size':(40,8),'varwidth':True},
	'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',5,1,20,'onloop',3,100) },
	'artist': { 'type':'text', 'format':'{0}', 'variables':['artist'], 'font':'small','varwidth':True,'effect':('scroll','left',5,1,20,'onloop',3,100)},
	'artistradio': { 'type':'text', 'format':'{0}', 'variables':['artist'], 'font':'small','varwidth':True},
	'album': { 'type':'text', 'format':'{0}', 'variables':['album'], 'font':'small','varwidth':True,'effect':('scroll','left',5,1,20,'onloop',3,100)},
	'time': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%-H:%M'], 'font':'large', 'just':'right', 'varwidth':True, 'size':(65,16) },
	'timesmall': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%-H:%M'], 'font':'small', 'just':'right', 'varwidth':True, 'size':(40,8) },
	#'time': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%-I:%M'], 'font':'large', 'just':'right', 'varwidth':True, 'size':(65,16) },
	#'timesmall': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%-I:%M'], 'font':'small', 'just':'right', 'varwidth':True, 'size':(40,8) },
	#'ampm': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%p'], 'font':'small', 'varwidth':True },
	'temp': { 'type':'text', 'format':'{0}', 'variables':['outside_temp_formatted'], 'font':'small', 'just':'right', 'size':(25,8) },
	'temphilow': { 'type':'text', 'format':'H {0}\nL {1}', 'variables':['outside_temp_max|int', 'outside_temp_min|int'], 'font':'small', 'just':'right', 'size':(25,16) },
	'conditions': { 'type':'text', 'format':'{0}', 'variables':['outside_conditions|capitalize'], 'font':'small','varwidth':True, 'size':(55,16), 'effect':('scroll','left',5,1,20,'onloop',3,55)},
	'conditions2': { 'type':'text', 'format':'{0}', 'variables':['outside_conditions|capitalize'], 'font':'small','varwidth':True, 'size':(75,8), 'effect':('scroll','left',5,1,20,'onloop',3,75)},
	'radio': { 'type':'text', 'format':"RADIO", 'font':'small', 'varwidth':True, 'size':(25,8), 'just':'right' },
	'volume': { 'type':'text', 'format':'VOLUME ({0})', 'variables':['volume'], 'font':'small', 'varwidth':True, 'just':'left', 'size':(95,8)},
	'volumebar': { 'type':'progressimagebar', 'image':'progbar','value':'volume', 'rangeval':(0,100) },
	'songprogresstext': { 'type':'text', 'format':'SONG', 'font':'small' },
	'songprogress': { 'type':'progressbar', 'value':'elapsed', 'rangeval':(0,'length'), 'size':(40,8) },
	'trkprogresstext': { 'type':'text', 'format':'LST', 'font':'small' },
	'trkprogress': { 'type':'progressbar', 'value':'playlist_position', 'rangeval':(1,'playlist_length'), 'size':(25,8) },
	'elapsed': { 'type':'text', 'format':'{0}', 'variables':['elapsed_formatted'], 'font':'small' },
	#'playstopsymbol': { 'type':'text', 'format':'{0}', 'variables':['state|select+play+\ue000+stop+\ue001'], 'font':'small', 'just':'left'  },
	'randomsymbol': { 'type':'text', 'format':'\ue002 ', 'font':'large', 'varwidth':True, 'size':(10,16) },
	'random': { 'type':'text', 'format':'Random\n{0}', 'variables':['random|onoff|Capitalize'], 'font':'small', 'varwidth':True, 'size':(65,16) },
	'repeatoncesymbol': { 'type':'text', 'format':'\ue003 ', 'font':'large', 'varwidth':True, 'size':(10,16) },
	'repeatonce': { 'type':'text', 'format':'Repeat Once\n{0}', 'variables':['single|onoff|Capitalize'], 'font':'small', 'varwidth':True, 'just':'center', 'size':(65,16) },
	'repeatallsymbol': { 'type':'text', 'format':'\ue004 ', 'font':'large', 'varwidth':True, 'size':(10,16) },
	'repeatall': { 'type':'text', 'format':'Repeat All\n{0}', 'variables':['repeat|onoff|Capitalize'], 'font':'small', 'varwidth':True, 'size':(65,16) },
	'showplay': { 'type':'text', 'format':'\ue000 PLAY', 'font':'large', 'varwidth':True, 'just':'left', 'size':(100,32) },
	'showstop': { 'type':'text', 'format':'\ue001 STOP', 'font':'large', 'varwidth':True, 'just':'left', 'size':(100,32) },
	'samplerate': { 'type':'text', 'format':'{0}', 'variables':['samplerate'], 'font':'small' },
	'bitdepth': { 'type':'text', 'format':'{0}', 'variables':['bitdepth'], 'font':'small', 'varwidth':True, 'just':'left' },
	'musictype': { 'type':'text', 'format':'{0}', 'variables':['tracktype'], 'font':'small', 'varwidht':True, 'just':'left' },
[... cut ...]
	{
		'name':'seqAnnounceSingle',
		'canvases': [ { 'name':'showrepeatonce', 'duration':2 } ],
		'conditional': "db['single'] != dbp['single']",
		'minimum':2,
	},
	{
		'name':'seqAnnounceRepeat',
		'canvases': [ { 'name':'showrepeatall', 'duration':2 } ],
		'conditional': "db['repeat'] != dbp['repeat']",
		'minimum':2,
	},
	{
		'name':'seqAnnounceTooHot',
		'canvases': [ { 'name':'temptoohigh', 'duration':5 } ],
		'conditional': "db['system_tempc'] > 85",
		'minimum':5,
		'coolingperiod':30
	}
]

Prfacia, 2020-02-07:
# My LCD for example

WIDGETS = {
	'splash': { 'type':'text', 'format':'pydPiper\nStarting...'   V O L U M I O\n      v 2.575\npydPiper plugin v0.3\n IP : ', 'font':'small' },
	'iptext': { 'type':'text', 'format':'{0}', 'variables':['ip'], 'font':'small', 'varwidth':True},
	'nowplaying': { 'type':'text', 'format':'{0}', 'variables':['actPlayer|upper'], 'font':'small', 'varwidth':True},
	#'nowplayingdata': { 'type':'text', 'format':'{0} OF {1}', 'variables':['playlist_position', 'playlist_length'], 'font':'small', 'just':'right','size':(40,8),'varwidth':True},
	'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',5,1,20,'onloop',3,100) },
	'artist': { 'type':'text', 'format':'{0}', 'variables':['artist'], 'font':'small','varwidth':True,'effect':('scroll','left',5,1,20,'onloop',3,100)},
	'artistradio': { 'type':'text', 'format':'{0}', 'variables':['artist'], 'font':'small','varwidth':True},
	'album': { 'type':'text', 'format':'{0}', 'variables':['album'], 'font':'small','varwidth':True,'effect':('scroll','left',5,1,20,'onloop',3,100)},
	'time': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%-IH:%M'], 'font':'large', 'just':'right', 'varwidth':True, 'size':(65,16) },
	'timesmall': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%-IH:%M'], 'font':'small', 'just':'right', 'varwidth':True, 'size':(40,8) },
		#'time': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%-I:%M'], 'font':'large', 'just':'right', 'varwidth':True, 'size':(65,16) },
	#'timesmall': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%-I:%M'], 'font':'small', 'just':'right', 'varwidth':True, 'size':(40,8) },
	#'ampm': { 'type':'text', 'format':'{0}', 'variables':['localtime|strftime+%p'], 'font':'small', 'varwidth':True },
	'temp': { 'type':'text', 'format':'{0}', 'variables':['outside_temp_formatted'], 'font':'small', 'just':'right', 'size':(25,8) },
	'temphilow': { 'type':'text', 'format':'H {0}\nL {1}', 'variables':['outside_temp_max|int', 'outside_temp_min|int'], 'font':'small', 'just':'right', 'size':(25,16) },
	'conditions': { 'type':'text', 'format':'{0}', 'variables':['outside_conditions|capitalize'], 'font':'small','varwidth':True, 'size':(55,16), 'effect':('scroll','left',5,1,20,'onloop',3,55)},
	'conditions2': { 'type':'text', 'format':'{0}', 'variables':['outside_conditions|capitalize'], 'font':'small','varwidth':True, 'size':(75,8), 'effect':('scroll','left',5,1,20,'onloop',3,75)},
	'radio': { 'type':'text', 'format':"RADIO", 'font':'small', 'varwidth':True, 'size':(25,8), 'just':'right' },
	'volume': { 'type':'text', 'format':'VOLUME ({0})', 'variables':['volume'], 'font':'small', 'varwidth':True, 'just':'left', 'size':(95,8)},
	'volumebar': { 'type':'progressimagebar', 'image':'progbar','value':'volume', 'rangeval':(0,100) },
	'songprogresstext': { 'type':'text', 'format':'SONG', 'font':'small' },
	'songprogress': { 'type':'progressbar', 'value':'elapsed', 'rangeval':(0,'length'), 'size':(40,8) },
	'trkprogresstext': { 'type':'text', 'format':'TRK''LST', 'font':'small' },
	'trkprogress': { 'type':'progressbar', 'value':'playlist_position', 'rangeval':(1,'playlist_length'), 'size':(25,8) },
	'elapsed': { 'type':'text', 'format':'{0}', 'variables':['elapsed_formatted'], 'font':'small' },
	#'playstopsymbol': { 'type':'text', 'format':'{0}', 'variables':['state|select+play+\ue000+stop+\ue001'], 'font':'small', 'just':'left'  },
	'randomsymbol': { 'type':'text', 'format':'\ue002 ', 'font':'large', 'varwidth':True, 'size':(10,16) },
	'random': { 'type':'text', 'format':'Random\n{0}', 'variables':['random|onoff|Capitalize'], 'font':'small', 'varwidth':True, 'size':(65,16) },
	'repeatoncesymbol': { 'type':'text', 'format':'\ue003 ', 'font':'large', 'varwidth':True, 'size':(10,16) },
	'repeatonce': { 'type':'text', 'format':'Repeat Once\n{0}', 'variables':['single|onoff|Capitalize'], 'font':'small', 'varwidth':True, 'just':'center', 'size':(65,16) },
	'repeatallsymbol': { 'type':'text', 'format':'\ue004 ', 'font':'large', 'varwidth':True, 'size':(10,16) },
	'repeatall': { 'type':'text', 'format':'Repeat All\n{0}', 'variables':['repeat|onoff|Capitalize'], 'font':'small', 'varwidth':True, 'size':(65,16) },
	'showplay': { 'type':'text', 'format':'\ue000 PLAY', 'font':'large', 'varwidth':True, 'just':'left', 'size':(100,32) },
	'showstop': { 'type':'text', 'format':'\ue001 STOP', 'font':'large', 'varwidth':True, 'just':'left', 'size':(100,32) },
	'samplerate': { 'type':'text', 'format':'{0}', 'variables':['samplerate'], 'font':'small' },
	'bitdepth': { 'type':'text', 'format':'{0}', 'variables':['bitdepth'], 'font':'small', 'varwidth':True, 'just':'left' },
	'musictype': { 'type':'text', 'format':'{0}', 'variables':['tracktype'], 'font':'small', 'varwidht':True, 'just':'left' },
	'temptoohigh': { 'type':'text', 'format':'\ue005 Warning System Too Hot ({0})', 'variables':['system_temp_formatted'], 'font':'large', 'varwidth':True, 'effect':('scroll','left',5,1,20,'onstart',3,80) }
}

CANVASES = {
	'playartist': { 'widgets': [ ('musictype',45,0), ('bitdepth',60,16), ('samplerate',0,16), ('artist',0,8), ('nowplaying',0,0), ('nowplayingdata',60,0), ('playstopsymbol', 50,16), ('songprogresstext',0,24), ('songprogress',20,24), ('trkprogresstext',60,24), ('trkprogress',75,24) ], 'size':(100,32) },
[... cut ...]
	},
	{
		'name':'seqAnnounceSingle',
		'canvases': [ { 'name':'showrepeatonce', 'duration':2 } ],
		'conditional': "db['single'] != dbp['single']",
		'minimum':2,
	},
	{
		'name':'seqAnnounceRepeat',
		'canvases': [ { 'name':'showrepeatall', 'duration':2 } ],
		'conditional': "db['repeat'] != dbp['repeat']",
		'minimum':2,
	},
	{
		'name':'seqAnnounceTooHot',
		'canvases': [ { 'name':'temptoohigh', 'duration':5 } ],
		'conditional': "db['system_tempc'] > 85",
		'minimum':5,
		'coolingperiod':30
	}

manolomalele, 2020-05-19:
Hi, i tryed adding this:
`'musictype': { 'type':'text', 'format':'{0}', 'variables':['tracktype'], 'font':'small', 'varwidht':True, 'just':'left' },`
but it only displays the path, not the codec like on your video...