dhrone/pydPiper #68: Rune install fails since 13/9, unable to install pillow

Alien8ions, 2019-09-15:
Hi

I've had the need to reinstall rune 0.5, and I'm testing the new Re 1.1 version as well.
Since yesterday the install script fails because PIL module can't load.
There are timeout errors at the end of the install script.

Is it possible to install manually?

Diego

dhrone, 2019-09-16:
Yes, that should work.

I'm wondering though if the problem might relate to the virtual environment that pipenv maintains.  Did the install script through an error when it was attempting to install Pillow?

Alien8ions, 2019-09-17:
Yes, it fails  with error. Url not found, timeout while downloading.

Couldn't catch it, but its a long block of errors.

Meanwhile I've managed to use volumio dockerized install and adjusting for rune.
It now runs again.

Don't know if it is useful but docker can be installed using pacman manager. At first that failed as well because of version error on containerd dependency. So installed that one seperately to get it to work.

I didn't try  to install pillow using pip.

Irony is that I now have a music making clock, because of reinstall on new Rune Re1 version it seems the status updates through redis are broken, so needed fresh install for debug
D

Alien8ions, 2019-09-27:
pacman -S docker works fine with adapted .service file
Script runs fine now...

Issue solved

closed 2019-09-27 by Alien8ions